openai/mujoco-py #539: Import error. Trying to rebuild mujoco_py. running build_ext building 'mujoco_py.cymj' extension distutils.errors.CompileError: command '/usr/local/bin/gcc-9' failed with exit status 1

MyJumperBroke23, 2020-05-14:
**Describe the bug**
Error Importing Mujoco-py

**To Reproduce**
import mujoco-py, then try to use it

**Expected behavior**
Mujoco-py should import fine without error

**Error Messages**
```
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -DONMAC -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py -I/Users/alex/.mujoco/mujoco200/include -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/numpy/core/include -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/include -I/Library/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/cymj.c -o /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/cymj.o -fopenmp -w
/usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -DONMAC -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py -I/Users/alex/.mujoco/mujoco200/include -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/numpy/core/include -I/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/include -I/Library/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.c -o /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.o -fopenmp -w
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/wait.h:110,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdlib.h:66,
                 from /Users/alex/.mujoco/mujoco200/include/mujoco.h:33,
                 from /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/gl/glshim.h:4,
                 from /Users/alex/.virtualenvs/ConditionalMultiTaskLearning/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.c:1:
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/resource.h: In function 'getiopolicy_np':
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/resource.h:447:34: error: expected declaration specifiers before '__OSX_AVAILABLE_STARTING'
```
Ending with: 
`distutils.errors.CompileError: command '/usr/local/bin/gcc-9' failed with exit status 1`

**Desktop (please complete the following information):**
 - OS: macOS 10.15.4
 - Python 3.7
 - Mujoco200_macOS
 - mujoco-py 2.0.2.10

I have tried the solutions outlined in #463 , but nothing has worked so far.

MyJumperBroke23, 2020-05-15:
To anyone else having this issue: I downloaded mjpro150 and it works now.

shengpu1126, 2020-05-21:
Hi @MyJumperBroke23, I am experiencing the same compile error when importing on macOS 10.15.4. Could you please provide more details on how you fixed it? Thank you very much!

MyJumperBroke23, 2020-05-22:
@shengpu1126 I downloaded MJPro 150 osx from the [Mujoco website](https://www.roboti.us/index.html), and unziped it into usr/.mujoco

shengpu1126, 2020-05-22:
I see. What version of mujoco-py are you using, and did you install from source? I installed the latest version from pip (2.0.2.10) but by default it's looking for the `mujoco200` directory, so even though I have `mjpro150` it's still using the `mujoco200` and fails to compile. Also, did you have gcc@9 from homebrew? 

https://github.com/openai/mujoco-py/blob/1fe312b09ae7365f0dd9d4d0e453f8da59fae0bf/mujoco_py/utils.py#L86

MyJumperBroke23, 2020-05-22:
@shengpu1126 I installed the latest version (2.0.2.10) using pip like you did. I installed gcc using homebrew, and its version is 9.3.0_1. I installed mjpro150 after changing my sdk as outlined in #463 but that shouldn't have made a difference.

shengpu1126, 2020-05-22:
@MyJumperBroke23 Thanks for your help! 

I tried a few different things but here's what worked for me:

1. create a virtual environment
2. install the latest version of mujoco-py (2.0.2.10) with pip
3. changing the `builder.py` script to use the default gcc at `/usr/local/bin/gcc` (which is an alias for clang)
https://github.com/openai/mujoco-py/blob/1fe312b09ae7365f0dd9d4d0e453f8da59fae0bf/mujoco_py/builder.py#L313

4. Install llvm and set the following env variables, this is from the [README on issues with openmp](https://github.com/openai/mujoco-py#youre-on-macos-and-you-see-clang-error-unsupported-option--fopenmp):
```
brew install llvm
export PATH="/usr/local/opt/llvm/bin:$PATH"
export CC="/usr/local/opt/llvm/bin/clang"
export CXX="/usr/local/opt/llvm/bin/clang++"
export CXX11="/usr/local/opt/llvm/bin/clang++"
export CXX14="/usr/local/opt/llvm/bin/clang++"
export CXX17="/usr/local/opt/llvm/bin/clang++"
export CXX1X="/usr/local/opt/llvm/bin/clang++"
export LDFLAGS="-L/usr/local/opt/llvm/lib"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
```
5. Build the cython extension by running `python -c 'import mujoco_py'`
6. Install an older version of openai gym (v0.15.3) that supports mujoco-py 2.0 version