openai/mujoco-py #542: Generate Different sim.data.qpos by Following Install instruction

Iwtbm, 2020-05-31:
**Describe the bug**
I install the mujoco.py following the instruction by pip3, and follow the step to play with mujoco.py. But the sim.data.qpos are different. 

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
The sim.data.qpos should be [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.] as shown on the instruction page.

**Error Messages**
But mine is [0.  0.  1.4 1.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0. 0.  0.  0.  0.  0.  0.  0.  0.  0.  0. ]. 
I am wondering if the mujoco works correct. 

**Desktop (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Python Version [3.7
 - Mujoco Version [2.0]
 - mujoco-py version [2.0.2.10]

guyuwuyu, 2021-07-05:
same issue. Has it solved?

guyuwuyu, 2021-07-05:
I have checked the issues on two different devices, finding the generated data are both [0. 0. 1.4 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. ].