openai/mujoco-py #543: Fix joint of mujoco xml file to certain angle

andreasbinder, 2020-06-03:
I have a model of a snake-like robot and wanted to train it under suboptimality. To do so, I wanted to fix one joint that connects the body parts to a certain angle. I was wondering which way is the best or recommended one. This might work be setting the action space for the one joint to zero but I am not sure where to perform that. I am grateful to get some ideas! Currently, I am using the baselines implementation of baselines to get trajectories.

wukongwong, 2024-07-16:
Did you find the solution of this? I found there is no "fixed" option in the joint type in xml.

certum-ai, 2024-07-16:
Hi, I actually did not continue looking into it :( I think restricting angles etc should be possible. Good luck!

andreasbinder, 2024-07-16:
Hi, I actually did not continue looking into it :( I think restricting angles etc should be possible. Good luck!