openai/mujoco-py #544: (Only) when importing in debug mode: name 'raw_input' is not defined

famura, 2020-06-08:
**Describe the bug**
When I run `import mujoco_py` using the debugger in CLion (v2020.1.2, but also with a pervious version), the program chrashed with the error message below. Running this script works perfectly fine.

**To Reproduce**
Set up a system as descibed below (I think Python 3.8 is decisive here), a debugger, and try importing the module.

**Expected behavior**
Same behavior as for running the py-file without a debugger.

**Error Messages**
pydev debugger: process 21411 is connecting
Connected to pydev debugger (build 201.7846.88)
Traceback (most recent call last):
  File "<string>", line 25, in <module>
NameError: name 'raw_input' is not defined

**Desktop (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Python Version [3.8.3]
 - Mujoco Version [2.0]
 - mujoco-py version [2.0.2.10]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
   /home/$USER/.mujoco/mujoco200/bin

**Additional Information**
I tried all possible settings of "PyQt compatible" in the IDE's settings. It didn't make a difference.

HosikChae, 2020-09-26:
Same here. I am running on PyCharm on macOS 10.15.6, and I meet two `raw_input` errors in a row only when debugging.
I tracked down along its import chain, and it seems the errors occur when loading some related cython extensions.

The errors are  happening at `loader = ExtensionFileLoader(name, path)` (line 123)
    called by `mod = load_dynamic_ext('cymj', cext_so_path)` (line 101)
    called by `cymj = load_cython_ext(mjpro_path)` (line 503) in `mujoco_py/builder.py`
when loading `cymj` from `mujoco_py/generated/cymj_1.50.1.68_38_macextensionbuilder_38.so`.

By the way, I am using `mujoco-py` through `gym`, it is referring to `~/.mujoco/mjpro150`.

Jendker, 2020-10-22:
Same issue here. Makes the debugging very difficult if one is using mujoco with python 3.8.

gauthamvasan, 2020-11-23:
I had the same issue. I temporarily sidestepped the issue by switching to `python 3.7`. Debugger works on pycharm if I use my conda env with python 3.7

Peter-zds, 2021-05-13:
> I had the same issue. I temporarily sidestepped the issue by switching to `python 3.7`. Debugger works on pycharm if I use my conda env with python 3.7

Thanks

PeterMitrano, 2021-11-09:
FWIW, turning off "attach to subprocesses" fixes this problem for me. See:
https://www.jetbrains.com/help/pycharm/debugger-python.html
https://youtrack.jetbrains.com/issue/PY-44481

closed 2021-11-20 by famura

nopaddleboat, 2023-08-11:
I have the same issue in vscode. As suggested by @PeterMitrano I added ""subProcess": false," to vscode launch file, problem solved.