openai/mujoco-py #545: GLEW initialization error: Missing GL version

maple52046, 2020-06-14:
Hi,

I'm a new user of Mujoco, I want to run a python program with `mujoco_py` on Windows 10:

```python
# model = xxx
sim = mujoco_py.MjSim(model)
sim.render(width=32, height=32, camera_name="birdview")
```

And I got an error when the `render` function is running:

```
ERROR: GLEW initialization error: Missing GL version
```

---

Here are the software version:

- Python: 3.7.6 (anaconda)
- Mujoco: 150
- Cuda: v10.0.130

And Python packages:

```
mujoco-py==2.0.2.0
PyOpenGL==3.1.5
```

> The `simulate.exe` of Mujoco is worked. So I think Mujoco and mujoco-py are installed well.

How can I fix this issue?

Niels-ns, 2021-05-27:
Did you find a solution?

maple52046, 2021-06-02:
Unfortunately, not yet.
We just move the development env to Linux.

Yiwen233, 2022-07-20:
Sad to use mojoco on windows..