openai/mujoco-py #547: Cannot build mujoco_py.cymj

shiningsunnyday, 2020-06-26:
Hello, I was following the instruction on https://github.com/openai/mujoco-py and hit this when trying to import mujoco_py in Python shell. 

**System information**
OS: Ubuntu 18.04
brew: Installed linuxbrew following https://idroot.us/install-homebrew-ubuntu-18-04/
GCC: in /home/ubuntu/.linuxbrew/Cellar/gcc/5.5.0_7
Python: 3.6.9

import mujoco_py
Compiling /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/cymj.pyx because it depends on /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/pxd/mujoco.pxd.
[1/1] Cythonizing /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py
creating /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/gl
gcc -pthread -B /home/ubuntu/anaconda3/envs/mujoco_conda/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py -I/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py -I/home/ubuntu/.mujoco/mujoco200/include -I/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/numpy/core/include -I/home/ubuntu/anaconda3/envs/mujoco_conda/include/python3.6m -c /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
In file included from /home/ubuntu/.linuxbrew/Cellar/gcc/5.5.0_7/lib/gcc/x86_64-unknown-linux-gnu/5.5.0/include-fixed/syslimits.h:7:0,
                 from /home/ubuntu/.linuxbrew/Cellar/gcc/5.5.0_7/lib/gcc/x86_64-unknown-linux-gnu/5.5.0/include-fixed/limits.h:34,
                 from /home/ubuntu/anaconda3/envs/mujoco_conda/include/python3.6m/Python.h:11,
                 from /home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/cymj.c:54:
/home/ubuntu/.linuxbrew/Cellar/gcc/5.5.0_7/lib/gcc/x86_64-unknown-linux-gnu/5.5.0/include-fixed/limits.h:168:61: fatal error: limits.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 273, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/ubuntu/anaconda3/envs/mujoco_conda/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1

arjung128, 2020-08-18:
Hi @shiningsunnyday, were you able to resolve this issue?

csingh27sewts, 2021-01-23:
I still have the same issue

hayday100, 2022-02-04:
The only work-around I found is to use Ubuntu 16.04 and gcc-7 to compile mujoco-py and it works.  In Ubuntu 18.04, I tried all gcc versions 7, 8, 9, 10, 11 and all of them gave the above error.  So if you are still craving for a work-around, try it.  Different Ubuntu versions have so much difference that I installed 16.04, 18.04 and 20.04 on different disks and switch between them for different tasks to workaround frequently.

hayday100, 2022-02-04:
If your read the error carefully, there is a missing header file limits.h.  This implicates you are missing certain libraries.  I receive a similar error and I googled it.  It turns out I missed libglew-dev, so my fix is:
sudo apt install libglew-dev

I also googled limits.h and it seems that the libc-dev has it.  So the fix might be:
sudo apt install libc-dev