openai/mujoco-py #549: libmujoco200.so: undefined symbol: __glewBindBuffer

guoyaq, 2020-07-02:
when I run 
mjlib = cdll.LoadLibrary(libfile)
please help me
Traceback (most recent call last):
  File "/home/gyq/anaconda2/envs/rllab3/lib/python3.5/site-packages/IPython/core/interactiveshell.py", line 2862, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-13-da7c8aad0fe8>", line 1, in <module>
    mjlib = cdll.LoadLibrary(libfile)
  File "/home/gyq/anaconda2/envs/rllab3/lib/python3.5/ctypes/__init__.py", line 425, in LoadLibrary
    return self._dlltype(name)
  File "/home/gyq/anaconda2/envs/rllab3/lib/python3.5/ctypes/__init__.py", line 347, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/gyq/ku/rllab/vendor/mujoco/libmujoco200.so: undefined symbol: __glewBindBuffer

ganlinzhang, 2020-09-07:
hello，did you solve this problem？ I meet the same problem.

frank-xwang, 2020-10-25:
Hi, I also met this problem, have you solved it? Thanks

Ji4chenLi, 2020-10-30:
Same problem here.

csufangyu, 2020-11-14:
same problem +1

csufangyu, 2020-11-15:
mybe you can try to : 

`sudo apt-get install libglfw3 libglew2.0`

xixinzhang, 2021-01-10:
same problem++

sheilaschoepp, 2021-07-08:
to fix this now, install libglew2.1

sheilaschoepp, 2021-07-08:
to fix this: `sudo apt-get install libglfw3 libglew2.1`

DanielTakeshi, 2021-08-03:
^ I can confirm, after getting the same error, I checked my machine:

```
(d4rl) seita@lowland:~/d4rl (master) $ (dpkg-query -W -f='${Status}' libglfw3 2>/dev/null | grep -c "ok installed")
1
(d4rl) seita@lowland:~/d4rl (master) $ (dpkg-query -W -f='${Status}' libglew2.0 2>/dev/null | grep -c "ok installed")
0
(d4rl) seita@lowland:~/d4rl (master) $
```

And I did not have libglew2.0. Getting that there worked.

@sheilaschoepp I think `libglew2.1` should be `libglew2.0`

sheilaschoepp, 2021-08-03:
@ DanielTakeshi
Maybe it depends on your system???  For me, libglew2.0 did not exist.  I am running Ubuntu 20.04.

Riften, 2022-07-23:
libglew2.1 should be libglew2.2 for ubuntu 22.04.

cxyang1997, 2022-07-27:
> to fix this: `sudo apt-get install libglfw3 libglew2.1`

I tried this. But still encountered `/.mujoco/mujoco210/bin/libmujoco210.so: undefined symbol: __glewBindBuffer`. Any one met this before?