openai/mujoco-py #550: macOS Catalina: "ld: unknown option: -platform_version" when importing and building mujoco_py

ferreirafabio, 2020-07-03:
**Describe the bug**
When importing mujoco_py, the cymj extension build fails with a clang-10 error.

**To Reproduce**
By simply following the installation from the readme on macOS Catalina (10.15.5) with the newest Xcode, CLT and miniconda3.

**Expected behavior**
The extension build to finish without an error.

**Error Messages**
```
$ python3
Python 3.7.7 (default, May  6 2020, 04:59:01)
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/opt/llvm/bin/clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/fabioferreira/miniconda3/envs/gtn/include -arch x86_64 -I/Users/fabioferreira/miniconda3/envs/gtn/include -arch x86_64 -I/usr/local/opt/llvm/include -DONMAC -I/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py -I/Users/fabioferreira/.mujoco/mujoco200/include -I/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/numpy/core/include -I/Users/fabioferreira/miniconda3/envs/gtn/include/python3.7m -c /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/cymj.c -o /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/cymj.o -fopenmp -w
/usr/local/opt/llvm/bin/clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/fabioferreira/miniconda3/envs/gtn/include -arch x86_64 -I/Users/fabioferreira/miniconda3/envs/gtn/include -arch x86_64 -I/usr/local/opt/llvm/include -DONMAC -I/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py -I/Users/fabioferreira/.mujoco/mujoco200/include -I/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/numpy/core/include -I/Users/fabioferreira/miniconda3/envs/gtn/include/python3.7m -c /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.c -o /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.o -fopenmp -w
/usr/local/opt/llvm/bin/clang -bundle -undefined dynamic_lookup -L/Users/fabioferreira/miniconda3/envs/gtn/lib -arch x86_64 -L/Users/fabioferreira/miniconda3/envs/gtn/lib -arch x86_64 -L/usr/local/opt/llvm/lib -I/usr/local/opt/llvm/include -arch x86_64 /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/cymj.o /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.o -L/Users/fabioferreira/.mujoco/mujoco200/bin -L/Users/fabioferreira/.mujoco/mujoco200/bin -lmujoco200 -lglfw.3 -o /Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.10_37_macextensionbuilder/lib.macosx-10.9-x86_64-3.7/mujoco_py/cymj.cpython-37m-darwin.so -fopenmp
ld: unknown option: -platform_version <------------------------------ error
clang-10: error: linker command failed with exit code 1 (use -v to see invocation) <-------------- error
Traceback (most recent call last):
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/unixccompiler.py", line 205, in link
    self.spawn(linker + ld_args)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/opt/llvm/bin/clang' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 338, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/command/build_ext.py", line 559, in build_extension
    target_lang=language)
  File "/Users/fabioferreira/miniconda3/envs/gtn/lib/python3.7/distutils/ccompiler.py", line 717, in link_shared_object
[... 39 lines cut ...]
GNU compiler collection
https://gcc.gnu.org/
/usr/local/Cellar/gcc/10.1.0 (1,463 files, 338.5MB) *
  Poured from bottle on 2020-07-03 at 19:31:35
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/gcc.rb
==> Dependencies
Required: gmp ✔, isl ✔, libmpc ✔, mpfr ✔
==> Options
--HEAD
```
I have the following lines in my ~/.zshrc:

```
 15 # mujoco
 16 export PATH="/usr/local/opt/llvm/bin:$PATH"
 17
 18 export CC="/usr/local/opt/llvm/bin/clang"
 19 export CXX="/usr/local/opt/llvm/bin/clang++"
 20 export CXX11="/usr/local/opt/llvm/bin/clang++"
 21 export CXX14="/usr/local/opt/llvm/bin/clang++"
 22 export CXX17="/usr/local/opt/llvm/bin/clang++"
 23 export CXX1X="/usr/local/opt/llvm/bin/clang++"
 24 export LDFLAGS="-L/usr/local/opt/llvm/lib"
 25 export CPPFLAGS="-I/usr/local/opt/llvm/include"
```

ferreirafabio, 2020-07-04:
Surprisingly, changing to an older Xcode version solved the issue:
`$ sudo xcode-select -switch /Applications/Xcode_11.3.1.app`

closed 2020-07-04 by ferreirafabio

carlee0, 2021-02-18:
Switching to an older Xcode also helped me! Thanks a lot for posting the solution.

One little thing, I had to run `sudo xcodebuild -license` to be able to use the CommandLineTool after switching Xcode.