openai/mujoco-py #555: AttributeError: module 'mujoco_py' has no attribute 'MjRenderContextOffscreen'

HankerSia, 2020-07-18:
**Describe the bug**
i installed keras_rl with tensorflow1.15, python 3.7.7, mujoco_py1.50.0.1, i had tried both mjppro150 and mjppro200 which ./simulate ../model/humanoid.xml can be executed successfully.
but when i test python ddpg_mujoco.py, it shows that problem
**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: [e.g. ubuntu20.13.6] with anaconda3
 - Python Version [e.g. 3.7.7]
 - Mujoco Version [e.g. 1.50, 2.0]
 - mujoco-py version [e.g. 1.50.0.1]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
/usr/local/cuda-10.0/lib64:/home/kai/.mujoco/mjpro150/bin:/home/kai/.mujoco/mjpro200/bin
 - output of: `echo $HOME`
/home/kai
 - output of: `echo $USER`
kai
**Additional context**
ning NUMA node zero
2020-07-18 16:06:29.322003: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1304] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 5148 MB memory) -> physical GPU (device: 0, name: GeForce RTX 2060, pci bus id: 0000:01:00.0, compute capability: 7.5)
WARNING:tensorflow:From /home/kai/anaconda3/envs/tf1_gpu/lib/python3.7/site-packages/keras/backend/tensorflow_backend.py:422: The name tf.global_variables is deprecated. Please use tf.compat.v1.global_variables instead.

Training for 1000000 steps ...
Traceback (most recent call last):
  File "ddpg_mujoco.py", line 69, in <module>
    agent.fit(env, nb_steps=1000000, visualize=False, verbose=1)
  File "/home/kai/anaconda3/envs/tf1_gpu/lib/python3.7/site-packages/rl/core.py", line 132, in fit
    observation = deepcopy(env.reset())
  File "/home/kai/gym/gym/wrappers/monitor.py", line 39, in reset
    self._after_reset(observation)
  File "/home/kai/gym/gym/wrappers/monitor.py", line 185, in _after_reset
    self.reset_video_recorder()
  File "/home/kai/gym/gym/wrappers/monitor.py", line 206, in reset_video_recorder
    self.video_recorder.capture_frame()
  File "/home/kai/gym/gym/wrappers/monitoring/video_recorder.py", line 101, in capture_frame
    frame = self.env.render(mode=render_mode)
  File "/home/kai/gym/gym/core.py", line 233, in render
    return self.env.render(mode, **kwargs)
  File "/home/kai/gym/gym/envs/mujoco/mujoco_env.py", line 145, in render
    self._get_viewer(mode).render(width, height, camera_id=camera_id)
  File "/home/kai/gym/gym/envs/mujoco/mujoco_env.py", line 174, in _get_viewer
    self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, -1)
AttributeError: module 'mujoco_py' has no attribute 'MjRenderContextOffscreen'

thanks for your help!

SparklingWhispering, 2021-07-08:
hello, I have the same problem, and I want to ask if you solved it？

hskAlena, 2021-08-22:
I solved the problem by installing mujoco-py==1.50.1.68 !