openai/mujoco-py #557: Mujoco-py : How to move model by using keyboard?

hoon159487, 2020-07-24:
---using version---
ubuntu 20.04
python 3.6.10
mjpro150
mujoco-py 1.50.1.68
And all of ready for working openAI hide and seek, successfully worked examples.

Our team want How to move model(or any objects) by using keyboard input. (ex. w,a,s,d or direction arrows)
We tried to find out on OpenAI's python codes like mjviewer.py or mujoco-py etc. but couldn't find them.

we want how to use something functions(for move) or where we add that.
please tell us how to do those. thanks.

hietalajulius, 2020-07-25:
@hoon159487 I have a branch that implements a custom callback for key presses in `MjViewer`: https://github.com/openai/mujoco-py/pull/558 You can pull the branch to use it 👍 I needed this myself for capturing expert demo actions during simulation.

hoon159487, 2020-08-04:
@hietalaljulius

Thanks for your comment Julius.
We are trynig to understand and apply your comment and changed codes to us.
but we are still struggling to find out how to move the objects in Mujoco. 
It should be somewhere in the codes, but we're new to Mujoco and doesn't have any idea what should we look for.
If you kindly will, can you teach us what should we look for to move the objects in Mujoco?

Thanks in advance.

hietalajulius, 2020-08-04:
> @hietalajulius
> 
> Thanks for your comment Julius.
> We are trynig to understand and apply your comment and changed codes to us.
> but we are still struggling to find out how to move the objects in Mujoco.
> It should be somewhere in the codes, but we're new to Mujoco and doesn't have any idea what should we look for.
> If you kindly will, can you teach us what should we look for to move the objects in Mujoco?
> 
> Thanks in advance.

Yeah sure!

There are 2 ways that I can think of:
**Using actuators**
-  Defining the actuator in the xml like here: https://github.com/openai/gym/blob/5404b39d06f72012f562ec41f60734bd4b5ceb4b/gym/envs/robotics/assets/fetch/pick_and_place.xml#L31
- Controlling it in the python code like here: https://github.com/openai/gym/blob/5404b39d06f72012f562ec41f60734bd4b5ceb4b/gym/envs/robotics/utils.py#L32 (i.e. setting values for the sim.data.ctrl array that refers to the actuators
**Using mocap bodies**
- Defining mocap bodies like here:  https://github.com/openai/gym/blob/601548dbc503dc6e58d7d4344e7c0ddee6650a94/gym/envs/robotics/assets/fetch/robot.xml#L2
- Defining weld constraints like here: https://github.com/openai/gym/blob/601548dbc503dc6e58d7d4344e7c0ddee6650a94/gym/envs/robotics/assets/fetch/shared.xml#L39
- Controlling the mocap in the python code like here: https://github.com/openai/gym/blob/5404b39d06f72012f562ec41f60734bd4b5ceb4b/gym/envs/robotics/utils.py#L55

The idea of my change was that with the callback function you could get the key strokes made during simulation that could be mapped to custom values that you give to the actuator/mocap body in the python code.

Tuning everything so that it works for your purposes is the tough part in all this and requires reading the mujoco docs etc :D But I think the gym Fetch environments are a good starting point to see how everything works.

hoon159487, 2020-09-02:
@hietalajulius 

Thanks to you, we were able to move objects via key callback. But we got another problem that need your help.
We know this is not right to ask you in this thread, and we're so sorry about that.

Currently we're trying to add the object named "projectile" as a stick that starts from the firing character to the point that intersects with raycast first and get removed in a short time later, and also we're working on removing characters when their HP goes below 0.

On delete part, I've heard that the mujoco in C has the command mjdelete, but we couldn't find anything similar in mujoco_py.
If mujoco_py has those functions, we'd like to know what are they called, and how to use them.

Thanks in advance.

hietalajulius, 2020-09-16:
> @hietalajulius
>
> Thanks to you, we were able to move objects via key callback. But we got another problem that need your help.
> We know this is not right to ask you in this thread, and we're so sorry [...]

Sorry for the late reply! I think you should be able to use all the functions and properties from here also in mujoco-py: http://www.mujoco.org/book/APIreference.html#API But there does not seem to be a function for removing bodies etc. and I'm not sure what properties of a body would need to be modified to make it "disappear".

Amir-Arsalan, 2021-07-08:
@hietalajulius This is a bit irrelevant but is there a way to change position of an object in a scene on the fly via the Python API?

hietalajulius, 2021-07-09:
> @hietalajulius This is a bit irrelevant but is there a way to change position of an object in a scene on the fly via the Python API?

Hello! In your xml you can define the body as a "mocap" ([http://www.mujoco.org/book/XMLreference.html#body](http://www.mujoco.org/book/XMLreference.html#body)) and then move it with `set_mocap_pos` [(https://openai.github.io/mujoco-py/build/html/reference.html?highlight=mocap#set_mocap_pos)](https://openai.github.io/mujoco-py/build/html/reference.html?highlight=mocap#set_mocap_pos)

Amir-Arsalan, 2021-07-09:
@hietalajulius Thank you so much. The reason I need this is because my goal is to make a body like an eye of a robot using a camera and then move the camera (i.e. the body) to capture images from the scene but I'm not sure how I can mount a camera to this moving body via the Python API. Would you be able to give me some pointers for this?

hietalajulius, 2021-07-09:
> @hietalajulius Thank you so much. The reason I need this is because my goal is to make a body like an eye of a robot using a camera and then move the camera (i.e. the body) to capture images from [...]

You should be able to set the camera as a child of the mocap body ([http://www.mujoco.org/book/XMLreference.html#camera](http://www.mujoco.org/book/XMLreference.html#camera)). I think something like this should/could work:

`
<worldbody>
  <body mocap="true" ... >
    <camera ... />
  </body>
</worldbody>
`

Alternatively you could define the camera as a child of the world body and move it in python with `model.cam_pos[camid] = new_position`.

Amir-Arsalan, 2021-07-09:
@hietalajulius Thank you so much! Is there a way to also define the camera in the Python API (i.e. not have a camera in the XML file) and use that camera as "eye" of the robot by mounting it to a body? Right now I am defining my cameras via `viewer = mujoco_py.MjRenderContext()` and then do `viewer.cam.xxx` to change camera settings.

hietalajulius, 2021-07-13:
> @hietalajulius Thank you so much! Is there a way to also define the camera in the Python API (i.e. not have a camera in the XML file) and use that camera as "eye" of the robot by mounting it to a [...]

I believe you need to define the camera in the XML if you want to attach it to a body.