openai/mujoco-py #569: What's the meaning of qpos and xpos?

HeegerGao, 2020-09-05:
Hello, I am new to mujoco, and I want to ask what is the difference between qpos and xpos? Thank you !

codeFairOfficial, 2021-02-05:
@HeegerGao it looks like no one is answering this question :disappointed: . My name is Matt and I work for a compnay named codeFair that is trying to help get issues/questions resolved quickly. One of our developers can get this question answered for you, we just need you to fill out the following form: https://sahchhnmtaj.typeform.com/to/nB94H3p1

yao07, 2021-03-03:
qpos is basically a position of each joint in the simulation and qvel are the velocities of these joints.

yao07, 2021-03-03:
qpos represents a position of each joint in the simulation and qvel denotes velocities of these joints.

nikhilweee, 2022-01-26:
I may be wrong, but as far as I understand, `qpos` is the position of the particular point wrt its parent, whereas `xpos` stores the cartesian coordinates of the point after forward kinematics. [(reference)](https://roboti.us/forum/index.php?threads/cartesian-coordinates-of-body.3376/#post-3654)

Engineering-Geek, 2022-07-17:
> Hello, I am new to mujoco, and I want to ask what is the difference between qpos and xpos? Thank you !

Hey @HeegerGao, there's some documentation on qpos and xpos [here](https://github.com/atabakd/MuJoCo-Tutorials/blob/master/include/mjdata.h). To answer your question, qpos is likely the position of the 'state', and xpos is the cartesian coordinate of the body frame. How this applies in your mode, I can't say. If you do need access to other variables that might better suit what you're trying to extract, take a look at the link above.

HeegerGao, 2022-07-18:
> Hey @HeegerGao, there's some documentation on qpos and xpos [here](https://github.com/atabakd/MuJoCo-Tutorials/blob/master/include/mjdata.h). To answer your question, qpos is likely the position of [...]

Thank you! It helps.

closed 2022-07-18 by HeegerGao