openai/mujoco-py #570: Dynamic constraints

prasoongoyal, 2020-09-07:
I'm trying to simulate a suction-type gripper, i.e., when the robot's gripper is sufficiently close to an object, the object is attached to the gripper. I suppose I can use [weld equality constraint](http://www.mujoco.org/book/XMLreference.html#equality-weld) for this, but I'm not sure how to activate/deactivate it dynamically. Apparently, mujoco [allows](http://www.mujoco.org/forum/index.php?threads/dynamic-constraints.3576/) enabling/disabling constraints dynamically, but I couldn't find how to do this in mujoco_py.
I can see the constraints in `env.sim.model.eq_type`, `env.sim.model.eq_active`, etc., but these fields are non-writable.
Thanks for any help!

prasoongoyal, 2020-09-14:
For anyone who runs into this issue -- I was not setting `env.sim.model.eq_active` correctly -- the entire variable cannot be updated, but the individual array elements are writable.

closed 2020-09-14 by prasoongoyal

v-wewei, 2020-11-02:
hi, @prasoongoyal .  I would like to use a suction gripper for bin-picking and I have some questions for suction gripper:
1) could you help to share with me the suction model and XML file?
2) do the suction work as well in the real world and can transform well into the real scenario?

Thanks, very much!

prasoongoyal, 2020-11-02:
You can add a weld constraint like shown in [this example](https://github.com/mazpie/mime/blob/f8cdfb86d5872723fe9e0ea16469083cce5f5858/metaworld/envs/assets_updated/sawyer_xyz/sawyer_table_with_hole_no_puck.xml#L24). Then you can activate/deactivate the constraint using `env.sim.model.eq_active` to grasp/release object.

However, the weld constraint wasn't working as expected -- see my post on MuJoCo forums [here](http://www.mujoco.org/forum/index.php?threads/weld-equality-not-working-as-expected.4223/).

An alternate solution I'm using is to turn off gravity on the object, and then setting the object position to the robot's gripper position at each timestep.

maheshr24, 2024-12-09:
I created a weld for both grippers to the object, and this happens any better equality constraint
https://github.com/user-attachments/assets/50653905-52f3-4ae9-b57e-e0b3c62daf57

NadeenElh, 2025-08-05:
I wonder if this weld equality constraint was solved in newer mujoco versions. Having the object follow the gripper manually certainly can't be as smooth.