openai/mujoco-py #571: Installing Mujoco 'ld' error

cyclone923, 2020-09-13:
> Agree with @bobyang9 . The problem has not been solved yet. Waiting for a solution.

(o2p2) yangchen@Chengxis-MacBook-Pro .mujoco % 
pip install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached mujoco-py-2.0.2.13.tar.gz (792 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/2f/48/b108057c1a23c8da9f4cdc7a7c46ab7cec49c3563c0706d50f2527de6ba0/mujoco-py-2.0.2.13.tar.gz#sha256=d6ae66276b565af9063597fda70683a89c7356290f5ac3961b794ee90ec50eea.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (2.9.0)
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.12.0)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.29.21)
Requirement already satisfied, skipping upgrade: fasteners~=0.15 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.15)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.16.1)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.14.2)
Requirement already satisfied, skipping upgrade: pillow in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (7.2.0)
Requirement already satisfied, skipping upgrade: six in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.15.0)
Requirement already satisfied, skipping upgrade: monotonic>=0.1 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.5)
Requirement already satisfied, skipping upgrade: pycparser in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.20)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/yangchen/opt/miniconda3/envs/o2p2/bin/python /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/tmp2udd9q81
       cwd: /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py
  Complete output (27 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/cymj_2.0.2.13_37_macextensionbuilder_37.so
  Compiling /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx because it depends on /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-build-env-yc969w4s/overlay/lib/python3.7/site-packages/Cython/Includes/libc/string.pxd.
  [1/1] Cythonizing /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/yangchen/opt/miniconda3/envs/o2p2/include -arch x86_64 -I/Users/yangchen/opt/miniconda3/envs/o2p2/include -arch x86_64 -DONMAC -Imujoco_py -I/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py -I/Users/yangchen/.mujoco/mujoco200/include -I/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-build-env-yc969w4s/overlay/lib/python3.7/site-packages/numpy/core/include -I/Users/yangchen/opt/miniconda3/envs/o2p2/include/python3.7m -c /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/yangchen/opt/miniconda3/envs/o2p2/include -arch x86_64 -I/Users/yangchen/opt/miniconda3/envs/o2p2/include -arch x86_64 -DONMAC -Imujoco_py -I/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py -I/Users/yangchen/.mujoco/mujoco200/include -I/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-build-env-yc969w4s/overlay/lib/python3.7/site-packages/numpy/core/include -I/Users/yangchen/opt/miniconda3/envs/o2p2/include/python3.7m -c /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/gl/dummyshim.c -o /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/ [... cut ...]
 ld: library not found for -lmujoco200
 collect2: error: ld returned 1 exit status
 error: command '/usr/local/bin/gcc-6' failed with exit status 1
 ----------------------------------------
ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

Anybody nows how to solve this 'ld' issue?
Thanks.

_Originally posted by @cyclone923 in https://github.com/openai/mujoco-py/issues/463#issuecomment-691632569_

yunzhe-tao, 2020-11-18:
Any update on this?

csingh27sewts, 2021-01-23:
I still have the same problem

stwykd, 2021-07-31:
I'm having this issue as well

stwykd, 2021-07-31:
I'm having this issue as well

```
(robotics) stwykd@Pierluigis-MacBook-Pro:~$ pip uninstall mujoco-py
Found existing installation: mujoco-py 2.0.2.8
Uninstalling mujoco-py-2.0.2.8:
  Would remove:
    /Users/stwykd/anaconda3/envs/robotics/lib/python3.9/site-packages/mujoco_py-2.0.2.8.dist-info/*
    /Users/stwykd/anaconda3/envs/robotics/lib/python3.9/site-packages/mujoco_py/*
  Would not remove (might be manually added):
    /Users/stwykd/anaconda3/envs/robotics/lib/python3.9/site-packages/mujoco_py/generated/mujocopy-buildlock
Proceed (y/n)? y
  Successfully uninstalled mujoco-py-2.0.2.8
(robotics) stwykd@Pierluigis-MacBook-Pro:~$ pip3 install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached mujoco-py-2.0.2.13.tar.gz (792 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/2f/48/b108057c1a23c8da9f4cdc7a7c46ab7cec49c3563c0706d50f2527de6ba0/mujoco-py-2.0.2.13.tar.gz#sha256=d6ae66276b565af9063597fda70683a89c7356290f5ac3961b794ee90ec50eea.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied: cffi>=1.10 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (1.14.6)
Requirement already satisfied: fasteners~=0.15 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (0.16.3)
Requirement already satisfied: glfw>=1.4.0 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (2.1.0)
Requirement already satisfied: numpy>=1.11 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (1.21.1)
Requirement already satisfied: Cython>=0.27.2 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (0.29.24)
Requirement already satisfied: imageio>=2.1.2 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (2.9.0)
Requirement already satisfied: pycparser in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.20)
Requirement already satisfied: six in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.16.0)
Requirement already satisfied: pillow in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (8.3.1)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/stwykd/anaconda3/envs/robotics/bin/python /Users/stwykd/anaconda3/envs/robotics/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/tmprr6dwnjw
       cwd: /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2
  Complete output (27 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/cymj_2.0.2.13_39_macextensionbuilder_39.so
  Compiling /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/cymj.pyx because it depends on /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-build-env-1_nshu86/overlay/lib/python3.9/site-packages/Cython/Includes/libc/string.pxd.
  [1/1] Cythonizing /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders/ww
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macos [... cut ...] 27ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/lib.macosx-10.9-x86_64-3.9/mujoco_py/cymj.cpython-39-darwin.so -fopenmp
  ld: library not found for -lmujoco200
  collect2: error: ld returned 1 exit status
  error: command '/usr/local/bin/gcc-8' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

```

stwykd, 2021-07-31:
I'm having this issue as well

```
(robotics) stwykd@Pierluigis-MacBook-Pro:~$ pip3 install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached mujoco-py-2.0.2.13.tar.gz (792 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/2f/48/b108057c1a23c8da9f4cdc7a7c46ab7cec49c3563c0706d50f2527de6ba0/mujoco-py-2.0.2.13.tar.gz#sha256=d6ae66276b565af9063597fda70683a89c7356290f5ac3961b794ee90ec50eea.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied: cffi>=1.10 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (1.14.6)
Requirement already satisfied: fasteners~=0.15 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (0.16.3)
Requirement already satisfied: glfw>=1.4.0 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (2.1.0)
Requirement already satisfied: numpy>=1.11 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (1.21.1)
Requirement already satisfied: Cython>=0.27.2 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (0.29.24)
Requirement already satisfied: imageio>=2.1.2 in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from mujoco-py<2.1,>=2.0) (2.9.0)
Requirement already satisfied: pycparser in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.20)
Requirement already satisfied: six in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.16.0)
Requirement already satisfied: pillow in ./anaconda3/envs/robotics/lib/python3.9/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (8.3.1)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/stwykd/anaconda3/envs/robotics/bin/python /Users/stwykd/anaconda3/envs/robotics/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/tmprr6dwnjw
       cwd: /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2
  Complete output (27 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/cymj_2.0.2.13_39_macextensionbuilder_39.so
  Compiling /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/cymj.pyx because it depends on /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-build-env-1_nshu86/overlay/lib/python3.9/site-packages/Cython/Includes/libc/string.pxd.
  [1/1] Cythonizing /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders/ww
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_64-3.9/private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T
  creating /private/var/folders/ww/fvjyk5gd00qgn1bd7qtsvtjm0000gn/T/pip-install-27k9382a/mujoco-py_8cda86751f1b40e49d2ec30cc6627ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/temp.macosx-10.9-x86_6 [... cut ...] 27ff2/mujoco_py/generated/_pyxbld_2.0.2.13_39_macextensionbuilder/lib.macosx-10.9-x86_64-3.9/mujoco_py/cymj.cpython-39-darwin.so -fopenmp
  ld: library not found for -lmujoco200
  collect2: error: ld returned 1 exit status
  error: command '/usr/local/bin/gcc-8' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

```

stwykd, 2021-07-31:
my issue was that whenever I run mujoco-py, MacOS complained the software is from an unidentified developer and asked me if I wanted to move a library to the bin. I didn't quite read the message and clicked "moved to bin".
I've downloaded the mujoco binaries again and placed a fresh copy under .mujoco. This time I've had to allow all the warnings that MacOS threw at me after running some command within bin (e.g. /simulate ../model/humanoid.xml). You allow by going into Settings > Security & Privacy  and click the "Allow" button you'll see there