openai/mujoco-py #572: INSTALLATION PROCEDURE THAT WORKS (update the README, please)

MartinaRuocco, 2020-09-14:
Don't know how many of you fellas are trying to install mujoco_py without succeeding. 
In case you are having troubles, I followed the instructions [here](https://github.com/reinforcement-learning-kr/pg_travel/wiki/Installing-Mujoco-py-on-Linux) and it worked (I work on Ubuntu 16.04)

1st Step: Download
you need to extract your mujoco zip file in a new hidden folder (.mujoco), along with the license file. 
The structure will be:

```
~/.mujoco/
├── mjkey.txt
└── mujoco200
    ├── bin
    ├── doc
    ├── include
    ├── model
    └── sample
```
2d Step: setup the environmental variable
`gedit ~/.bashrc`
and add the line
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/.mujoco/mujoco200/bin`

3rd Step: (optional) create a python3 virtual environment 
```
virtualenv ve_mujoco -p python3.6
source ve_mujoco/bin/activate
```
this step can also be done with conda

4th Step: Installation
```
git clone https://github.com/openai/mujoco-py.git
cd mujoco_py
python3 setup.py install
```

to test, just try to import mujoco_py in python.

NB: you might be asked to install some packages, in my case I had to install the following:
```
sudo apt-get install libglfw3-dev libosmesa6-dev libgl1-mesa-glx libglfw3
pip install patchelf-wrapper
```

Tylersuard, 2020-09-17:
You have the dubious honor of being my favorite internet person for the day.  Thank you!

PeterMitrano, 2020-10-20:
Alternatively, `sudo apt install patchelf` also works

quinnliu, 2021-01-10:
Anyone get this working on MacOS 10.15.7?