openai/mujoco-py #576: render window does not close

joeljosephjin, 2020-09-24:
**Describe the bug**
This is the code I am running.

```
#Importing OpenAI gym package and MuJoCo engine
import gym
import mujoco_py

#Setting MountainCar-v0 as the environment
env = gym.make('HalfCheetah-v2')
# env = gym.make('MountainCar-v0')

#Sets an initial state
env.reset()

# Rendering our instance 300 times
for _ in range(300):
  #renders the environment
  env.render()

  #Takes a random action from its action space 
  # aka the number of unique actions an agent can perform
  env.step(env.action_space.sample())
env.close()
```
It runs fine, but then the render window doesn't close. Even when I do env.close() again, it doesn't close.
I am runnning the code on jupyter notebook, in ubuntu.
When I try to manually click close, the window becomes non-responsive and the dialog box pops up where I click 'Force Quit'. The window closes but then the jupyter kernel restarts.

nerkulec, 2020-12-30:
I also experience the same issue on Jupyter

kaigett, 2021-04-27:
I have the same issue, I fixed it by:

```
import glfw
glfw.terminate()
```

raeidsaqur, 2021-05-06:
@kaigett it works - you'll need to restart the kernel though. :)

suvajit790, 2021-12-13:
```
env = gym.make("CarRacing-v0")
env.reset()
for i in range(1000):
    env.render()
    env.step(env.action_space.sample())
env.reset()
env.close()
```

This worked for me.

UseryzhfromCN, 2022-08-29:
> 

I think CarRacing is in Box2d environment but not MUJOCO. MUJOCO requires glfw.terminate() for closing the window.