openai/mujoco-py #578: ERROR: could not initialize GLFW

HanBing0802, 2020-09-27:
**Describe the bug**
# ./simulate ../model/humanoid.xml
MuJoCo Pro version 2.00
ERROR: could not initialize GLFW

Press Enter to exit ...

**To Reproduce**
This bug occurs when testing for successful installation of mujoco200.

**Expected behavior**
How to solve this bug?

**Desktop (please complete the following information):**
 - CentOS: [CentOS 7.7.1908]
 - Python Version [3.8.5]
 - Mujoco Version [2.00]
 - mujoco-py version [2.0.2.8]

Crazyalltnt, 2020-10-03:
I also have the same problem on Ubuntu18.04

ymr12, 2020-10-20:
> I also have the same problem on Ubuntu18.04
Me too. Have you fix it?

zhihaocheng, 2020-12-24:
Same, have anyone solved this problme?

zanellar, 2021-04-22:
same error on Ubuntu 20.04.2 LTS

thj926, 2021-08-19:
same error on Ubuntu 18.04

TUY3, 2021-11-03:
maybe，no header？

MR-BENjie, 2021-11-29:
maybe you use the  remote desktop to access you system ,and the problem happened when you don't have a display

Grantsydney2020, 2022-02-26:
same question, no answer,help!

XiangyuSu611, 2022-07-09:
maybe using virtualGL?

JacopoDapueto, 2025-03-26:
Did someone find a solution?
I'm experiencing the same issue while not having a display, and I can't do it any other way.