openai/mujoco-py #582: Installing error: sre_constants.error: bad character range r-n at position 7

rgap, 2020-10-20:
**Describe the bug**

Does anyone else get this when installing / when importing mujoco_py?
 
File "/usr/local/lib/python3.6/sre_parse.py", line 553, in _parse
    raise source.error(msg, len(this) + 1 + len(that))
sre_constants.error: bad character range r-n at position 7

**To Reproduce**
Install mujoco with the new readme steps.

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - Python Version Python 3.6.3
 - Mujoco Version 200
 - mujoco-py version Installed from source

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: :/home/rel/.mujoco/mujoco200/bin:/usr/lib/nvidia-000:/home/[usr-name]/.mujoco/mujoco200/bin:/usr/lib/nvidia-[driver-ver
 - output of: `echo $HOME`: /home/rel

closed 2020-10-21 by rgap

Lynn-lol, 2021-09-18:
Sorry to bother you,bro. My problem is absolutely the same as yours. Could you please add some details about the error in LD_LIBRARY_PATH? Your kind help means a lot to me.

rgap, 2021-09-18:
Hi,

I don't remember what it was. I think it was because of wrong quotation marks or that I forgot to change the <user> in the path. I followed this
 https://github.com/openai/mujoco-py/issues/572