openai/mujoco-py #588: Not able to visualize contact forces

hermanjakobsen, 2020-11-05:
**Describe the bug**
Pressing "C" on the keyboard has no effect on visualization of the contact forces. Whenever I press the key, the menu says that the contact forces are toggled on but the forces are not visualized. 

**To Reproduce**
Press "C" while simulating an arbitrary environment.

**Expected behavior**
Expected to see visualization of vectors representing contact forces.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Python Version 3.8.5
 - Mujoco Version 2.0
 - mujoco-py version 2.0.2.9

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: /usr/lib/cuda/include:/usr/lib/cuda/lib64:/home/hermankj/webots/lib:/home/hermankj/.mujoco/mujoco200/bin:/home/hermankj/.mujoco/mjpro150/bin

 - output of: `echo $HOME`: 
/home/hermankj

 - output of: `echo $USER`:
hermankj

krzentner, 2020-12-01:
It might be helpful in diagnosing this that openai/gym disabled using mujoco 2.0 because [contact forces were always zero.](https://github.com/openai/gym/issues/1541) It looks like that issue was never fixed in this repo, [although this fork claims to fix it.](https://github.com/openai/mujoco-py/commit/e869e76e668efdf2857eb978284283087f33513c)

TheButlah, 2020-12-14:
Is a merge request going to be submitted for that fork? Its only a 1 line change. Any reason it couldn't be merged?

rohanpsingh, 2021-04-12:
Was anybody able to solve this issue? I am unable to visualize the contact forces even after applying the patch mentioned by @krzentner.
Is moving to mujoco 1.5 the only solution?

rohanpsingh, 2021-04-13:
Update: PR #487 is able to visualize contact forces!