openai/mujoco-py #591: mujoco_py.cymj.GlfwError: Failed to create GLFW window

ashBabu, 2020-11-19:
Hi, 
I get the error ```mujoco_py.cymj.GlfwError: Failed to create GLFW window. 
GLFW error (code %d): %s 65543 b'GLX: Failed to create context: BadValue (integer parameter out of range for operation) ```
I am on Ubuntu 18.04. 
Any help would be really appreciated

MianchuWang, 2020-11-30:
The same issue :(

a-z-e-r-i-l-a, 2021-01-11:
same issue...

a-z-e-r-i-l-a, 2021-01-11:
@MianchuWang  @ashBabu  I have the same issue, have you found a workaround for it?

MianchuWang, 2021-01-11:
> @MianchuWang @ashBabu I have the same issue, have you found a workaround for it?

Hi, fortunately I solved the problem at last. The problem was raising when I was using a headless Linux server. The problem is that `glfw` is unavailable without a screen. However, we can use a virtual screen  `xvfb`, like `xvfb-run python foo.py`. You can read http://elementalselenium.com/tips/38-headless for more details.

I don't know whether my situation is similar to yours, but I hope you can solve it soon. : )

rojas70, 2021-04-21:
@MianchuWang 
I am running on a standard workstation but get the same problem. 

I tried to install `xvfb` and run as you did, when I do that there is no error, but now window shows up... It might work for a headless server but for a standard workstation it seems a different situation. 

The strange thing is that this was working just recently. I don't believe I had changed anything. 

Any feedback is appreciated.

MissuQAQ, 2021-07-06:
> Hi, fortunately I solved the problem at last. The problem was raising when I was using a headless Linux server. The problem is that `glfw` is unavailable without a screen. However, we can use a [...]
Hi Mianchu 
Thanks a lot. Your solution helps me and I wanna ask could we set up a GUI for mujoco in a headless Linux server? 
I set up the environment in a docker container. 

Thanks a lot.

MissuQAQ, 2021-07-06:
> Hi, fortunately I solved the problem at last. The problem was raising when I was using a headless Linux server. The problem is that `glfw` is unavailable without a screen. However, we can use a [...]

Hi Mianchu, Thanks a lot and your solution helps  me. I wanna ask could we set up a GUI for a headless Linux server? I set up the environments in a docker container.

MianchuWang, 2021-07-07:
> Hi Mianchu, Thanks a lot and your solution helps me. I wanna ask could we set up a GUI for a headless Linux server? I set up the environments in a docker container.

Hi. There are some ways to set up a GUI for a headless Linux server, e.g., TigerVNC. But I’m not very clear about the setup procedure. Sorry.

dogged1021, 2021-07-25:
I met the same issue and I rebooted the computer system, then the code worked without the problem. I have no idea what happend...

longfeizhang617, 2022-05-28:
> Hi Mianchu, Thanks a lot and your solution helps me. I wanna ask could we set up a GUI for a headless Linux server? I set up the environments in a docker container.

I have  fixed the same issue by your method , thank you very much.

wkkdhj, 2024-07-21:
Thank you very much, it's really helpful !!!

meghbhalerao, 2024-11-18:
I am still facing the same issue even with xvfb-run python foo.py - any one facing similar issue? thanks!