openai/mujoco-py #597: ERROR: Command errored out with exit status 1

ylan01, 2020-12-30:
` ERROR: Command errored out with exit status 1:
     command: 'C:\ProgramData\Anaconda3\envs\py37\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\bit\\AppData\\Local\\Temp\\pip-install-ok8f250m\\mujoco-py_054a26951f0e464184b61e328536bbfe\\setup.py'"'"'; __file__='"'"'C:\\Users\\bit\\AppData\\Local\\Temp\\pip-install-ok8f250m\\mujoco-py_054a26951f0e464184b61e328536bbfe\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\bit\AppData\Local\Temp\pip-record-p140o3u4\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\ProgramData\Anaconda3\envs\py37\Include\mujoco-py'
         cwd: C:\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe\
    Complete output (7 lines):
    running install
    running build
    Removing old mujoco_py cext C:\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe\mujoco_py\generated\cymj_1.50.1.68_37_windowsextensionbuilder_37.so
    running build_ext
    building 'mujoco_py.cymj' extension
    creating C:\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe
    error: could not create 'C:\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\bit\AppData\Local\Temp\pip-install-ok8f250m\mujoco-py_054a26951f0e464184b61e328536bbfe': 系统找不到指定的路径。
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\ProgramData\Anaconda3\envs\py37\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\bit\\AppData\\Local\\Temp\\pip-install-ok8f250m\\mujoco-py_054a26951f0e464184b61e328536bbfe\\setup.py'"'"'; __file__='"'"'C:\\Users\\bit\\AppData\\Local\\Temp\\pip-install-ok8f250m\\mujoco-py_054a26951f0e464184b61e328536bbfe\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\bit\AppData\Local\Temp\pip-record-p140o3u4\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\ProgramData\Anaconda3\envs\py37\Include\mujoco-py' Check the logs for full command output.`

quinnliu, 2021-01-10:
I've written the following [guide](https://github.com/ARISE-Initiative/robosuite/issues/160) summarizing everything I've tried to install on MacOS 10.15.7 but still not successful

xiaowkong, 2021-05-23:
> ` ERROR: Command errored out with exit status 1: command: 'C:\ProgramData\Anaconda3\envs\py37\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = [...]

Have you ever solved this problem? I met the same problem while pip the mujoco 1.15.

BlueShit-1120, 2021-05-27:
> @BlueShit-1120 Did you manage to solve it? I am running into the same problem now

After restarting the computer, this problem disappears. Sounds very funny. Maybe you can try.

Niels-ns, 2021-05-27:
haha that does sound funny. I found a different solution. Hello future people readying this trying to install mujoco-py on windows 10, python 3. You need to:

follow this guide up to and including step 3.1.4.
Then you need to install pywin32 (pip install pywin32)
Then you need to go here: https://github.com/openai/mujoco-py/issues/68#issuecomment-451562560. and apply the linked code as described. Don't forget to change the syntax to python 3 syntax, so change error line to ```raise WindowsError(f'The system cannot find the path specified: {PartPath}'```
Then you need to go here: https://github.com/openai/mujoco-py/compare/master...aaronsnoswell:fix-windows-support
and apply these changes and follow these steps to make it all work. 

With this I now have mujoco-py working

BlueShit-1120, 2021-05-27:
> haha that does sound funny. I found a different solution. Hello future people readying this trying to install mujoco-py on windows 10, python 3. You need to:
> 
> follow this guide up to and including step 3.1.4: https://www.programmersought.com/article/38134632752/#_12
> Then you need to install pywin32 (pip install pywin32)
> Then you need to go here: [#68 (comment)](https://github.com/openai/mujoco-py/issues/68#issuecomment-451562560). and apply the linked code as described. Don't forget to change the syntax to python 3 syntax, so change error line to `raise WindowsError(f'The system cannot find the path specified: {PartPath}'`
> Then you need to go here: [master...aaronsnoswell:fix-windows-support](https://github.com/openai/mujoco-py/compare/master...aaronsnoswell:fix-windows-support)
> and apply these changes and follow these steps to make it all work.
> 
> With this I now have mujoco-py working

Thanks for your solution and sharing.