openai/mujoco-py #598: Error: Could not initialize EGL. and RuntimeError: Failed to initialize OpenGL

lknownothing, 2021-01-02:
Environment: 
+ ubuntu 18.04 mujoco_py==2.0.2.5  

```
Found 3 GPUs for rendering. Using device 1.
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
Could not initialize EGL
Traceback (most recent call last):
  File "meld/scripts/run_meld.py", line 1023, in <module>
    app.run(main)
......
File "/home/CDJ/testRL/meld/meld/environments/gym_wrappers.py", line 143, in _reset
    return self._modify_observation(observation)
  File "/home/CDJ/testRL/meld/meld/environments/gym_wrappers.py", line 125, in _modify_observation
    image = self._env.sim.render(**self._render_kwargs)[::-1, :, :]
  File "mujoco_py/mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjrendercontext.pyx", line 45, in mujoco_py.cymj.MjRenderContext.__init__
  File "mujoco_py/mjrendercontext.pyx", line 113, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "mujoco_py/opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```
[image]
I read the code from `mujoco_py/mjrendercontext.pyx`
- Even I `export  GPUS=1` and `CUDA_VISIABLE_DEVICE=0` , still Using device 1. Since using 'env.sim.render("mode=rgb_array")`, I also set `export LD_PRELOAD=''`
- I've been suffering from this problem for days. Does anyone know how to modify or set it.
- I wonder where the info `Using device 1` come from.

dbtu, 2021-04-20:
I have the same issue. Did you manage to solve it?

LeoZDong, 2021-06-29:
Same issue here...

QUIlToT, 2022-05-12:
Same issue....

fuzhoulwl, 2024-05-13:
> 同样的问题......

你解决了吗？

dannyXSC, 2025-01-10:
you shold first resolve libegl warning