openai/mujoco-py #601: Error when installing Mujoco-py - RuntimeError: Could not find GCC executable.

GomezH30, 2021-01-11:
When I use pip3 install -U 'mujoco-py<2.1,>=2.0' to install mujoco-py I get this error message:
RuntimeError: Could not find GCC executable.
  
  HINT: On OS X, install GCC with `brew install gcc`. or `port install gcc`.
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

I have tried using both `brew install gcc`. or `port install gcc` and I get command not found: brew or port

 - OS: [macOS 11.0.1]
 - Python Version [3.8]
 - Mujoco Version [200]
 - mujoco-py version [2.0.2.5]

ekanshgupta92, 2021-01-18:
Same problem, but this `pip install mujoco-py==2.0.2.8` worked. However, when I import the library, it throws the same error. Don't know how to fix that.

ekanshgupta92, 2021-01-18:
`brew install gcc@7`
`pip uninstall mujoco_py`
`pip install mujoco_py`
 
Worked for me. Don't know if this is correct, but this gave a few warnings and worked as expected. Maybe this occurred due to some bug in gcc and gets fixed later.

gautams3, 2021-05-26:
The solution above also works for [gcc 8](https://github.com/openai/mujoco-py/issues/254#issuecomment-666417052)

PTJohn0122, 2024-05-01:
Running on same issue with mujoco version 200. My gcc is 13.2.0, and the above solution doesn't work for me.

chrisantuseze, 2024-05-30:
> Running on same issue with mujoco version 200. My gcc is 13.2.0, and the above solution doesn't work for me.

Same here. The solution fix doesn't work for me. gcc version 13.2.0. I tried using the Mujoco from https://github.com/google-deepmind/mujoco/releases and the python binding specified in the repo (pip install mujoco), even though that works, it doesn't support gymnasium. Please whoever is familiar with a fix should help me.