openai/mujoco-py #607: numpy.ndarray size changed

bebbo203, 2021-02-22:
**Describe the bug**
As for the title, I got this error message:
`ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject`

It's solvable by updating numpy from 1.19.2 to 1.20.0 but for other dependecies (tensorflow), I can't do this

**To Reproduce**
Just run Mujoco with numpy 1.19.2 installed

**Error Messages**
`Traceback (most recent call last):
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bebbo203/Scrivania/Tesi/rnd_algo/RND/vec_env.py", line 210, in worker
    env = env_fn_wrapper.x()
  File "/home/bebbo203/Scrivania/Tesi/rnd_algo/RND/cmd_util.py", line 26, in _thunk
    env = gym.make(env_id)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/registration.py", line 156, in make
    return registry.make(id, **kwargs)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/registration.py", line 101, in make
    env = spec.make(**kwargs)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/registration.py", line 72, in make
    cls = load(self.entry_point)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/registration.py", line 17, in load
    mod = importlib.import_module(mod_name)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/mujoco/__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/gym/envs/mujoco/mujoco_env.py", line 12, in <module>
    import mujoco_py
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/mujoco_py/builder.py", line 101, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/bebbo203/anaconda3/envs/rl2/lib/python3.7/site-packages/mujoco_py/builder.py", line 125, in load_dynamic_ext
    return loader.load_module()
  File "cymj.pyx", line 1, in init mujoco_py.cymj
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject`

**Desktop (please complete the following information):**
 - OS: Linux Mint 20.1
 - Python Version 3.7.9
 - mujoco-py version 2.0.2.9

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` :/usr/lib/nvidia-460:/home/bebbo203/.mujoco/mujoco200/bin
 - output of: `echo $HOME` /home/bebbo203
 - output of: `echo $USER` bebbo203

Jiayuan-Gu, 2021-02-25:
I encountered the same issue. However, I thought the error message was quite clear, and led to [this answer](https://stackoverflow.com/questions/66060487/valueerror-numpy-ndarray-size-changed-may-indicate-binary-incompatibility-exp). Installing the latest numpy works, but I wonder what has changed in fact. I find that only numpy 1.20 and <=1.17 can work.

bebbo203, 2021-02-25:
Thanks, the solution to make it works for numpy 1.19 was to use mujoco_py==2.0.2.5 or eralier

samarth-robo, 2021-04-20:
If you need latest mujoco_py (e.g. 2.0.2.9 because of robotsuite.ai), with numpy 1.19.2 (because tensorflow does not accept numpy 1.20), you can use the following command to build the mujoco-py wheel locally after installing numpy

`pip install mujoco-py==2.0.2.9 --no-cache-dir --no-binary :all: --no-build-isolation`

with help from https://github.com/scikit-learn-contrib/hdbscan/issues/457#issuecomment-774033215