openai/mujoco-py #609: I can't import mujoco_py because of gcc error MacOSX BigSur

vittorione94, 2021-02-24:
It seems I can't import mujoco_py

python --version
Python 3.6.12 :: Anaconda, Inc.

```
Package                Version       Location
---------------------- ------------- ------------------------------------------------------------
absl-py                0.11.0
appdirs                1.4.4
appnope                0.1.2
astunparse             1.6.3
attrs                  20.3.0
audioread              2.1.9
backcall               0.2.0
cached-property        1.5.2
cachetools             4.2.1
certifi                2020.12.5
cffi                   1.14.4
chardet                4.0.0
cloudpickle            1.6.0
cycler                 0.10.0
Cython                 0.29.21
decorator              4.4.2
dm-control             0.0.322773188
dm-env                 1.3
dm-tree                0.1.5
docopt                 0.6.2
fasteners              0.16
flake8                 3.8.4
flatbuffers            1.12
future                 0.18.2
gast                   0.3.3
glfw                   2.0.0
google-auth            1.25.0
google-auth-oauthlib   0.4.2
google-pasta           0.2.0
grpcio                 1.32.0
gym                    0.18.0
h5py                   2.10.0
idna                   2.10
imageio                2.9.0
imageio-ffmpeg         0.4.3
importlib-metadata     3.4.0
iniconfig              1.1.1
ipopt                  0.3.0
ipython                7.16.1
ipython-genutils       0.2.0
jedi                   0.18.0
joblib                 1.0.0
Keras                  2.4.3
Keras-Preprocessing    1.1.2
kiwisolver             1.3.1
labmaze                1.0.3
librosa                0.8.0
llvmlite               0.35.0
lockfile               0.12.2
lxml                   4.6.2
Markdown               3.3.3
matplotlib             3.3.3
mccabe                 0.6.1
MeshLabXML             2018.3
moviepy                1.0.3
mujoco-py              2.0.2.13      /Users/vittorio/Documents/mujoco-py
npy-append-array       0.9.6
numba                  0.52.0
numpy                  1.19.5
oauthlib               3.1.0
opt-einsum             3.3.0
packaging              20.8
[... 245 lines cut ...]
  File "/Users/vittorio/anaconda3/envs/py36/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/Users/vittorio/anaconda3/envs/py36/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Users/vittorio/anaconda3/envs/py36/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-8' failed with exit status 1
```

Running on MacOS BigSur 11.1

PS: I also tried everything listed here:  [issue #463](https://github.com/openai/mujoco-py/issues/463 ) but without success

I've tried also all sort of gcc versions 6, 7, 8, 9 

Any help would be much appreciated!

echo $HOME
/Users/vittorio

echo $USER
vittorio

echo $LD_LIBRARY_PATH
(blank)

mfogelson, 2021-02-25:
Try this it might work for you as well and a little less hacky: 
https://github.com/openai/mujoco-py/issues/463#issuecomment-549874271

vittorione94, 2021-02-26:
Unfortunately I already tried that and I get the exact same error. :(

parze, 2021-03-13:
I have the same problem. Tried all gcc's with the same result.

liujiang833, 2021-08-06:
Finally got it work. Maybe? But at least I am able to run spinningup's test case.  
1. Get the latest sdk, not previous versions in [catalina issue](https://github.com/openai/mujoco-py/issues/463). Older versions fail with different reasons.  However, I don't know the proper way to do this.  xcode-select thing didn't work on my laptop. After doing random changes, here is how my sdk directory looks like.  
 ```
lrwxr-xr-x   1 root  wheel   14 Aug  6 11:38 MacOSX.sdk -> MacOSX11.3.sdk
drwxr-xr-x   8 root  wheel  256 Aug  6 11:16 MacOSX10.14.sdk
drwxr-xr-x  10 root  wheel  320 Aug  6 11:38 MacOSX10.15.sdk
drwxr-xr-x   7 root  wheel  224 Nov 30  2020 MacOSX11.1.sdk
drwxr-xr-x   7 root  wheel  224 Aug  6 11:37 MacOSX11.3.sdk
lrwxr-xr-x   1 root  wheel   14 Aug  6 11:36 MacOSX11.sdk -> MacOSX11.3.sdk
```
2. I used gcc installed by ```brew install gcc@7```. 
3. There will be some privacy issue to deal with.  

If you have further questions, I can try to help.

onnoeberhard, 2021-10-11:
I get a very similar error when trying to import `mujoco_py`:

```
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -DONMAC -Imujoco-py/mujoco_py -I/Users/onno/MPI/mbrl/mujoco-py/mujoco_py -I/Users/onno/.mujoco/mujoco200/include -I/Users/onno/.local/share/virtualenvs/mbrl-E31sOzwR/lib/python3.8/site-packages/numpy/core/include -I/Users/onno/.local/share/virtualenvs/mbrl-E31sOzwR/include -I/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c /Users/onno/MPI/mbrl/mujoco-py/mujoco_py/cymj.c -o /Users/onno/MPI/mbrl/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_38_macextensionbuilder/temp.macosx-10.14-x86_64-3.8/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc-8: error: this compiler does not support arm64
gcc-8: error: unrecognized command line option '-iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders'
gcc-8: error: unrecognized command line option '-iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers'
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/unixccompiler.py", line 118, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] +
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/spawn.py", line 157, in _spawn_posix
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-8' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 335, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/onno/.local/share/virtualenvs/mbrl-E31sOzwR/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 192, in run
    _build_ext.build_ext.run(self)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Users/onno/MPI/mbrl/mujoco-py/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/unixccompiler.py", line 121, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-8' failed with exit status 1
```

I don't know if it's the same problem as above, but it's also a gcc error.. I am not sure why it says "error: this compiler does not support arm64", as I have an Intel CPU. Maybe I am misinterpreting that error.

rw422scarlet, 2023-01-15:
> Finally got it work. Maybe? But at least I am able to run spinningup's test case.
>
> 1. Get the latest sdk, not previous versions in [catalina issue](https://github.com/openai/mujoco-py/issues/463). [...]

This worked for me on Mac Monterey. I deleted all previous ``CommandLineTools`` folders and installed them from scratch with ``xcode-select --install``. Deleted ``~/.mujoco210`` and did ``pip uninstall mujoco_py`` and then installed them again. I used ``gcc@7`` and indeed had to give some privacy permissions in mac setting which says "libmujoco210.dylib was installed from browser". The permission thing was in the middle of running ``import mujoco_py`` for the first time. Then I was able to run the test code in README.