openai/mujoco-py #611: distutils.errors.LinkError: command 'gcc' failed with exit status 1

seolhokim, 2021-03-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior.
AWS Ubuntu Server 18.04 LTS (HVM), SSD Volume Type
g4dn.xlarge
sudo apt-get unzip and followed [this](https://www.programmersought.com/article/70704324024/) way.
**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: [e.g. ubuntu 18.04 LTS 10.13.6]
 - Python Version [e.g. 3.6.10]

**Environment**

**Additional context**
Add any other context about the problem here.

Any kinds of already suggested ways did not work for me in this repository issues like `apt install libgl1-mesa-glx --reinstall`
How can i fix this?

Schortenger, 2021-04-23:
sudo apt-get install libglew-dbg

wardnath, 2021-07-27:
try installing gcc if you don't have it `sudo apt-get install gcc`