openai/mujoco-py #615: Running setup.py install for mujoco-py ... error

jgp-github, 2021-04-01:
**Describe the bug**
ERROR: Command errored out with exit status 1:
     command: 'D:\anaconda3\envs\py36\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jgp\\AppData\\Local\\Temp\\pip-install-3hovz2ry\\mujoco-py_77abc2bca99347949f8ff21e735ca545\\setup.py'"'"'; __file__='"'"'C:\\Users\\jgp\\AppData\\Local\\Temp\\pip-install-3hovz2ry\\mujoco-py_77abc2bca99347949f8ff21e735ca545\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\jgp\AppData\Local\Temp\pip-record-d71pqdhc\install-record.txt' --single-version-externally-managed --compile --install-headers 'D:\anaconda3\envs\py36\Include\mujoco-py'
         cwd: C:\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545\
    Complete output (7 lines):
    running install
    running build
    Removing old mujoco_py cext C:\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545\mujoco_py\generated\cymj_1.50.1.68_36_windowsextensionbuilder_36.so
    running build_ext
    building 'mujoco_py.cymj' extension
    creating C:\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545
    error: could not create 'C:\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\Users\jgp\AppData\Local\Temp\pip-install-3hovz2ry\mujoco-py_77abc2bca99347949f8ff21e735ca545': 系统找不到指定的路径。
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'D:\anaconda3\envs\py36\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jgp\\AppData\\Local\\Temp\\pip-install-3hovz2ry\\mujoco-py_77abc2bca99347949f8ff21e735ca545\\setup.py'"'"'; __file__='"'"'C:\\Users\\jgp\\AppData\\Local\\Temp\\pip-install-3hovz2ry\\mujoco-py_77abc2bca99347949f8ff21e735ca545\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\jgp\AppData\Local\Temp\pip-record-d71pqdhc\install-record.txt' --single-version-externally-managed --compile --install-headers 'D:\anaconda3\envs\py36\Include\mujoco-py' Check the logs for full command output.

Niels-ns, 2021-05-26:
I am also getting a similar error where is it also gives a: "error: could not create 'C:\Users\user\AppData\Local\Temp\pip-install-3..." error when trying to install version 1.50.1.68 on windows. 

Did you find a solution?

GehenHe, 2022-02-25:
I found a solution. Do not install using `pip install -r requirements.txt`, use `python setup.py install`

This works for me. 

More details can be found in https://github.com/openai/mujoco-py/issues/253#issuecomment-446025520

jgp-github, 2022-04-27:
ok, I will try it, thanks