openai/mujoco-py #617: Installing mujoco_py

elenasotoh, 2021-04-16:
Whenever I go into Python3, and do "import mujoco_py" I get this:
Traceback (most recent call last):
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/unixccompiler.py", line 117, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] +
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/spawn.py", line 157, in _spawn_posix
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-6' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 307, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 225, in _build_impl
    dist.run_commands()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/builder.py", line 125, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Users/elenasotoh/opt/anaconda3/lib/python3.8/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-6' failed with exit status 1
>>> 
Anyone knows how to fix this?

danielstankw, 2021-10-07:
Use my newly written guide, let me know if it helps
https://github.com/danielstankw/Robosuite_Installation

RayYoh, 2021-11-23:
Use the command
```
 sudo apt-get update -q
 sudo DEBIAN_FRONTEND=noninteractive apt-get install -y curl  libgl1-mesa-dev libgl1-mesa-glx libglew-dev libosmesa6-dev 
 sudo apt-get clean
 sudo rm -rf /var/lib/apt/lists/* 
```