openai/mujoco-py #622: ERROR: GLEW initalization error: Missing GL version

KitanoFumiya, 2021-06-05:
I'm thinking of getting a camera elephant using mujoco. However, I get an error when I try to do the following:

"Img_rgb = self.sim.render (camera_name ='track', width = 64, height = 64, depth = False)"

ERROR: GLEW initalization error: Missing GL version

OS: Win10
mujoco200 win64
GPU: GeForce RTX 2060 SUPER
OpenGL: 4.6

Is this due to the GPU?
I would like to know if there is any solution or other way to get the camera image.

Thank you.

sprhawk, 2021-07-29:
On Linux, I used `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so.2.1` to fix this issue

RayYoh, 2021-11-23:
On Linux, you can uese:
```
sudo gedit ~/.bashrc
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so
source ~/.bashrc
```

sprhawk, 2021-11-23:
> On Linux, you can uese:
> 
> ```
> sudo gedit ~/.bashrc
> export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so
> source ~/.bashrc
> ```

You don't need to use `sudo` to edit your own `.bashrc`

ChouYuhao, 2022-04-13:
the same problem