openai/mujoco-py #627: How to install mujoco-py in a server without root privileges?

luckeciano, 2021-08-19:
Hello,

I started to use Mujoco in a shared server (Ubuntu 18.04). I would like to install `mujoco-py` package there, but I don't have root privileged to install dependencies (e.g., I can't execute `sudo apt install libosmesa6-dev libgl1-mesa-glx libglfw3`)

Is there any way to install it without root access?

Thanks!

PierreBoyeau, 2021-09-01:
Did you find a solution to this problem? I am facing the same issue.

ezhang7423, 2022-01-07:
I solved this issue with conda.

```
conda install -c conda-forge glew
conda install -c conda-forge mesalib
conda install -c menpo glfw3
```
Then add your conda environment include to CPATH (put this in your .bashrc to make it permanent):

```
export CPATH=<path_to_conda>/envs/<ENV_NAME>/include
```

Finally, install patchelf with `pip install patchelf`

luckeciano, 2022-04-10:
Thank you so much @ezhang7423! Really appreciate your help.

After installing these conda packages, I also had an issue to find lGL (`cannot find -lGL`). Mujoco-py README says to create a symlink using the libs in `/usr`, but I would need sudo privileges for that.

What I did was to copy the `libGL.so.1` library to the `lib` directory on my conda env and then create a symlink there (`ln -s libGL.so.1 libGL.so`).

qureshinomaan, 2022-11-09:
> I solved this issue with conda.
> 
> ```
> conda install -c conda-forge glew
> conda install -c conda-forge mesalib
> conda install -c menpo glfw3
> ```
> 
> Then add your conda environment include to CPATH (put this in your .bashrc to make it permanent):
> 
> ```
> export CPATH=$CONDA_PREFIX/include
> ```
> 
> Finally, install patchelf with `pip install patchelf`

Note that this works for python 3.8

RisingAuroras, 2023-01-06:
> Thank you so much @ezhang7423! Really appreciate your help.
>
> After installing these conda packages, I also had an issue to find lGL (`cannot find -lGL`). Mujoco-py README says to create a symlink [...]

@luckeciano  i had the same problem,have you solved it yet

Kelym, 2023-01-15:
> I solved this issue with conda.
>
> ```
> conda install -c conda-forge glew
> conda install -c conda-forge mesalib
> conda install -c menpo glfw3
> ```
>
> Then add your conda environment include to CPATH (put [...]

For people who installed these libraries to the non-base conda environment and have to batch script on a cluster machine: you might want to add this to your `.bashrc`. Note how `CPATH` comes after `conda activate`
```
conda activate BLABLA
export CPATH=$CONDA_PREFIX/include
```

leonasting, 2023-02-13:
> For people who installed these libraries to the non-base conda environment and have to batch script on a cluster machine: you might want to add this to your `.bashrc`. Note how `CPATH` comes after [...]

Still not working for me.

sachlatis, 2023-02-20:
Thanks for the suggestions. After these commands I have the following problem:
`/usr/bin/ld: cannot find -lOSMesa
/usr/bin/ld: cannot find -lGL`

Any suggestions?

gabrieletiboni, 2023-04-23:
> Still not working for me.
@leonasting have you got it working? I'm having the same issue when running mujoco 2.1 on a cluster

Epicato, 2023-11-30:
> Thank you so much @ezhang7423! Really appreciate your help.
>
> After installing these conda packages, I also had an issue to find lGL (`cannot find -lGL`). Mujoco-py README says to create a symlink [...]

I had the same problem and it works for me. 
Thanks a lot!

sophistz, 2024-04-06:
> Thank you so much @ezhang7423! Really appreciate your help.
>
> After installing these conda packages, I also had an issue to find lGL (`cannot find -lGL`). Mujoco-py README says to create a symlink [...]

Hi, I solved this issue. But I still got missing -IOSMesa. Is there a similar way to solve this one as well? I don't have root access and use conda to install everything. Thank you.

Epiphanye30, 2024-05-26:
> Hi, I solved this issue. But I still got missing -IOSMesa. Is there a similar way to solve this one as well? I don't have root access and use conda to install everything. Thank you.

Hi! I met the same issue. I checked my $CONDA_PREFIX/lib and found `libOSMesa32.so`. It seemed to be lOSMesa in 32bit system, so I added at the end of `~/.bashrc:
```
export LD_LIBRARY_PATH=$CONDA_PREFIX/lib:$LD_LIBRARY_PATH
if [ -f "$CONDA_PREFIX/lib/libOSMesa32.so" ] && [ ! -f "$CONDA_PREFIX/lib/libOSMesa.so" ]; then
    ln -s $CONDA_PREFIX/lib/libOSMesa32.so $CONDA_PREFIX/lib/libOSMesa.so
fi
```
After `source ~/.bashrc`
My problem was solved, hope it would be helpful to you :)