openai/mujoco-py #635: No root access: fatal error: GL/osmesa.h: No such file or directory

shivanimall, 2021-10-04:
how to resolve this issue with mujoco-py installation with no root access- PLEASE NOTE: `can't run apt-get install`

```
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /sailhome/small02/miniconda2/envs/robos/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco-py/mujoco_py -I/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py -I/sailhome/small02/.mujoco/mujoco200/include -I/sailhome/small02/miniconda2/envs/robos/lib/python3.8/site-packages/numpy/core/include -I/sailhome/small02/miniconda2/envs/robos/include/python3.8 -c /cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/cymj.c -o /cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxcpuextensionbuilder/temp.linux-x86_64-3.8/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /sailhome/small02/miniconda2/envs/robos/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco-py/mujoco_py -I/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py -I/sailhome/small02/.mujoco/mujoco200/include -I/sailhome/small02/miniconda2/envs/robos/lib/python3.8/site-packages/numpy/core/include -I/sailhome/small02/miniconda2/envs/robos/include/python3.8 -c /cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/gl/osmesashim.c -o /cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxcpuextensionbuilder/temp.linux-x86_64-3.8/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/unixccompiler.py", line 117, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] +
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/spawn.py", line 157, in _spawn_posix
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 273, in _build_impl
    so_file_path = super()._build_impl()
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/site-packages/Cython-3.0.0a9-py3.8-linux-x86_64.egg/Cython/Distutils/old_build_ext.py", line 192, in run
    _build_ext.build_ext.run(self)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/cvgl2/u/small02/robosuite/mujoco-py/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/sailhome/small02/miniconda2/envs/robos/lib/python3.8/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
```

dmklee, 2021-12-13:
I was able to get it to work using anaconda, without needing root access.  
```
conda activate <env-name>
conda install -c menpo osmesa
pip install mujoco-py
C_INCLUDE_PATH=:$HOME/.conda/envs/<env-name>/include python -c "import mujoco_py"
```
This solution was originally suggested in [#96 ](https://github.com/openai/mujoco-py/issues/96#issuecomment-346685411), but the suggested osmesa package was no longer available.

EmiliyanGospodinov, 2023-01-14:
@dmklee Thanks man, it worked for me!