openai/mujoco-py #645: I cannot import mujoco_py

JhonPool4, 2021-11-02:
Hi,
I want to use mujoco for an university project. Thus, I installed mujoco_py following indications on [link](https://github.com/openai/mujoco-py#install-mujoco) However when I try to import mujoco_py I get error:
```
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jhon/.local/lib/python3.8/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/jhon/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/jhon/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 101, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/jhon/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 125, in load_dynamic_ext
    return loader.load_module()
  File "mujoco_py/cymj.pyx", line 1, in init mujoco_py.cymj
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
```

Thanks!

wookayin, 2021-11-02:
Which numpy version did you have?

You should try reinstalling numpy (w/ the latest version):
```
pip uninstall numpy
pip install --upgrade numpy
```

JhonPool4, 2021-11-02:
Now is working, thanks!!!

closed 2021-11-02 by JhonPool4