openai/mujoco-py #647: No matching distribution found for mujoco-py<2.2,>=2.1

nrkumar93, 2021-11-02:
I tried to install `mujoco-py 2.1` . This is the error I am getting. 

```
 ~ pip3 install -U 'mujoco-py<2.2,>=2.1'
Collecting mujoco-py<2.2,>=2.1
  Could not find a version that satisfies the requirement mujoco-py<2.2,>=2.1 (from versions: 0.2.4, 0.2.5, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.5.0, 0.5.1, 0.5.2, 0.5.3, 0.5.4, 0.5.5, 0.5.6, 0.5.7, 1.5.0.0.1, 1.50.0.1, 1.50.0.2, 1.50.0.3, 1.50.0.4, 1.50.0.5, 1.50.0.6, 1.50.0.7, 1.50.0.8, 1.50.0.9, 1.50.0.10, 1.50.0.11, 1.50.0.12, 1.50.0.13, 1.50.0.14, 1.50.1.0, 1.50.1.1, 1.50.1.2, 1.50.1.3, 1.50.1.4, 1.50.1.5, 1.50.1.6, 1.50.1.7, 1.50.1.8, 1.50.1.9, 1.50.1.10, 1.50.1.11, 1.50.1.12, 1.50.1.13, 1.50.1.14, 1.50.1.15, 1.50.1.16, 1.50.1.17, 1.50.1.18, 1.50.1.19, 1.50.1.20, 1.50.1.21, 1.50.1.22, 1.50.1.23, 1.50.1.24, 1.50.1.25, 1.50.1.26, 1.50.1.27, 1.50.1.28, 1.50.1.29, 1.50.1.30, 1.50.1.31, 1.50.1.32, 1.50.1.33, 1.50.1.34, 1.50.1.35, 1.50.1.36, 1.50.1.37, 1.50.1.38, 1.50.1.39, 1.50.1.40, 1.50.1.41, 1.50.1.42, 1.50.1.43, 1.50.1.44, 1.50.1.45, 1.50.1.46, 1.50.1.47, 1.50.1.48, 1.50.1.49, 1.50.1.50, 1.50.1.51, 1.50.1.52, 1.50.1.53, 1.50.1.54, 1.50.1.55, 1.50.1.56, 1.50.1.59, 1.50.1.62, 1.50.1.63, 1.50.1.64, 1.50.1.65a1, 1.50.1.65, 1.50.1.65.1, 1.50.1.65.2, 1.50.1.65.3, 1.50.1.66, 1.50.1.67, 1.50.1.68, 2.0.2.0a1, 2.0.2.0, 2.0.2.1, 2.0.2.2, 2.0.2.3, 2.0.2.4, 2.0.2.5, 2.0.2.6, 2.0.2.7, 2.0.2.8, 2.0.2.9, 2.0.2.10, 2.0.2.11, 2.0.2.12, 2.0.2.13)
No matching distribution found for mujoco-py<2.2,>=2.1
```

wookayin, 2021-11-02:
This is not released yet. The OpenAI team will make a release soon.

In the meantime installing the latest version via

    pip install git+https://github.com/openai/mujoco-py

will work for you.

nrkumar93, 2021-11-02:
That worked (only with Python 3.7 though because of Numpy dependency). 

Thanks for the quick reply.

closed 2021-11-02 by nrkumar93

MotorCityCobra, 2021-11-13:
> This is not released yet. The OpenAI team will make a release soon.
> 
> In the meantime installing the latest version via
> 
> ```
> pip install git+https://github.com/openai/mujoco-py
> ```
> 
> will work for you.

I'm getting a nasty error message  

(mbot) iii@iii-MS-7A32:~$ pip install git+https://github.com/openai/mujoco-py
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/openai/mujoco-py
  Cloning https://github.com/openai/mujoco-py to /tmp/pip-req-build-lb1nmvxz
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for git+https://github.com/openai/mujoco-py.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied: imageio>=2.1.2 in ./.local/lib/python3.8/site-packages (from mujoco-py==2.0.2.13) (2.9.0)
Requirement already satisfied: fasteners~=0.15 in /usr/local/lib/python3.8/dist-packages (from mujoco-py==2.0.2.13) (0.16.3)
Requirement already satisfied: glfw>=1.4.0 in /usr/local/lib/python3.8/dist-packages (from mujoco-py==2.0.2.13) (2.4.0)
Requirement already satisfied: numpy>=1.11 in /usr/local/lib/python3.8/dist-packages (from mujoco-py==2.0.2.13) (1.19.5)
Requirement already satisfied: Cython>=0.27.2 in /usr/local/lib/python3.8/dist-packages (from mujoco-py==2.0.2.13) (0.29.24)
Requirement already satisfied: cffi>=1.10 in ./.local/lib/python3.8/site-packages (from mujoco-py==2.0.2.13) (1.14.4)
Requirement already satisfied: pycparser in ./.local/lib/python3.8/site-packages (from cffi>=1.10->mujoco-py==2.0.2.13) (2.20)
Requirement already satisfied: six in ./.local/lib/python3.8/site-packages (from fasteners~=0.15->mujoco-py==2.0.2.13) (1.15.0)
Requirement already satisfied: pillow in /usr/local/lib/python3.8/dist-packages (from imageio>=2.1.2->mujoco-py==2.0.2.13) (8.4.0)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /home/iii/.local/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpmm0u3s12
       cwd: /tmp/pip-req-build-lb1nmvxz
  Complete output (20 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/cymj_2.0.2.13_38_linuxgpuextensionbuilder_38.so
  Compiling /tmp/pip-req-build-lb1nmvxz/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-req-build-lb1nmvxz/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-lb1nmvxz
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-lb1nmvxz/mujoco_py
  creating /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-lb1nmvxz/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-req-build-lb1nmvxz/mujoco_py -I/home/iii/.mujoco/mujoco210/include -I/usr/local/lib/python3.8/dist-packages/numpy/core/include -I/tmp/pip-req-build-lb1nmvxz/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /tmp/pip-req-build-lb1nmvxz/mujoco_py/cymj.c -o /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-lb1nmvxz/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-req-build-lb1nmvxz/mujoco_py -I/home/iii/.mujoco/mujoco210/include -I/usr/local/lib/python3.8/dist-packages/numpy/core/include -I/tmp/pip-req-build-lb1nmvxz/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /tmp/pip-req-build-lb1nmvxz/mujoco_py/gl/eglshim.c -o /tmp/pip-req-build-lb1nmvxz/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-lb1nmvxz/mujoco_py/gl/eglshim.o -fopenmp -w
  /tmp/pip-req-build-lb1nmvxz/mujoco_py/gl/eglshim.c:4:10: fatal error: GL/glew.h: No such file or directory
      4 | #include <GL/glew.h>
        |          ^~~~~~~~~~~
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

wookayin, 2021-11-13:
@MotorCityCobra  You need to put your messages into codeblock (three backticks).

Regarding your problem: I bet you didn't run `sudo apt-get install libglew-dev`.

tfboyd, 2021-11-14:
@wookayin   If it is not released why did you update the README.md to say to run the command. I do the the releases for TF-Agents and Reverb (for DeepMind) and I don't update the instructions until I am pushing out the release.  You updated the README 11 days ago.  There is a lot of sloppiness in research which makes these issues more difficult.    TF-Agents has released CQL-SAC but we used D4RL and D4RL depends on dm_control (but they do so via github because they are also a bit sloppy) and dm_control is now looking for Mujoco 2.1. So I moved to Mujoco 2.1 but now I cannot build mujoco-py.  None of this is your fault in anyway other than why update the README before you release the code. It is 10pm on a Saturday night.  It is might fault for even working on it.  Oh well.  Hopefully I will notice a work around tomorrow or wait a few weeks until everything related to MuJoCo 2.1 is sorted out with the major libraries.

Thank you for maintaining this project, I know how thankless it can be from all directions. 

I tried running `LD_LIBRARY_PATH=/root/.mujoco/mujoco210/bin $python_version -m pip install -U 'mujoco-py<2.1,>=2.0'` but it seems to still look for `You appear to be missing MuJoCo.  We expected to find the file here: /root/.mujoco/mujoco200`.  I will look around and heck maybe just copy everything to mujoco200 for the short-term and see if that works so I can move forward.

tfboyd, 2021-11-14:
I give up and built from head given the other changes that look like they are also needed to use MuJoCo 2.1.  I can go with way for a few weeks (I hope less) with a big TODO(b/xxxxxxx) until an official release occurs and then swap it. Everything seems ok now. Smelly but ok.

wookayin, 2021-11-14:
@tfboyd  Actually, I am not a maintainer of the project -- I just happened to contribute to mujoco_py recently and be helping other users in several github issues as a community contributor because the official maintainers have been not responding as much.

README.me was updated in the PR #640 (by @nimrod-gileadi, not me :P ); I agree that README.md should have not updated yet but only after the official release. @matthiasplappert, Can we please have your attention on this so that mujoco_py 2.1 can be released **as soon as possible**, given that **a lot** of research projects depend on mujoco-py?

wookayin, 2021-11-18:
@matthiasplappert @jonasschneider @MillionIntegrals Any updates?

tfboyd, 2021-11-18:
I am working off head for the docker build.  I seeing hanging 10-20% of the
time.  I have no idea where it is hanging and this is in no way a bug
report.  I likely should not mention it in this thread.  One of those, is
anyone else seeing this? and I do not want to suggest a root cause as I
have no idea.

matthiasplappert, 2021-11-19:
I've just released a new version of `mujoco-py` on PyPI: https://pypi.org/project/mujoco-py/2.1.2.14/. This should now work with the free version of MuJoCo 2.1. Enjoy and let me know if it does not work for some reason.

tfboyd, 2021-11-19:
@matthiasplappert   THANK YOU.  You rock.  I own a bunch of our OSS releases and I know it is always as easy as people might think.