openai/mujoco-py #650: GL/osmesa.h: No such file or directory

GoingMyWay, 2021-11-12:
On anaconda, how can I install `GL/osmesa.h` with conda? I cannot use sudo and the system is red-hat.

wookayin, 2021-11-18:
Installing on a conda will be very tricky. Getting sudo will be a less-painful solution for you. 

If you want to give a try, follow the instructions at https://docs.mesa3d.org/download.html and https://docs.mesa3d.org/install.html

What you'll need to add is to add `--prefix=$CONDA_PREFIX` (e.g. `--prefix=/home/user/miniconda3/envs/your_virtual_env`) (or `--prefix=$HOME/.local/bin`). This will also assume that your `LDLIBRARY_PATH`, `C_INCLUDE_PATH`, etc. are also configured accordingly.  

Another reference: https://gist.github.com/SiyuanQi/600d1ce536791b7a3bd2e59fdbe69e66

dmklee, 2021-12-13:
Try my suggestion from #635, you can install `GL/osmesa.h` using `conda install -c menpo osmesa`.  Then, set the environment variable `C_INCLUDE_PATH=:$HOME/.conda/envs/<env-name>/include` when you import mujoco_py the first time.