openai/mujoco-py #651: Please add: gcc (MacPorts gcc11 11.2.0_1) 11.2.0

dbl001, 2021-11-14:
**Describe the bug**
Please add: gcc (MacPorts gcc11 11.2.0_1) 11.2.0 to the list of c_compilers in the file: mujoco_py/builder.py

**To Reproduce**
In project decision-transformer, in ~/decision-transformer/gym:
 $ conda env create -f conda_env.yml

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
RuntimeError: Could not find GCC executable.
  
  HINT: On OS X, install GCC with `brew install gcc`. or `port install gcc`.
  
**Desktop (please complete the following information):**
 - OS: [e.g. macOS 10.13.6]
 - OS X Monterey 12.1 Beta
 - 
 - Python Version [e.g. 3.6.6]
 - 
 - Mujoco Version [e.g. 1.50]
 - ```
 -  % ls -l ~/.mujoco 
total 8
-rw-r--r--@ 1 davidlaxer  staff  768 Nov 14 06:29 mjkey.txt
drwxr-xr-x@ 7 davidlaxer  staff  224 Oct  7 11:48 mujoco200
drwxr-xr-x@ 7 davidlaxer  staff  224 Oct  7 11:48 mujoco210
 - ```
 - mujoco-py version [e.g. 1.50.1.59]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`
```
**Additional context**
 c_compilers = [
                '/usr/local/bin/gcc-9',
                '/usr/local/bin/gcc-8',
                '/usr/local/bin/gcc-7',
                '/usr/local/bin/gcc-6',
                '/opt/local/bin/gcc-mp-9',
                '/opt/local/bin/gcc-mp-8',
                '/opt/local/bin/gcc-mp-7',
                '/opt/local/bin/gcc-mp-6',
            ]
            available_c_compiler = None
            for c_compiler in c_compilers:
                if distutils.spawn.find_executable(c_compiler) is not None:
                    available_c_compiler = c_compiler
                    break
            if available_c_compiler is None:
                raise RuntimeError(
                    'Could not find GCC executable.\n\n'
                    'HINT: On OS X, install GCC with '
                    '`brew install gcc`. or '
                    '`port install gcc`.')
            os.environ['CC'] = available_c_compiler
```

wookayin, 2021-11-18:
I think this is duplicate of #605. I agree with you that gcc detection in mujoco-py installation script should be improved.

In the meantime, a workaround for you would be:
```
export CC=/opt/homebrew/bin/gcc-11
pip install mujoco-py
```

wookayin, 2021-11-18:
@dbl001 May I ask if you are using Intel mac or M1 mac? What is your path to homebrew-installed gcc? i.e., `which gcc`

dbl001, 2021-11-19:
I am on the Intel iMac 2020.  I run MacPorts.

$ which gcc
/opt/local/bin/gcc

gcc 11.2.0