openai/mujoco-py #652: No such file or directory: 'patchelf' on mujoco-py installation

hbonnavaud, 2021-11-15:
**Describe the bug**
Fail to build Mujoco wheel, be cause of error "error: [Errno 2] No such file or directory: 'patchelf'".

**To Reproduce**

 - Download Mujoco for ubuntu at https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz
 - add to your ~/.bashrc the following line before to source it
```
 export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/disc/h.bonnavaud/.mujoco/mujoco210/bin
 export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
```
 - run "pip install git+https://github.com/openai/mujoco-py"

**Expected behavior**
mujoco-py is installed

**Error Messages**
```
Collecting git+https://github.com/openai/mujoco-py
  Cloning https://github.com/openai/mujoco-py to /tmp/pip-req-build-ybqkruux
  Running command git clone -q https://github.com/openai/mujoco-py /tmp/pip-req-build-ybqkruux
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for git+https://github.com/openai/mujoco-py.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Collecting glfw>=1.4.0
  Using cached glfw-2.4.0-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-manylinux2014_x86_64.whl (205 kB)
Requirement already satisfied: numpy>=1.11 in ./.local/lib/python3.8/site-packages (from mujoco-py==2.0.2.13) (1.21.4)
Collecting fasteners~=0.15
  Using cached fasteners-0.16.3-py2.py3-none-any.whl (28 kB)
Collecting cffi>=1.10
  Using cached cffi-1.15.0-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (446 kB)
Collecting Cython>=0.27.2
  Using cached Cython-0.29.24-cp38-cp38-manylinux1_x86_64.whl (1.9 MB)
Collecting imageio>=2.1.2
  Using cached imageio-2.10.4-py3-none-any.whl (3.3 MB)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from fasteners~=0.15->mujoco-py==2.0.2.13) (1.14.0)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Collecting pillow>=8.3.2
  Using cached Pillow-8.4.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.1 MB)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmpt20u729w build_wheel /tmp/tmp4ebpijd0
       cwd: /tmp/pip-req-build-ybqkruux
  Complete output (19 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-req-build-ybqkruux/mujoco_py/generated/cymj_2.0.2.13_38_linuxgpuextensionbuilder_38.so
  Compiling /tmp/pip-req-build-ybqkruux/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-req-build-ybqkruux/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-req-build-ybqkruux/mujoco_py -I/home/disc/h.bonnavaud/.mujoco/mujoco210/include -I/tmp/pip-build-env-auccvjc5/overlay/lib/python3.8/site-packages/numpy/core/include -I/tmp/pip-req-build-ybqkruux/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /tmp/pip-req-build-ybqkruux/mujoco_py/cymj.c -o /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-req-build-ybqkruux/mujoco_py -I/home/disc/h.bonnavaud/.mujoco/mujoco210/include -I/tmp/pip-build-env-auccvjc5/overlay/lib/python3.8/site-packages/numpy/core/include -I/tmp/pip-req-build-ybqkruux/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /tmp/pip-req-build-ybqkruux/mujoco_py/gl/eglshim.c -o /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py/gl/eglshim.o -fopenmp -w
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8
  creating /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8/mujoco_py
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py/cymj.o /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/tmp/pip-req-build-ybqkruux/mujoco_py/gl/eglshim.o -L/home/disc/h.bonnavaud/.mujoco/mujoco210/bin -Wl,--enable-new-dtags,-R/home/disc/h.bonnavaud/.mujoco/mujoco210/bin -lmujoco210 -lglewegl -o /tmp/pip-req-build-ybqkruux/mujoco_py/generated/_pyxbld_2.0.2.13_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8/mujoco_py/cymj.cpython-38-x86_64-linux-gnu.so -fopenmp
  error: [Errno 2] No such file or directory: 'patchelf'
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS
 - Python Version: Python 2.7.18 but same problem occurs with pip3, python3 version: Python 3.8.10
 - Mujoco Version: 2.1
 - mujoco-py version: Not installed

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
:/home/disc/h.bonnavaud/.mujoco/mujoco210/bin:/usr/lib/nvidia
 - output of: `echo $HOME`
/home/disc/h.bonnavaud
 - output of: `echo $USER`
h.bonnavaud

**Additional context**

ou-ais, 2021-11-16:
I'm having the same problem.

wookayin, 2021-11-16:
Try:

`sudo apt-get install patchelf`

hbonnavaud, 2021-11-17:
Ty, it worked for me

closed 2021-11-17 by hbonnavaud