openai/mujoco-py #653: `has_offscreen_render=True` does not work (GLEW initalization error: Missing GL version)

RayYoh, 2021-11-24:
**Describe the bug**  **To Reproduce**
When I use the Quickstart codes of mujoco_py, it work perfectly when I use:
```
env = suite.make(
    env_name="Lift", # try with other tasks like "Stack" and "Door"
    robots="Panda",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)
```
But when I use:
```
env = suite.make(
    env_name="Lift", # try with other tasks like "Stack" and "Door"
    robots="Panda",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=False,
    has_offscreen_renderer=True,
    use_camera_obs=True,
)
```
it dose not work, and there is an error `GLEW initalization error: Missing GL version`.

**Error Messages**
GLEW initalization error: Missing GL version

**Desktop (please complete the following information):**
 - OS: Ubuntu20.04
 - Python Version 3.8.12
 - Mujoco Version 200
 - mujoco-py version 2.0.2.13

I have try below to slove this problem, but this don't help.
 * https://github.com/openai/mujoco-py/issues/408#issuecomment-566796804
How can I do it?

lorepieri8, 2022-02-18:
The problem is still there, any solutions?

SATE001, 2022-04-27:
> The problem is still there, any solutions?
Give up immediately.

Instead, use 'OpenGL.GL.glReadPixels ()'.

famishedrover, 2023-08-28:
@SATE001 can you elaborate more on your response? (like some small example with some environment)? 
Thanks!