openai/mujoco-py #658: Installing mujoco-py on linux:

escape-velocity-labs, 2021-12-06:
Hi, I'm trying to install mujoco-py on a linux environment to be used with the Gym library. However. The installation seems to be fine but when I first import the library, it performs the following compilation:

**[1/1] Cythonizing /usr/local/lib/python3.7/dist-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension**

And it spits out the following error:

**ImportError: libglewosmesa.so: cannot open shared object file: No such file or directory**

On Mac, the installation went fine. Does anyone know how to solve this?

renxiongwang, 2021-12-13:
I am getting the same error with python 3.8

houyiwen, 2022-06-25:
My code works well in linux terminal, however i got the same error `ImportError: libglewosmesa.so: cannot open shared object file: No such file or directory` when running in Pycharm. 

I fix this problem by adding `LD_LIBRARY_PATH` in pycharm:
`Run --> Edit Configurations --> Enviroment variables`
Then add the `LD_LIBRARY_PATH` :
Name: LD_LIBRARY_PATH
Value: $LD_LIBRARY_PATH:/home/your_username/.mujoco/mujoco210/bin