openai/mujoco-py #662: Support MuJoCo 2.1.1 (including arm64 mac support)

wookayin, 2021-12-16:
I hope this can be a tracking issue for supporting [MuJoCo 2.1.1](https://github.com/deepmind/mujoco/releases/tag/2.1.1), which was just released few hours ago. There are some incompatible changes, so the python binding `mujoco-py` also needs to catch up those changes.

Required changes include:

* The path to the shared library has changed to `lib/libmujoco.so` (previously `bin/libmujoco210.so`)
* Perhaps expected installation paths might be `$HOME/.mujoco/mujoco211` per the convention. However, this naming convention is because of the directory name in the archive, which is `mujoco210`;  the new release 2.1.1 no longer uses this convention. Therefore mujoco installation path might also need to be changed (discussion/opinion needed).
* For MacOS, MuJoCo is now provided as a Framework bundle, so it is no longer expected to have a load path under `$HOME/.mujoco`. 

In addition, the builder script also needs to be updated because currently GCC only up to v9 is supported: https://github.com/openai/mujoco-py/issues/605.

I already managed to make mujoco-py work locally on my M1 machines, so creating and submitting a patch will be quite straightforward but we'll have to make a few decisions.

wookayin, 2021-12-16:
Until `mujoco-py` gets updated to officially support MuJoCo 2.1.1, the following should work as a *hacky* workaround.

```bash
mkdir -p $HOME/.mujoco/mujoco210         # Remove existing installation if any
ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/Headers/ $HOME/.mujoco/mujoco210/include
mkdir -p $HOME/.mujoco/mujoco210/bin
ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib $HOME/.mujoco/mujoco210/bin/libmujoco210.dylib
ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib /usr/local/lib/

# For M1 (arm64) mac users:
# The released binary doesn't ship glfw3, so need to install on your own
brew install glfw
ln -sf /opt/homebrew/lib/libglfw.3.dylib $HOME/.mujoco/mujoco210/bin
export CC=/opt/homebrew/bin/gcc-11         # see GH-605

pip install mujoco-py && python -c 'import mujoco_py'
```

I assumed that you copied `MuJoCo.app` into `/Applications`. Or you can feel free to change the path as you'd like, for example, `/Volume/MuJoCo/MuJoCo.app` if you simply mounted the `dmg` file from the downloaded archive file.

Excuse the (wrong) directory name `mujoco210` and library name `libmujoco210.dylib` because `mujoco-py==2.1.*` expects the mujoco installation directory/library name as if they were mujoco 2.1.0. This is, again, just a dirty workaround.

After this step, you should be able to run MuJoCo on a Apple Silicon Mac!

[image]

hueds, 2021-12-17:
Thanks for the tutorial!

My brew packages installed to different paths than the ones above, so I had to make the following adjustments:
```bash
# glfw
ln -sf /usr/local/Cellar/glfw/3.3.6/lib/libglfw.3.3.dylib $HOME/.mujoco/mujoco210/bin

# gcc
export CC=/usr/local/Cellar/gcc/11.2.0_3/bin/gcc-11
```

hueds, 2021-12-17:
[image]

I got this error, even though everything compiled smoothly. I suspect this is because I am using the x86_64 emulation mode on m1. I'll try again in arm64 mode.

wookayin, 2021-12-17:
If you are using an Intel mac, glfw installation would be different. Please feel free to change the path accordingly.

The x86_64 binary requires AVX instruction, which is not supported on M1 machines. The only way is to use **arm64** python: I recommend using miniforge3 (arm64) rather than miniconda3.

hueds, 2021-12-17:
Thanks for the tip! I got it running. There were some things I had to undo to get it working.
1. In general, my M1 terminal and brew was set to x86_64 emulation mode, so I had to undo that (e.g. turn off Rosetta emulation for iTerm.app, install brew for arm64). 
2. Uninstall glfw / gcc for the x86_64 brew, and reinstall  glfw/gcc using the arm64 brew.  
3. My miniconda env was built for intel. [Miniconda actually supports Arm now](https://docs.conda.io/en/master/miniconda.html.),  so I installed that version.

After doing these changes, my paths exactly matched the [original post's paths](https://github.com/openai/mujoco-py/issues/662#issuecomment-996081734). Then everything works.

raviswaroop-98, 2021-12-17:
`RuntimeError: Could not find supported GCC executable.

HINT: On OS X, install GCC 9.x with `brew install gcc@9`. or `port install gcc9``

I already have gcc 11.2.0_3 installed. I'm not sure why I'm getting this error

hueds, 2021-12-17:
Did you make sure the CC variable was set, i.e. `export CC=/opt/homebrew/bin/gcc-11 `

raviswaroop-98, 2021-12-17:
Yeah my bad made a spelling mistake. But now I have a different error when I try to import MuJoCo_py

`Import error. Trying to rebuild mujoco_py.
running build_ext
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/generated/cymj_2.1.2.14_39_macextensionbuilder_39.so, 0x0002): tried: '/Users/raviswarooprayavarapu/opt/anaconda3/lib/python3.9/site-packages/mujoco_py/generated/cymj_2.1.2.14_39_macextensionbuilder_39.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/cymj_2.1.2.14_39_macextensionbuilder_39.so' (no such file), '/usr/lib/cymj_2.1.2.14_39_macextensionbuilder_39.so' (no such file)`

raviswaroop-98, 2021-12-17:
anaconda is not native for m1 and its trying to compile mujoco_py for x86. Uninstalling anaconda and installing miniforge3 which is native for m1 solved the error.

XuJing1022, 2022-01-07:
Thanks for the tutorial!!

guptav96, 2022-01-17:
> Until `mujoco-py` gets updated to officially support MuJoCo 2.1.1, the following should work as a _hacky_ workaround.
> 
> First, make sure your python is running **arm64** (NOT x86_64 under Rosetta 2).
> 
> ```shell
> mkdir -p $HOME/.mujoco/mujoco210         # Remove existing installation if any
> ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/Headers/ $HOME/.mujoco/mujoco210/include
> mkdir -p $HOME/.mujoco/mujoco210/bin
> ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib $HOME/.mujoco/mujoco210/bin/libmujoco210.dylib
> ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib /usr/local/lib/
> 
> # For M1 (arm64) mac users:
> # The released binary doesn't ship glfw3, so need to install on your own
> brew install glfw
> ln -sf /opt/homebrew/lib/libglfw.3.dylib $HOME/.mujoco/mujoco210/bin
> export CC=/opt/homebrew/bin/gcc-11         # see https://github.com/openai/mujoco-py/issues/605
> 
> pip install mujoco-py && python -c 'import mujoco_py'
> ```
> 
> I assumed that you copied `MuJoCo.app` into `/Applications`. Or you can feel free to change the path as you'd like, for example, `/Volume/MuJoCo/MuJoCo.app` if you simply mounted the `dmg` file from the downloaded archive file.
> 
> Excuse the (wrong) directory name `mujoco210` and library name `libmujoco210.dylib` because `mujoco-py==2.1.*` expects the mujoco installation directory/library name as if they were mujoco 2.1.0. This is, again, just a temporary workaround.
> 
> After this step, you should be able to run MuJoCo on a Apple Silicon Mac!
> 
> [image]

Even after following these steps, I am getting the following error:

```
Import error. Trying to rebuild mujoco_py.
running build_ext
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/generated/cymj_2.1.2.14_39_macextensionbuilder_39.so, 2): Library not loaded: @rpath/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/mujoco_py/generated/cymj_2.1.2.14_39_macextensionbuilder_39.so
  Reason: image not found
```

enajx, 2022-01-18:
> Thanks for the tutorial!
> 
> My brew packages installed to different paths than the ones above, so I had to make the following adjustments:
> 
> ```shell
> # glfw
> ln -sf /usr/local/Cellar/glfw/3.3.6/lib/libglfw.3.3.dylib $HOME/.mujoco/mujoco210/bin
> 
> # gcc
> export CC=/usr/local/Cellar/gcc/11.2.0_3/bin/gcc-11
> ```

Got it to work in intel mac by changing the version in the symbolic link to .3 rather than 3.3

`ln -sf /usr/local/Cellar/glfw/3.3.6/lib/libglfw.3.dylib $HOME/.mujoco/mujoco210/bin`

Tanmay-Kulkarni101, 2022-01-26:
`(raps) tanmaykulkarni@Tanmays-MacBook-Pro bin % python -c 'import mujoco_py'
Compiling /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py
creating /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/gl
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -DONMAC -I/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py -I/Users/tanmaykulkarni/.mujoco/mujoco210/include -I/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/numpy/core/include -I/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/cymj.c -o /Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.14.6-arm64-3.8/Users/tanmaykulkarni/Courses/Spring/RA/Research/RAPS/raps/lib/python3.8/site-packages/mujoco_py/cymj.o -fopenmp -w
gcc-11: warning: this compiler does not support X86 (arch flags ignored)
gcc-11: error: unrecognized command-line option '-iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders'
gcc-11: error: unrecognized command-line option '-iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers'
Traceback (most recent call last):
[... cut ...]
distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed with exit status 1`

I followed the instructions given in 

> Until `mujoco-py` gets updated to officially support MuJoCo 2.1.1, the following should work as a _hacky_ workaround.
>
> First, make sure your python is running **arm64** (NOT x86_64 under Rosetta [...]

But on running ` python -c 'import mujoco_py'` it is not able to build successfully. I am currently using the arm64 version of python as well as gcc. Can someone help me with this...

nikhilweee, 2022-01-27:
@wookayin Thank you so much for putting in a workaround for now.  In my case, `/usr/local/lib` did not exist so I had to `sudo mkdir /usr/loca/lib` manually. This didn't seem necessary before but I had another error which required this.

TachikakaMin, 2022-02-10:
killed     python -c 'import mujoco_py'

Does anyone have idea? Thanks.

bikcrum, 2022-02-13:
I followed the steps indicated by @wookayin and it worked perfectly in my Macbook m1 max machine running Monterey 12.2.1. After a couple of weeks when I updated my mac to the Monterey beta 12.3 version, I begin to get `distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed with exit status 5`. After doing everything I could, I had to factory reset my MacBook to a non-beta version (back to 12.2.1) and was able to install it correctly again. Now, I am skeptical to run any update. Not sure if I won't see this update in the next stable version of mac.

alexyalunin, 2022-02-16:
What if I need to mujoco 1.5 (aka mjpro150) for gym, does the same workaround work?

alexyalunin, 2022-02-16:
> Until `mujoco-py` gets updated to officially support MuJoCo 2.1.1, the following should work as a _hacky_ workaround.
>
> First, make sure your python is running **arm64** (NOT x86_64 under Rosetta [...]

What if I need to mujoco 1.5 (aka mjpro150) for gym, does the same workaround work?

zdhNarsil, 2022-03-17:
I got `distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed: No such file or directory` on my M1 macbookpro.

TianhongDai, 2022-03-17:
> I got `distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed: No such file or directory` on my M1 macbookpro.

@zdhNarsil I suggest you need to first go to confirm if you have installed the `homebrew` in your macbook. Once you have installed it, you can use `which brew` to identify the location of `homebrew`. Finally, you can use `brew install gcc@11` to install `gcc-11`. I have successfully installed MuJoCo on my machine according to @wookayin 's instruction.

voidism, 2022-03-27:
I've done all the setups suggested by @wookayin  but still getting this error: `ld: library not found for -lmujoco210`
I've also tried the following possible steps:

```
❯ export LIBRARY_PATH=$LIBRARY_PATH:/Users/$USER/.mujoco
❯ export LIBRARY_PATH=$LIBRARY_PATH:/Users/$USER/.mujoco/mujoco210/
❯ export LIBRARY_PATH=$LIBRARY_PATH:/Users/$USER/.mujoco/mujoco210/bin
❯ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/$USER/.mujoco
❯ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/$USER/.mujoco/mujoco210/
❯ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/$USER/.mujoco/mujoco210/bin
```
But none of them works... Do you have nny ideas?

The full error message is:
```
❯ python -c 'import mujoco_py'
running build_ext
building 'mujoco_py.cymj' extension
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/jexus/miniforge3/include -arch arm64 -fPIC -O2 -isystem /Users/jexus/miniforge3/include -arch arm64 -DONMAC -I/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py -I/Users/jexus/.mujoco/mujoco210/include -I/Users/jexus/miniforge3/lib/python3.9/site-packages/numpy/core/include -I/Users/jexus/miniforge3/include/python3.9 -c /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.c -o /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-3.9/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.o -fopenmp -w
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/jexus/miniforge3/include -arch arm64 -fPIC -O2 -isystem /Users/jexus/miniforge3/include -arch arm64 -DONMAC -I/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py -I/Users/jexus/.mujoco/mujoco210/include -I/Users/jexus/miniforge3/lib/python3.9/site-packages/numpy/core/include -I/Users/jexus/miniforge3/include/python3.9 -c /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/gl/dummyshim.c -o /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-3.9/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/gl/dummyshim.o -fopenmp -w
/opt/homebrew/bin/gcc-11 -bundle -undefined dynamic_lookup -Wl,-rpath,/Users/jexus/miniforge3/lib -L/Users/jexus/miniforge3/lib -Wl,-rpath,/Users/jexus/miniforge3/lib -L/Users/jexus/miniforge3/lib /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-3.9/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.o /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-3.9/Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/gl/dummyshim.o -L/Users/jexus/.mujoco/mujoco210/bin -Wl,-rpath,/Users/jexus/.mujoco/mujoco210/bin -lmujoco210 -lglfw.3 -o /Users/jexus/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/lib.macosx-11.0-arm64-3.9/mujoco_py/cymj.cpython-39-darwin.so -fopenmp
ld: library not found for -lmujoco210
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/Users/jexus/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 204, in link
    self.spawn(linker + ld_args)
  File "/Users/jexus/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 917, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/Users/jexus/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/spawn.py", line 68, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1
```

liusf15, 2022-03-29:
I was also getting the error 
distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1

I was using Mujoco-2.1.3. Then I deleted it and downloaded mujoco-2.1.1-macos-universal2.dmg. I copied the MuJoCo.app to /Applications and double-clicked it to open the app. I ran the installation script again and it worked. 

(I'm using miniforge3, python=3.9.1, macOS Monterey 12.1.)

Somjit77, 2022-04-13:
I keep getting this error.
`ld: warning: ignoring file /Users/somjitnath/.mujoco/mujoco210/bin/libmujoco210.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: dylib (/Users/somjitnath/.mujoco/mujoco210/bin/libglfw.3.dylib) was built for newer macOS version (12.0) than being linked (11.0)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniforge/base/envs/tensorflow/lib/python3.9/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/opt/homebrew/Caskroom/miniforge/base/envs/tensorflow/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/opt/homebrew/Caskroom/miniforge/base/envs/tensorflow/lib/python3.9/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/opt/homebrew/Caskroom/miniforge/base/envs/tensorflow/lib/python3.9/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/opt/homebrew/Caskroom/miniforge/base/envs/tensorflow/lib/python3.9/site-packages/mujoco_py/generated/cymj_2.1.2.14_39_macextensionbuilder_39.so, 0x0002): symbol not found in flat namespace '_mjcb_act_bias'`

wookayin, 2022-05-24:
I updated [my instruction above](https://github.com/openai/mujoco-py/issues/662#issuecomment-996081734) to catch up the recent Mujoco [v2.2.0 release](https://github.com/deepmind/mujoco/releases/tag/2.2.0). Summary of changes are:

- Added an instruction to remove previous mujoco-py extensions after upgrading Mujoco to v2.2 (needs recompilation)
- Updated the installation instruction to correctly handle libmujoco.2.2.0.dylib  (due to version changes)
- Add more fool-proof instructions (like `CC` variables).

jaku-jaku, 2022-07-07:
> I updated [my instruction above](https://github.com/openai/mujoco-py/issues/662#issuecomment-996081734) to catch up with the recent Mujoco [v2.2.0 release](https://github.com/deepmind/mujoco/releases/tag/2.2.0). Summary of changes are:
> 
> * Added an instruction to remove previous mujoco-py extensions after upgrading Mujoco to v2.2 (needs recompilation)
> * Updated the command line instructions to correctly handle libmujoco.2.2.0.dylib  (due to version/filename change)
> * Added more fool-proof instructions (like `CC` variables).
> 
> **UPDATE**: It looks like the ABI of Mujoco 2.2.0 is NOT compatible with mujoco-py, resulting in Segmentation fault:
> 
> ```
> AssertionError: 
> Exception ignored in: 'mujoco_py.cymj.PyMjModel._set'
> Traceback (most recent call last):
>   File "wrappers.pxi", line 1139, in mujoco_py.cymj.PyMjModel._extract_mj_names
> ```
> 
> Until somebody figure out a workaround, please stick to mujoco 2.1.1 to use the obsolete mujoco-py bindings. The installation instruction will work for both versions.

I guess only 2.1.1 works. Failed to install for 2.2.0 due to gcc compiler error.

zhuyuanyang, 2022-07-13:
I try it again, and find the issue which may be resulting in the above errors.
When i install and execute the "pip install mujoco-py && python -c 'import mujoco_py'", i obtain the following errors:
The operation is based on Macos 12.4 and Apple clang version 13.1.6 (clang-1316.0.21.2.5).
Collecting mujoco-py
  Downloading mujoco_py-2.1.2.14-py3-none-any.whl (2.4 MB)
     |████████████████████████████████| 2.4 MB 1.0 MB/s 
Requirement already satisfied: glfw>=1.4.0 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (2.5.3)
Requirement already satisfied: imageio>=2.1.2 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (2.19.3)
Requirement already satisfied: fasteners~=0.15 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (0.17.3)
Requirement already satisfied: cffi>=1.10 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (1.15.0)
Requirement already satisfied: numpy>=1.11 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (1.22.3)
Requirement already satisfied: Cython>=0.27.2 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from mujoco-py) (0.29.30)
Requirement already satisfied: pycparser in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from cffi>=1.10->mujoco-py) (2.21)
Requirement already satisfied: pillow>=8.3.2 in ./opt/anaconda3/envs/RL/lib/python3.9/site-packages (from imageio>=2.1.2->mujoco-py) (9.0.1)
Installing collected packages: mujoco-py
Successfully installed mujoco-py-2.1.2.14
Compiling /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib/python3.9
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py
creating /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/gl
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/jack/opt/anaconda3/envs/RL/include -arch arm64 -I/Users/jack/opt/anaconda3/envs/RL/include -fPIC -O2 -isystem /Users/jack/opt/anaconda3/envs/RL/include -arch arm64 -DONMAC -Iopt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py -I/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py -I/Users/jack/.mujoco/mujoco210/include -I/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/numpy/core/include -I/Users/jack/opt/anaconda3/envs/RL/include/python3.9 -c /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/cymj.c -o /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.1-arm64-3.9/Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/cymj.o -fopenmp -w
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/jack/opt/anaconda3/envs/RL/include -arch arm64 -I/Users/jack/opt/anaconda3/envs/RL/include -fPIC -O2 -isystem /Users/jack/opt/anaconda3/envs/RL/include -arch arm64 -DONMAC -Iopt/anaconda3/envs/ [... cut ...] s/RL/lib/python3.9/site-packages/mujoco_py/gl/dummyshim.o -L/Users/jack/.mujoco/mujoco210/bin -Wl,-rpath,/Users/jack/.mujoco/mujoco210/bin -lmujoco210 -lglfw.3 -o /Users/jack/opt/anaconda3/envs/RL/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/lib.macosx-11.1-arm64-3.9/mujoco_py/cymj.cpython-39-darwin.so -fopenmp
**### ld: warning: dylib (/Users/jack/.mujoco/mujoco210/bin/libglfw.3.dylib) was built for newer macOS version (12.0) than being linked (11.1)**

alper111, 2022-11-23:
[This comment](https://github.com/openai/mujoco-py/issues/662#issuecomment-996081734) seems to be working. Can we at least include it in the README file so people can find it easily?

kclauw, 2022-12-17:
Is anyone able to run mujoco_py on macOS Monterey 12.6.2? I have tried everything mentioned here but I still get the following error : 

Import error. Trying to rebuild mujoco_py.
running build_ext
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/generated/cymj_2.1.2.14_310_macextensionbuilder_310.so, 0x0002): Library not loaded: '@rpath/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib'
  Referenced from: '/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/generated/cymj_2.1.2.14_310_macextensionbuilder_310.so'
  Reason: tried: '/opt/homebrew/Caskroom/miniforge/base/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/kenzoclauw/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/generated/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/gcc/aarch64-apple-darwin21/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/kenzoclauw/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/generated/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/gcc/aarch64-apple-darwin21/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.3.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/usr/local/lib/libmujoco.2.1.1.dylib' (no such file), '/usr/lib/libmujoco.2.1.1.dylib' (no such file)

kclauw, 2022-12-17:
> ps: I downloaded MuJoCo from https://github.com/deepmind/mujoco/releases/download/2.1.1/mujoco-2.1.1-macos-universal2.dmg

amine0el, 2023-04-28:
Import error. Trying to rebuild mujoco_py.
running build_ext
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so, 0x0002): symbol not found in flat namespace '_mjcb_act_bias'

you know why i get this erorr, i have a mac M1 pro

Nockda, 2023-06-14:
sh install-mujoco.sh
ln: /Users/hyunsuklee/.mujoco/mujoco210/include/Headers: Operation not permitted
ln: /usr/local/lib/: No such file or directory
Warning: glfw 3.3.8 is already installed and up-to-date.
To reinstall 3.3.8, run:
  brew reinstall glfw
Requirement already satisfied: mujoco-py in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (2.1.2.14)
Requirement already satisfied: glfw>=1.4.0 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (2.5.9)
Requirement already satisfied: Cython>=0.27.2 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (0.29.35)
Requirement already satisfied: cffi>=1.10 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (1.15.1)
Requirement already satisfied: imageio>=2.1.2 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (2.31.1)
Requirement already satisfied: fasteners~=0.15 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (0.18)
Requirement already satisfied: numpy>=1.11 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from mujoco-py) (1.24.3)
Requirement already satisfied: pycparser in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from cffi>=1.10->mujoco-py) (2.21)
Requirement already satisfied: pillow>=8.3.2 in /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages (from imageio>=2.1.2->mujoco-py) (9.5.0)
Import error. Trying to rebuild mujoco_py.
In file included from /opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/cymj.c:806:
/Users/hyunsuklee/.mujoco/mujoco210/include/mjmodel.h:18:10: fatal error: mujoco/mjtnum.h: No such file or directory
   18 | #include <mujoco/mjtnum.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/unixccompiler.py", line 186, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] + extra_postargs)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/ccompiler.py", line 1007, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/spawn.py", line 70, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 343, in _build_impl
    so_file_path = super()._build_impl()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 249, in _build_impl
    dist.run_commands()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/dist.py", line 1208, in run_command
    super().run_command(command)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/mujoco_py/builder.py", line 149, in build_extensions
    build_ext.build_extensions(self)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 468, in build_extensions
    self._build_extensions_serial()
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 494, in _build_extensions_serial
    self.build_extension(ext)
  File "/opt/homebrew/Caskroom/miniforge/base/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 549, in build_extension
    objects = self.compiler.compile(
[... cut ...] ersions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Caskroom/miniforge/base/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/usr/local/lib/libmujoco.2.1.1.dylib' (no such file), '/usr/lib/libmujoco.2.1.1.dylib' (no such file, not in dyld cache)
  
  
  I type the sh install-mujoco.sh
  and then there is error like this.
  what should i do

PrakashSrinivsan, 2023-08-08:
..

HollowGL, 2023-10-12:
I met an error: `command '/opt/homebrew/bin/gcc-11' failed with exit code 1`
Then I found that I had installed mujoco2.2.1 by mistake. So I reinstall mujoco2.1.1, and now it works.
Thanks for @wookayin so much!

maiyetum95, 2023-11-07:
> anaconda is not native for m1 and its trying to compile mujoco_py for x86. Uninstalling anaconda and installing miniforge3 which is native for m1 solved the error.

i use miniforge3 but seems that the problem is still there. how do you fix it?

Tridu33, 2024-02-09:
```bash
% python -c 'import mujoco_py'
Compiling /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx
performance hint: /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_warning_callback' to allow an error code to be returned.
performance hint: /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:104:5: Exception check on 'c_error_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_error_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_error_callback' to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const char *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const char *) except * nogil'.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 343, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 239, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/Users/tridu33/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "/Users/tridu33/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx
tridu33@tridu33deMacBook-Pro Downloads % where gcc-111
gcc-111 not found
tridu33@tridu33deMacBook-Pro Downloads % where gcc-11
/opt/homebrew/bin/gcc-11
tridu33@tridu33deMacBook-Pro Downloads % export CC=/opt/homebrew/bin/gcc-11
tridu33@tridu33deMacBook-Pro Downloads % python -c 'import mujoco_py'
Compiling /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx
performance hint: /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
[... 89 lines cut ...]
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const char *) except * nogil'.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
[... cut ...]
  File "/Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 239, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/Users/tridu33/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "/Users/tridu33/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: /Users/tridu33/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx
```

shreeram-murali, 2024-07-01:
@Tridu33 Were you able to resolve this issue? What was the fix?

timoklein, 2024-11-05:
I was able to fix @Tridu33's issue by running
```bash
pip install "cython<3"
```

See https://github.com/openai/mujoco-py/issues/773

gom168, 2025-05-13:
> Import error. Trying to rebuild mujoco_py. running build_ext Traceback (most recent call last): File "", line 1, in File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/**init**.py", line 2, in from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in cymj = load_cython_ext(mujoco_path) File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext mod = load_dynamic_ext('cymj', cext_so_path) File "/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext return loader.load_module() ImportError: dlopen(/Users/amineelblidi/Documents/Bachlor vorbereitung/code/PGMORL/.env/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so, 0x0002): symbol not found in flat namespace '_mjcb_act_bias'
> 
> you know why i get this erorr, i have a mac M1 pro

have you fixed this?