openai/mujoco-py #665: Failed to load OpenGL: Found 0 GPUs for rendering. Using device 0.

hbonnavaud, 2021-12-29:
**Describe the bug**
Unable to generage rgb array of environment state / image using environment.render (mode="rgb_array").

**To Reproduce**
Run the following code:
```
import mujoco_py
import gym
from gym.envs.mujoco.half_cheetah_v3 import HalfCheetahEnv
import numpy as np
from mujoco_py import GlfwContext

GlfwContext(offscreen=True)  # Trying to initialize an OpenGL context

env = HalfCheetahEnv()
test = env.reset()
mode_human = False  # Work well in human mode, don't work in "rgb_array" mode
image = env.render(mode="human" if mode_human else "rgb_array")
print("image is: ")
print(image)
```

**Expected behavior**
An image is returned and stored inside `image` variable, as a numpy.ndarray of pixels values.

**Error Messages**
```
Creating offscreen glfw
Found 0 GPUs for rendering. Using device 0.
Device id outside of range of available devices.
Traceback (most recent call last):
  File "mujoco_test.py", line 11, in <module>
    image = env.render(mode="human" if mode_human else "rgb_array")
  File "/home/disc/h.bonnavaud/.local/lib/python3.8/site-packages/gym/envs/mujoco/mujoco_env.py", line 157, in render
    self._get_viewer(mode).render(width, height, camera_id=camera_id)
  File "/home/disc/h.bonnavaud/.local/lib/python3.8/site-packages/gym/envs/mujoco/mujoco_env.py", line 186, in _get_viewer
    self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, -1)
  File "mujoco_py/mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
  File "mujoco_py/mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "mujoco_py/opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Python Version 3.8.10
 - Mujoco Version 2.1
 - mujoco-py version 2.0.2.13

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` -> /home/disc/h.bonnavaud/.mujoco/mujoco210/bin:/usr/lib/nvidia
 - output of: `echo $HOME` -> /home/disc/h.bonnavaud
 - output of: `echo $USER` -> h.bonnavaud
 - output of: `echo $LD_PRELOAD` -> /usr/lib/x86_64-linux-gnu/libGLEW.so

 - output of: `glxinfo | grep "OpenGL version"` -> OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3

**Additional context**
I found many hints about this problem but nothing about an error like "Found 0 GPUs for rendering. Using device 0.". Reading this line, the problem looks obvious since indice 0 of an empty list is out of range, but I have no idea how to solve it. Do not hesitate to redirect me if I'm at the wrong place, or if you know any helpful other issues reports. Thanks a lot.

tudorjnu, 2022-01-24:
I found out that if I comment the loading of the library, it works to render an image. I am really not sure what is wrong with it, maybe someone with more experience can help us here. 
```#export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so```

kaixindelele, 2022-02-18:
Thank you！ In other [issues](https://github.com/openai/mujoco-py/issues/187), I wrote the `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so` into ~/.bashrc, and then I couldn't get an rgb_array for env.render('rgb_array') directly,  I must first env.render(), then get an img from env.render ("rgb_array").

kaixindelele, 2022-02-18:
What's very annoying is, different Ubuntu versions also affect the render of mujoco (version 2.0 and below). I list some common permutations here：
[Conflict between render and image of robosuite](https://github.com/kaixindelele/Mujoco-Issues#robosuite%E7%9A%84render%E5%92%8Cimage%E5%86%B2%E7%AA%81%E9%97%AE%E9%A2%98)

Of course, you may need to Google translate~

nikhilweee, 2022-02-24:
What worked for me was to change [this line](https://github.com/openai/gym/blob/c8321e68bbd7e452ef65fc237525669979bb45af/gym/envs/mujoco/mujoco_env.py#L193) in `gym/envs/mujoco/mujoco_env.py` to:
```python
self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, None, -1)
```

hbonnavaud, 2022-02-24:
Ty @nikhilweee, maybe this correction should be included inside the mujoco.py code, and shouldn't be something the user should change?

tudorjnu, 2022-02-24:
@nikhilweee, you just made my life so much easier! Thank you for your answer!

tudorjnu, 2022-02-24:
@nikhilweee, are you able to render without starting the terminal or id with the graphics card? For me it throws the error of ```libGL error: failed to load driver: iris```.

closed 2022-05-19 by hbonnavaud

pjhae, 2022-06-19:
> There have been multiple issues which relate to the same error (#598, #187, #390) but unfortunately none of them worked for me. What worked was to change [this line](https://github.com/openai/gym/blob/c8321e68bbd7e452ef65fc237525669979bb45af/gym/envs/mujoco/mujoco_env.py#L193) in `<site_packages>/gym/envs/mujoco/mujoco_env.py` to:
> 
> ```python
> self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, None, -1)
> ```
> 
> You can find the location of `<site_packages>` in your system by using `pip show gym`.

Thankyou!!! you make my life very easier  :)
감사합니다

herene, 2023-02-22:
> 
I use this method, but there are two Creating window glfw in my program, and one is stuck because it doesn't get any output display information. Do you know why? Thank you very much for your answer!!

kaixindelele, 2023-02-22:
没太懂你的问题，建议报错贴详细信息，另外可以先参考一下我的这个教程：https://github.com/kaixindelele/Mujoco-Issues#robosuite%E7%9A%84render%E5%92%8Cimage%E5%86%B2%E7%AA%81%E9%97%AE%E9%A2%98

kaixindelele, 2023-02-22:
> > 
> 
> I use this method, but there are two Creating window glfw in my program, and one is stuck because it doesn't get any output display information. Do you know why? Thank you very much for your answer!!

没太懂你的问题，建议报错贴详细信息，另外可以先参考一下我的这个教程：https://github.com/kaixindelele/Mujoco-Issues#robosuite%E7%9A%84render%E5%92%8Cimage%E5%86%B2%E7%AA%81%E9%97%AE%E9%A2%98

herene, 2023-02-22:
> 没太懂你的问题，建议报错贴详细信息，另外可以先参考一下我的这个教程：https://github.com/kaixindelele/Mujoco-Issues#robosuite%E7%9A%84render%E5%92%8Cimage%E5%86%B2%E7%AA%81%E9%97%AE%E9%A2%98

我看到了一个同学可以同时拿到渲染场景和图片（https://blog.csdn.net/zhangkefang45/article/details/107313009?spm=1001.2014.3001.5502），所以就想尝试一下，但是我这样做之后画面一闪而过，且提示我Failed to initialize OpenGL，
我尝试按照上述说法更改MjRenderContextOffscreen，得到了下面这样的结果
[image]

kaixindelele, 2023-02-22:
> 我看到了一个同学可以同时拿到渲染场景和图片（[https://blog.csdn.net/zhangkefang45/article/details/107313009?spm=1001.2014.3001.5502），所以就想尝试一下，但是我这样做之后画面一闪而过，且提示我Failed](https://blog.csdn.net/zhangkefang45/article/details/10 [...]

你看我的帖子了么？这个很大程度上是因为你的Ubuntu版本是16.04，而你参考的博客的版本可能是18以上。至于为什么，我还不太清楚。如果说错了，可以先贴出来看看~

herene, 2023-02-23:
> 

> 你看我的帖子了么？这个很大程度上是因为你的Ubuntu版本是16.04，而你参考的博客的版本可能是18以上。至于为什么，我还不太清楚。如果说错了，可以先贴出来看看~

我的ubuntu系统是20.04的，我也没搞清楚为什么在会出现两个viewer，我并没有更改过底层的rende函数，现在采取的方案就是不使用gym，只在mujoco中渲染得到我自己设定的camera的rgb和depth。

sainavaneet, 2024-04-27:
> > 
> 
> > > > 没太懂你的问题,建议报错贴详细信息,另外可以先参考一下我的这个教程:https://github.com/kaixindelele/Mujoco-Issues#robosuite%E7%9A%84render%E5%92%8Cimage%E5%86%B2%E7%AA%81%E9%97%AE%E9%A2%98
> > > 
> > > 
> > > I saw a classmate who could get both the rendered scene and the image ([https://blog.csdn.net/zhangkefang45/article/details/107313009?spm=1001.2014.3001.5502), so I wanted to try it, but when I did so, the screen flashed and I was prompted to Failed](https://blog.csdn.net/zhangkefang45/article/details/107313009?spm=1001.2014.3001.5502%EF%BC%89%EF%BC%8C%E6%89%80%E4%BB%A5%E5%B0%B1%E6%83%B3%E5%B0%9D%E8%AF%95%E4%B8%80%E4%B8%8B%EF%BC%8C%E4%BD%86%E6%98%AF%E6%88%91%E8%BF%99%E6%A0%B7%E5%81%9A%E4%B9%8B%E5%90%8E%E7%94%BB%E9%9D%A2%E4%B8%80%E9%97%AA%E8%80%8C%E8%BF%87%EF%BC%8C%E4%B8%94%E6%8F%90%E7%A4%BA%E6%88%91Failed) To initialize OpenGL, I tried changing the MjRenderContextOffscreen as described above and got something like this [image]
> > 
> > 
> > 你看我的帖子了么？这个很大程度上是因为你的Ubuntu版本是16.04，而你参考的博客的版本可能是18以上。至于为什么，我还不太清楚。如果说错了，可以先贴出来看看~
> 
> 我的ubuntu系统是20.04的，我也没搞清楚为什么在会出现两个viewer，我并没有更改过底层的rende函数，现在采取的方案就是不使用gym，只在mujoco中渲染得到我自己设定的camera的rgb和depth。

i am also getting the mujoco_py viewer but i dont required it did you solve this problem ??

pjhae, 2024-04-29:
If you are running rendering and an RGB display for training simultaneously, it is normal for two windows to open, and seeing a black screen is typical.

If you have attempted training, it should be working properly even if the screen is black.
To check if the values are being outputted correctly, you can use matplotlib to visualize them. 

One tip would be not to render twice; instead, concatenate the RGB vectors and create a program that automatically saves the video, which is much more efficient.

Have a nice day :)

sainavaneet, 2024-04-29:
> If you are running rendering and an RGB display for training simultaneously, it is normal for two windows to open, and seeing a black screen is typical.
>
> If you have attempted training, it should be [...]

thank you got it