openai/mujoco-py #672: Cant install mujoco-py-1.50.1.68 in Windows

RonAvr, 2022-01-08:
Im trying to install mujoco_py v1.50.1.68 and i have some issues with the installtion.
I got "Failed building wheel for mujoco-py".
Couldnt find any solution at the other issues, hope someone can help me here.
Thank you.

My Desktop:
 - OS: Windows 10 x64
 - Python Version [e.g. 3.6.6]
 - Mujoco Version 1.50
 - mujoco-py version 1.50.1.68

The full error message is -

**Collecting mujoco-py==1.50.1.68
  Using cached mujoco-py-1.50.1.68.tar.gz (120 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: glfw>=1.4.0 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (2.5.0)
Requirement already satisfied: numpy>=1.11 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (1.21.5)
Requirement already satisfied: Cython>=0.27.2 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (0.29.26)
Requirement already satisfied: imageio>=2.1.2 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (2.13.5)
Requirement already satisfied: cffi>=1.10 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (1.15.0)
Requirement already satisfied: lockfile>=0.12.2 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from mujoco-py==1.50.1.68) (0.12.2)
Requirement already satisfied: pycparser in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from cffi>=1.10->mujoco-py==1.50.1.68) (2.21)
Requirement already satisfied: pillow>=8.3.2 in c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages (from imageio>=2.1.2->mujoco-py==1.50.1.68) (9.0.0)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (setup.py): started
  Building wheel for mujoco-py (setup.py): finished with status 'error'
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Attempting uninstall: mujoco-py
    Found existing installation: mujoco-py 0.5.7
    Uninstalling mujoco-py-0.5.7:
      Successfully uninstalled mujoco-py-0.5.7
    Running setup.py install for mujoco-py: started
    Running setup.py install for mujoco-py: finished with status 'error'
  Rolling back uninstall of mujoco-py
  Moving to c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages\mujoco_py-0.5.7.dist-info\
   from c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages\~ujoco_py-0.5.7.dist-info
  Moving to c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages\mujoco_py\
   from c:\users\ron\desktop\mujoco\p1\venv\pycharmprojects\lib\site-packages\~ujoco_py

  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Ron\Desktop\Mujoco\p1\venv\PycharmProjects\Scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Ron\\AppData\\Local\\Temp\\pip-install-8dz0bxi7\\mujoco-py_617d790c270e46039851b83c99ada0e8\\setup.py'"'"'; __file__='"'"'C:\\Users\\Ron\\AppData\\Local\\Temp\\pip-install-8dz0bxi7\\mujoco-py_617d790c270e46039851b83c99ada0e8\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Ron\AppData\Local\Temp\pip-wheel-_slwt5u3'
       cwd: C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\
  Complete output (428 lines):
  running bdist_wheel
  running build
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:121:4: 'mjtJoint' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:127:4: 'mjtGeom' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:149:4: 'mjtCamLight' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:157:4: 'mjtTexture' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:168:4: 'mjtCollision' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:174:4: 'mjtCone' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:179:4: 'mjtJacobian' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:185:4: 'mjtSolver' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:191:4: 'mjtImp' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:198:4: 'mjtRef' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:203:4: 'mjtEq' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:211:4: 'mjtWrap' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:220:4: 'mjtTrn' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:230:4: 'mjtDyn' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:237:4: 'mjtGain' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:242:4: 'mjtBias' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:248:4: 'mjtObj' redeclared
  warning: mujoco_py\generated\../pxd\mjmodel.pxd:274:4: 'mjtConstraint' redeclared
[... 395 lines cut ...]
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData\Local
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData\Local\Temp
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7
  creating C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8
  error: could not create 'C:\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8\mujoco_py\generated\_pyxbld_1.50.1.68_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\Ron\AppData\Local\Temp\pip-install-8dz0bxi7\mujoco-py_617d790c270e46039851b83c99ada0e8': The system cannot find the path specified
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\Ron\Desktop\Mujoco\p1\venv\PycharmProjects\Scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Ron\\AppData\\Local\\Temp\\pip-install-8dz0bxi7\\mujoco-py_617d790c270e46039851b83c99ada0e8\\setup.py'"'"'; __file__='"'"'C:\\Users\\Ron\\AppData\\Local\\Temp\\pip-install-8dz0bxi7\\mujoco-py_617d790c270e46039851b83c99ada0e8\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptool [... cut ...] n)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Ron\AppData\Local\Temp\pip-record-2l1bgbzg\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Ron\Desktop\Mujoco\p1\venv\PycharmProjects\include\site\python3.7\mujoco-py' Check the logs for full command output.**

Evanwu1125, 2022-03-28:
Have you solved your problem

capriciousZihao, 2022-12-11:
In my case, 'offline' install solved this issue. 
1) download _mujoco-py-1.50.1.68.tar.gz_ _from https://pypi.org/project/mujoco-py/1.50.1.68/_
2) _pip install ./mujoco-py-1.50.1.68.tar.gz_

Enigma-cit, 2023-10-30:
@capriciousZihao can you pls provide the detailed installation steps of mujoco_py in Windows 10?
It's quite urgent!

Thank you!