openai/mujoco-py #676: Dockerfile Build Error

peterdavidfagan, 2022-01-23:
Issue:

The current Dockerfile fails to build, it appears that the Ubuntu 16.04 xenial package is no longer being hosted within ppa:deadsnakes/ppa. Below is the error I have observed when building the image.
 
```
Step 4/26 : RUN DEBIAN_FRONTEND=noninteractive apt-get install --yes python3.6-dev python3.6 python3-pip
 ---> Running in 06c42aa3641f
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3.6-dev
E: Couldn't find any package by glob 'python3.6-dev'
E: Couldn't find any package by regex 'python3.6-dev'
E: Unable to locate package python3.6
E: Couldn't find any package by glob 'python3.6'
E: Couldn't find any package by regex 'python3.6'
The command '/bin/sh -c DEBIAN_FRONTEND=noninteractive apt-get install --yes python3.6-dev python3.6 python3-pip' returned a non-zero code: 100
```

**Reproduce steps:**
Run the below command from the root directory of this repository.
`docker build -t mujoco_py .`

**Expected behavior:**
A successful build of the image.

**Fix:**
I plan to open a pull request to fix this issue but I would also be happy to hear comments/suggestions on the above from others.

peterdavidfagan, 2022-01-23:
FYI @nimrod-gileadi, is it time to update the image to Focal? It would be cool to discuss this further, I am happy to work towards the required updates if necessary. Maybe the following base image: nvidia/cuda@sha256:8480ffb4a547ba36cb9b9553eac5cdbb3fd33c346351c41a947279838817c7d8

I also had a question relating to the symbolic links, what is the advantage of using these versus simply activating the virtual environment? I thought it would be cleaner to keep the default python installation separate to the create virtual environment.

peterdavidfagan, 2022-01-23:
I have submitted the above pull request to rectify this issue, looking forward to feedback on this and potentially implementing further changes where needed. 

Have a great rest of the day.

peterdavidfagan, 2022-01-24:
Also please find the following related issue: https://github.com/deadsnakes/issues/issues/195.

JiahuiSun, 2022-04-24:
I make a simpler Dockerfile that installs ubuntu18.04, mujoco2.1.0 and anaconda.

# Install nvidia docker for GPU support
# https://hub.docker.com/r/nvidia/cuda
FROM nvidia/cuda:11.3.0-devel-ubuntu18.04

# Install all needed tools
RUN apt-get update -q \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y \
    openssh-server \
    curl \
    git \
    vim \
    wget \
    net-tools \
    software-properties-common \
    xserver-xorg-dev \
    libgl1-mesa-dev \
    libgl1-mesa-glx \
    libglew-dev \
    libosmesa6-dev \
    patchelf \
    build-essential \
    libssl-dev \
    libffi-dev \
    libxml2-dev \
    libxslt1-dev \
    zlib1g-dev \
    libglew1.5 \
    libegl1-mesa \
    libxrandr2 \ 
    libxss1 \
    libxcursor1 \
    libxcomposite1 \
    libasound2 \
    libxi6 \
    libxtst6 \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/*

# Install mujoco
RUN mkdir -p /root/.mujoco \
    && wget https://github.com/deepmind/mujoco/releases/download/2.1.0/mujoco210-linux-x86_64.tar.gz -O mujoco.tar.gz \
    && tar -xf mujoco.tar.gz -C /root/.mujoco \
    && rm mujoco.tar.gz
ENV LD_LIBRARY_PATH=/root/.mujoco/mujoco210/bin:${LD_LIBRARY_PATH}
ENV LD_LIBRARY_PATH=/usr/local/nvidia/lib64:${LD_LIBRARY_PATH}

# Install anaconda
RUN wget \
    https://repo.anaconda.com/archive/Anaconda3-2021.11-Linux-x86_64.sh \
    && mkdir /root/.conda \
    && bash Anaconda3-2021.11-Linux-x86_64.sh -b \
    && rm -f Anaconda3-2021.11-Linux-x86_64.sh
ENV PATH="/root/anaconda3/bin:${PATH}"

EXPOSE 22
EXPOSE 6006

# After create the image, users can run a container.
# The remaining steps are about creating the python env for mujoco-py. 
# Here is our example.

# conda init bash
# source .bashrc
# conda create -n mujoco python=3.7
# conda activate mujoco
# conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch
# pip install -U 'mujoco-py<2.2,>=2.1'