openai/mujoco-py #682: A Working Installation Guide For M1 Mac

geyang, 2022-02-06:
Issue #662  is getting a big mangled, so here is a working script that worked for me.  A copy of this guide can be found at [jaynes-starter-kit/docker/m1_mac_setup](https://github.com/geyang/jaynes-starter-kit/blob/master/docker/m1_mac_mujoco_setup.md)

## Background

Mujoco-py currently does not support `mujoco2.1.1`. The first arm64 release, which is needed for M1 Macs, came out a few weeks ago. Therefore `mujoco2.1.1` is needed in order to run MuJoCo natively on the M1 Mac.

## Pre-requisits:

- make sure you use `Miniforge` as your Conda environment
- install `glfw` via `brew install glfw`. **Note the location for the installation**
- download MuJoCo2.1.1 image that ends with a `*.dmg`. The new `mujoco2.1.1` is released as a Framework. You can copy the `MuJoCo.app` into `/Applications/` folder.

## Installation Script

Make a file locally called `install-mujoco.sh`, and put the following into it.

```bash
mkdir -p $HOME/.mujoco/mujoco210
ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/Headers/ $HOME/.mujoco/mujoco210/include

mkdir -p $HOME/.mujoco/mujoco210/bin
ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib $HOME/.mujoco/mujoco210/bin/libmujoco210.dylib
sudo ln -sf /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib /usr/local/lib/

# For M1 (arm64) mac users:
# brew install glfw
ln -sf /opt/homebrew/lib/libglfw.3.dylib $HOME/.mujoco/mujoco210/bin

# remove old installation
rm -rf /opt/homebrew/Caskroom/miniforge/base/lib/python3.9/site-packages/mujoco_py

# which python
# exit

export CC=/opt/homebrew/bin/gcc-11         # see https://github.com/openai/mujoco-py/issues/605
pip install mujoco-py && python -c 'import mujoco_py'
```

Enjoy!

t46, 2022-03-12:
This is awesome!!! Thanks for sharing your workaround!!

zdhNarsil, 2022-03-17:
Get error on M1 new macbook pro
`distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed: No such file or directory`

xfl03, 2022-03-29:
I tried it, but get following:
```
gcc-11: warning: this compiler does not support X86 (arch flags ignored)
gcc-11: error: unrecognized command-line option '-iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders'
gcc-11: error: unrecognized command-line option '-iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers'
```

Szepi, 2022-04-28:
On:
`command '/opt/homebrew/bin/gcc-11' failed: No such file or directory`
I got the same error. gcc-11 is not here on my Mac. You can install gcc using `brew install gcc` and hopefully this installs gcc-11 to the path. Then you won't need the `/opt/homebrew/bin` portion above, i.e., just remove that from the script.

DanqingZ, 2022-06-06:
I first had error
```
distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit status 1
```
and then I run
```
brew install gcc@11
```
but then I had error
```
gcc-11: warning: this compiler does not support x86 ('-arch' option ignored)
ld: warning: object file (/Users/danqing0703/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.9-x86_64-3.8/Users/danqing0703/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/gl/dummyshim.o) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: object file (/Users/danqing0703/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_macextensionbuilder/temp.macosx-10.9-x86_64-3.8/Users/danqing0703/opt/anaconda3/lib/python3.8/site-packages/mujoco_py/cymj.o) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: dylib (/Users/danqing0703/.mujoco/mujoco210/bin/libmujoco210.dylib) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: dylib (/Users/danqing0703/.mujoco/mujoco210/bin/libglfw.3.dylib) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: ignoring file /Users/danqing0703/opt/anaconda3/lib/libgomp.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: object file (/opt/homebrew/Cellar/gcc/11.3.0_1/lib/gcc/11/gcc/aarch64-apple-darwin20/11/libgcc.a(ldadd_4_4.o)) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: object file (/opt/homebrew/Cellar/gcc/11.3.0_1/lib/gcc/11/gcc/aarch64-apple-darwin20/11/libgcc.a(lse-init.o)) was built for newer macOS version (11.0) than being linked (10.9)
```

Mr-13th, 2022-06-15:
Thank you! Following your steps help me fix this bug!

anuragajay, 2022-06-15:
You can follow @geyang steps with python 3.8 as well. If you don't already have `cffi` installed, you will need to install it with `pip install cffi==1.14` (rather than `pip install cffi` given 1.15 might not be compatible with python3.8)

JWK7, 2022-06-23:
When i run the sh file it brings the error: 
running build_ext
building 'mujoco_py.cymj' extension
/opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/johnchung/miniforge3/include -arch arm64 -fPIC -O2 -isystem /Users/johnchung/miniforge3/include -arch arm64 -I/opt/homebrew/opt/llvm/include -DONMAC -I/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py -I/Users/johnchung/.mujoco/mujoco210/include -I/Users/johnchung/miniforge3/lib/python3.9/site-packages/numpy/core/include -I/Users/johnchung/miniforge3/include/python3.9 -c /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.c -o /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-cpython-39/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.o -fopenmp -w
In file included from /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.c:754:
/Users/johnchung/.mujoco/mujoco210/include/mjmodel.h:18:10: fatal error: mujoco/mjtnum.h: No such file or directory
   18 | #include <mujoco/mjtnum.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 173, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] +
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 917, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/spawn.py", line 68, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 343, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 249, in _build_impl
    dist.run_commands()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/dist.py", line 1229, in run_command
    super().run_command(command)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 149, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 176, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1

Can someone help?

grahamannett, 2022-06-28:
> When i run the sh file it brings the error: running build_ext building 'mujoco_py.cymj' extension /opt/homebrew/bin/gcc-11 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/johnchung/miniforge3/include -arch arm64 -fPIC -O2 -isystem /Users/johnchung/miniforge3/include -arch arm64 -I/opt/homebrew/opt/llvm/include -DONMAC -I/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py -I/Users/johnchung/.mujoco/mujoco210/include -I/Users/johnchung/miniforge3/lib/python3.9/site-packages/numpy/core/include -I/Users/johnchung/miniforge3/include/python3.9 -c /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.c -o /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_39_macextensionbuilder/temp.macosx-11.0-arm64-cpython-39/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.o -fopenmp -w In file included from /Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/cymj.c:754: /Users/johnchung/.mujoco/mujoco210/include/mjmodel.h:18:10: fatal error: mujoco/mjtnum.h: No such file or directory 18 | #include <mujoco/mjtnum.h> | ^~~~~~~~~~~~~~~~~ compilation terminated. Traceback (most recent call last): File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 173, in _compile self.spawn(compiler_so + cc_args + [src, '-o', obj] + File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 917, in spawn spawn(cmd, dry_run=self.dry_run, **kwargs) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/spawn.py", line 68, in spawn raise DistutilsExecError( distutils.errors.DistutilsExecError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "", line 1, in File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/**init**.py", line 2, in from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in cymj = load_cython_ext(mujoco_path) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext cext_so_path = builder.build() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 226, in build built_so_file_path = self._build_impl() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 343, in _build_impl so_file_path = super()._build_impl() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 249, in _build_impl dist.run_commands() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands self.run_command(cmd) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/dist.py", line 1229, in run_command super().run_command(command) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command cmd_obj.run() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run _build_ext.build_ext.run(self) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run self.build_extensions() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/mujoco_py/builder.py", line 149, in build_extensions build_ext.build_extensions(self) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions _build_ext.build_ext.build_extensions(self) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 448, in build_extensions self._build_extensions_serial() File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 473, in _build_extensions_serial self.build_extension(ext) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 528, in build_extension objects = self.compiler.compile(sources, File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 574, in compile self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts) File "/Users/johnchung/miniforge3/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 176, in _compile raise CompileError(msg) distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1
> 
> Can someone help?

i was getting errors like this a lot before trying to get it working.  not exactly sure what i did to fix it but followed the instructions in the OP and its working.  main thing I think might have played an issue for me was using brew that is x86 specific and thus all the pythons you install with conda/miniforge/pyenv were i386 (at least that is my understanding)

required me to reinstall brew and removed old brew that is i386 specific

KouGZ, 2022-07-26:
i run 
```
export CC=/opt/homebrew/bin/gcc-11         # see https://github.com/openai/mujoco-py/issues/605
pip install mujoco-py && python -c 'import mujoco_py'
```
but got error
```
gcc-11: warning: this compiler does not support X86 (arch flags ignored)
ld: warning: object file (/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/cymj.o) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: object file (/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.o) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: dylib (/Users/guozhaokou/.mujoco/mujoco210/bin/libmujoco210.dylib) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: dylib (/Users/guozhaokou/.mujoco/mujoco210/bin/libglfw.3.dylib) was built for newer macOS version (11.0) than being linked (10.9)
ld: warning: dylib (/opt/homebrew/Cellar/gcc/11.2.0_1/lib/gcc/11/libgomp.dylib) was built for newer macOS version (11.5) than being linked (10.9)
ld: warning: dylib (/opt/homebrew/Cellar/gcc/11.2.0_1/lib/gcc/11/libgcc_s.1.1.dylib) was built for newer macOS version (11.0) than being linked (10.9)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/builder.py", line 111, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen(/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/generated/cymj_2.1.2.14_37_macextensionbuilder_37.so, 0x0002): tried: '/Users/guozhaokou/opt/anaconda3/envs/mujoco/lib/python3.7/site-packages/mujoco_py/generated/cymj_2.1.2.14_37_macextensionbuilder_37.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
```

natolambert, 2022-08-03:
Thanks @geyang, this is still useful!

geyang, 2022-08-03:
:)

jaku-jaku, 2022-08-03:
I believe mujoco-py only works with 2.1.x.

MuJoCo 2.2.x comes with m1 release support https://github.com/deepmind/mujoco/releases/tag/2.2.1 
And MuJoCo 2.2.x comes with native Python bindings officially, so you may refer to migration instructions: https://mujoco.readthedocs.io/en/latest/python.html#migration-notes-for-mujoco-py

All you need to do is `$pip install mujoco` and use mujoco directly.

Someone made similar to mujocopy viewer with native MuJoCo 2.2.x: https://github.com/rohanpsingh/mujoco-python-viewer, which is a great starting point. 
I have been doing refactoring lately based on that repo: https://github.com/jaku-jaku/jx-mujoco-python-viewer

I also found [dm_control](https://github.com/deepmind/dm_control) has many features and well designed.

Personal Thoughts: (I maybe wrong), but I feel mujoco-py maybe less significant as MuJoCo releases newer versions with native viewer officially in python bindings. As for now, you may follow mujoco-python-viewer as starting point, and combine with native MuJoCo and dm_control.

natolambert, 2022-08-03:
@jaku-jaku A lot of RL repositories still include `mujoco_py` as a dependency, so people are going to be dealing with this for years to come. Though I totally agree that it shouldn't really be the case. 🫠

jaku-jaku, 2022-08-03:
> @jaku-jaku A lot of RL repositories still include `mujoco_py` as a dependency, so people are going to be dealing with this for years to come. Though I totally agree that it shouldn't really be the case. 🫠
> 
> Does `pip install mujoco` let you `import mujoco_py`. I should try this.

Yeah, Based on my attempts with early installation with `2.2.x` (starting in May) and `mujoco-py`, MuJoCo seems to have change some of the naming styles in `2.2.x`. I am currently just getting into Mujoco, so I decided to skip the wrapper to reduce the dependency issue later on. 

`import mujoco` , you can see how it works with https://github.com/jaku-jaku/jx-mujoco-python-viewer
I think the dm_control seems to be more actively maintained and has many control interfaces.

louieworth, 2022-08-15:
When I followed this instruction and install Mujoco_py. There still comes with this error, and I think it comes to a new version of M1?
> ImportError: dlopen(/opt/anaconda3/envs/jl/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so, 0x0002): tried: '/opt/anaconda3/envs/jl/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))

Anyone, any suggestion?

wookayin, 2022-08-15:
@louieworth your python binary is probably not arm64 one. Ensure you have installed arm64 version of anaconda3:
```
$ lipo -archs $(which python3)
arm64
```

leochen1234321, 2022-08-22:
perfect! i have worked on it for my whole day! and it works for me,thank you!!!

edouardfouche, 2022-10-27:
Thank you @geyang ! It works on my M1 macbook with Python 3.9. 
I did not have a `/usr/local/lib` folder so I had to create it (`mkdir /usr/local/lib`)
Make sure to adjust the path `.../site-packages/mujoco_py` if you are working in another environment than `base`. 
I also had to remove `rm -rf .../site-packages/mujoco_py-2.1.2.14.dist-info`
Otherwise `pip install mujoco-py` thinks I already have `mujoco_py`

sachlatis, 2022-12-16:
Thank you! It worked for me too.
Regarding this error:
`
distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed with exit code 1
`
I had this error because I was using the version mujoco2.3.1.  I solved this error by going back to version **mujoco2.1.1**.

DDDOH, 2023-04-10:
> Get error on M1 new macbook pro
> `distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed: No such file or directory`

I use `export CC=/opt/homebrew/bin/gcc-12` instead and it worked for me. Check the version of gcc stored in the `/opt/homebrew/bin` folder.

kierad, 2023-07-13:
> On:
> `command '/opt/homebrew/bin/gcc-11' failed: No such file or directory`
> This is a sign that the brew install is running on emulation mode on the new Mac. You may want to upgrade to the arm64 (native) version of brew if you see this. See, e.g., this [page](https://blog.smittytone.net/2021/02/07/how-to-migrate-to-native-homebrew-on-an-m1-mac/).
> After installing brew, make sure you install glfw again, and also gcc. This seemed to do the trick for me. (Probably, the best order of doing this is to first have the native brew install, and then miniforge etc.)

To expand on this: On my M1 Mac I had to uncheck `Open using Rosetta` for my terminal. Then check that `uname -m` outputs `arm64`. Then install homebrew (if successful, it should install to /opt/homebrew/, not /usr/local/). Then install https://formulae.brew.sh/formula/gcc@11, then try original walkthrough again.

grahamannett, 2023-08-14:
anytime i have to create a new env for local dev of something that uses mujoco/gym/similar i run into slightly different issues somehow

DogWY, 2023-09-28:
Thanks for your workaround! Your solution helped me solve the same problem I faced before.
But when I tried to solve the problem according to the solution you gave, I faced the following error while executing the code:  `import mujoco_py`. 

```plain
[1/1] Cythonizing /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'
Traceback (most recent call last):
  File "main.py", line 8, in <module>
    import d4rl
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/__init__.py", line 14, in <module>
    import d4rl.locomotion
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/locomotion/__init__.py", line 2, in <module>
    from d4rl.locomotion import ant
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/locomotion/ant.py", line 20, in <module>
    import mujoco_py
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 278, in _build_impl
    so_file_path = super()._build_impl()
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 239, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1134, in cythonize
    cythonize_one(*args)
  File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx
```
My laptop is Macbook Pro with M1Pro, and my solution is to change the version of Cython by this code: `pip install Cython==3.0.0a10`. 

Hope my solution can help others.

promiseve, 2023-12-08:
> Get error on M1 new macbook pro `distutils.errors.CompileError: command '/opt/homebrew/bin/gcc-11' failed: No such file or directory`

Did anyone figure how to handle this error?

wookayin, 2023-12-08:
Try `ls /opt/homebrew/bin/gcc-*` and update the path accordingly. Don't just blindly execute the script.

In 2023 homebrew will install gcc-13 not gcc-11

joanvelja, 2024-01-24:
Hi all,

I am having trouble installing MuJoCo 2.1.1, specifically over this:

```
Successfully installed mujoco-py-2.1.2.14
Compiling /Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/cymj.pyx
ld: warning: duplicate -rpath '/Users/joanvelja/miniconda3/envs/mujoco_env/lib' ignored
ld: library 'glfw.3' not found
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 266, in link
    self.spawn(linker + ld_args)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 1041, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/spawn.py", line 70, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/opt/homebrew/Cellar/gcc/13.2.0/bin/gcc-13' failed with exit code 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 343, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 249, in _build_impl
    dist.run_commands()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/dist.py", line 989, in run_command
    super().run_command(command)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
    self.build_extensions()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/mujoco_py/builder.py", line 149, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 467, in build_extensions
    self._build_extensions_serial()
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 493, in _build_extensions_serial
    self.build_extension(ext)
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 572, in build_extension
    self.compiler.link_shared_object(
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 752, in link_shared_object
    self.link(
  File "/Users/joanvelja/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/setuptools/_distutils/unixccompiler.py", line 268, in link
    raise LinkError(msg)
distutils.errors.LinkError: command '/opt/homebrew/Cellar/gcc/13.2.0/bin/gcc-13' failed with exit code 1
```

Where it says that `ld: library 'glfw.3' not found`. I did install glfw through brew though, and the symbolic link is being made.

Any clue on what I might be missing, @geyang @wookayin? I've been bumping my head on this the entire day.  

Best

promiseve, 2024-01-25:
Hey @joanvelja , have you been able to figure this out ? Perhaps we can try troubleshooting together?

Anri-Lombard, 2024-02-18:
> Thanks for your workaround! Your solution helped me solve the same problem I faced before. But when I tried to solve the problem according to the solution you gave, I faced the following error while running to this code: `pip install mujoco-py && python -c 'import mujoco_py'`.
> 
> ```
> [1/1] Cythonizing /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx
> 
> Error compiling Cython file:
> ------------------------------------------------------------
> ...
>     See c_warning_callback, which is the C wrapper to the user defined function
>     '''
>     global py_warning_callback
>     global mju_user_warning
>     py_warning_callback = warn
>     mju_user_warning = c_warning_callback
>                        ^
> ------------------------------------------------------------
> 
> /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'
> 
> Error compiling Cython file:
> ------------------------------------------------------------
> ...
>     See c_warning_callback, which is the C wrapper to the user defined function
>     '''
>     global py_error_callback
>     global mju_user_error
>     py_error_callback = err_callback
>     mju_user_error = c_error_callback
>                      ^
> ------------------------------------------------------------
> 
> /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'
> Traceback (most recent call last):
>   File "main.py", line 8, in <module>
>     import d4rl
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/__init__.py", line 14, in <module>
>     import d4rl.locomotion
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/locomotion/__init__.py", line 2, in <module>
>     from d4rl.locomotion import ant
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/d4rl/locomotion/ant.py", line 20, in <module>
>     import mujoco_py
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
>     from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
>     cymj = load_cython_ext(mujoco_path)
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
>     cext_so_path = builder.build()
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 226, in build
>     built_so_file_path = self._build_impl()
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 278, in _build_impl
>     so_file_path = super()._build_impl()
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/builder.py", line 239, in _build_impl
>     dist.ext_modules = cythonize([self.extension])
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1134, in cythonize
>     cythonize_one(*args)
>   File "/root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
>     raise CompileError(None, pyx_file)
> Cython.Compiler.Errors.CompileError: /root/anaconda3/envs/mujoco_env/lib/python3.8/site-packages/mujoco_py/cymj.pyx
> ```
> 
> My laptop is Macbook Pro with M1Pro and the version of python is 3.8, and my solution is to change the version of Cython by this code: `pip install Cython==3.0.0a10`.
> 
> Hope my solution can help others.

Legend!

JacopoD, 2024-03-20:
> Hey @joanvelja , have you been able to figure this out ? Perhaps we can try troubleshooting together?

Did you manage to fix?

Binjian, 2024-07-02:
The fix still works till now on M3 MacOS Sonoma14.5. Here's the further patch
I use conda virtual enviroment with Python 3.10 (3.8 should also work).

```bash
pip install cffi=1.16.0
brew install gcc@11 

#actually installed Cython 0.29.37
pip install "Cython<3.0" 

#copy or make a softlink
cp -r /Applications/MuJoCo.app/Contents/Frameworks/MuJoCo.framework /Path/to/Conda/Env/lib/
```

KisaraBlue, 2024-10-14:
For me (M2 Sonoma) the script finally worked after these steps:
- using brew in /opt/homebrew to install `gcc@11` and `glfw`
- using miniforge3 
- installing cffi and cython with `pip install cffi==1.14` and `pip install Cython==3.0.0a10`
- following the instruction from @geyang

pranay8297, 2025-01-06:
> For me (M2 Sonoma) the script finally worked after these steps:
> 
> * using brew in /opt/homebrew to install `gcc@11` and `glfw`
> * using miniforge3 and python 3.8.20
> * installing cffi and cython with `pip install cffi==1.14` and `pip install Cython==3.0.0a10`
> * following the instructions from @geyang

I followed it and still it did not work for me

It was throwing this error `fatal error: mjmodel.h: No such file or directory` 

To solve this error, you have to download source code zip file from mujoco-2.1.1 release and copy the include folder and paste it $HOME/.mujoco/mujoco210

and then run python -c 'import mujoco_py'

This worked for me