openai/mujoco-py #683: distutils.errors.CompileError

DijieDeng, 2022-02-08:
**Describe the bug**
I am trying to install mujoco on my windows10 laptop. But it report the error as follow.

**To Reproduce**
I have installed other gym environments like the CartPole, BipedalWalkers and they run well.  I have tried to reinstall my vs tools as the picture shows below  
[image]

**Expected behavior**

**Error Messages**

(base) C:\year4\Hons\Evolutionary-Reinforcement-Learning-master>python main.py --env Humanoid-v2
running build_ext
building 'mujoco_py.cymj' extension
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\56325\anaconda3\lib\site-packages\mujoco_py -IC:\Users\56325\.mujoco\mjpro150\include -IC:\Users\56325\anaconda3\lib\site-packages\numpy\core\include -IC:\Users\56325\anaconda3\include -IC:\Users\56325\anaconda3\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /TcC:\Users\56325\anaconda3\lib\site-packages\mujoco_py/gl/dummyshim.c /FoC:\Users\56325\anaconda3\lib\site-packages\mujoco_py\generated\_pyxbld_1.50.1.59_39_windowsextensionbuilder\temp.win-amd64-3.9\Release\Users\56325\anaconda3\lib\site-packages\mujoco_py/gl/dummyshim.obj -fopenmp -w
cl : Command line warning D9025 : overriding '/W3' with '/w'
cl : Command line warning D9002 : ignoring unknown option '-fopenmp'
dummyshim.c
C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\gl\glshim.h(4): fatal error C1083: Cannot open include file: 'mujoco.h': No such file or directory
Traceback (most recent call last):
  File "C:\Users\56325\anaconda3\lib\distutils\_msvccompiler.py", line 408, in compile
    self.spawn(args)
  File "C:\Users\56325\anaconda3\lib\distutils\_msvccompiler.py", line 514, in spawn
    return super().spawn(cmd)
  File "C:\Users\56325\anaconda3\lib\distutils\ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "C:\Users\56325\anaconda3\lib\distutils\spawn.py", line 91, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\year4\Hons\Evolutionary-Reinforcement-Learning-master\main.py", line 46, in <module>
    env_constructor = EnvConstructor(args.env_name, args.frameskip)
  File "C:\year4\Hons\Evolutionary-Reinforcement-Learning-master\envs_repo\constructor.py", line 19, in __init__
    dummy_env = self.make_env()
  File "C:\year4\Hons\Evolutionary-Reinforcement-Learning-master\envs_repo\constructor.py", line 29, in make_env
    env = GymWrapper(self.env_name, self.frameskip)
  File "C:\year4\Hons\Evolutionary-Reinforcement-Learning-master\envs_repo\gym_wrapper.py", line 19, in __init__
    self.env = gym.make(env_name)
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\registration.py", line 142, in make
    return registry.make(id, **kwargs)
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\registration.py", line 87, in make
    env = spec.make(**kwargs)
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\registration.py", line 58, in make
    cls = load(self.entry_point)
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\registration.py", line 17, in load
    mod = importlib.import_module(mod_name)
  File "C:\Users\56325\anaconda3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\mujoco\__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "C:\Users\56325\anaconda3\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 12, in <module>
    import mujoco_py
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\builder.py", line 476, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\builder.py", line 97, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\builder.py", line 210, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\builder.py", line 233, in _build_impl
    dist.run_commands()
  File "C:\Users\56325\anaconda3\lib\distutils\dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "C:\Users\56325\anaconda3\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\56325\anaconda3\lib\site-packages\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Users\56325\anaconda3\lib\distutils\command\build_ext.py", line 340, in run
    self.build_extensions()
  File "C:\Users\56325\anaconda3\lib\site-packages\mujoco_py\builder.py", line 133, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Users\56325\anaconda3\lib\site-packages\Cython\Distutils\old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "C:\Users\56325\anaconda3\lib\distutils\command\build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "C:\Users\56325\anaconda3\lib\distutils\command\build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "C:\Users\56325\anaconda3\lib\distutils\command\build_ext.py", line 529, in build_extension
    objects = self.compiler.compile(sources,
  File "C:\Users\56325\anaconda3\lib\distutils\_msvccompiler.py", line 410, in compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

**Desktop (please complete the following information):**
 - OS: windows 10
 - Python Version 3.9.7
 - gym version 0.15.6
 - Mujoco Version 1.50
 - mujoco-py version 1.50.1.59

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
Add any other context about the problem here.

Evanwu1125, 2022-03-27:
The same error, any suggestions...
And I wonder if you are Chinese because of your name, if yes, I hope I can have your WeChat because I find that you know a lot about ERL which I am recently dedicating to. @DijieDeng

DijieDeng, 2022-03-27:
> This email was sent to you by someone outside the University.
> You should only click on links or attachments if you are certain that the email is genuine and the content is safe.

[Image.jpeg]

获取 Outlook for iOS<https://aka.ms/o0ukef>

The University of Edinburgh is a charitable body, registered in Scotland, with registration number SC005336. Is e buidheann carthannais a th’ ann an Oilthigh Dhùn èideann, clàraichte an Alba, àireamh clàraidh SC005336.

Evanwu1125, 2022-03-27:
I can't see the image you posted.  think you can search my QQ instead. 793034850 @DijieDeng

vida0812, 2022-04-18:
Have you installed it now? I have encountered the same issues.

ruanyanghui, 2024-09-08:
o ,I  also have the same issues ,spent my time for  8 hours,but  do not over