openai/mujoco-py #690: "command 'x86_64-linux-gnu-gcc' failed with exit status 1" error while trying to install MuJoCo on Google Colab

Saptam04, 2022-02-20:
**Describe the bug**
I'm trying to install `MuJoCo` on Google Colab. I followed all the steps mentioned in [`mujoco-py`'s GitHub page](https://github.com/openai/mujoco-py) (i.e. setting the environment variables and installing additional packages). When I import `mujoco_py` or use `gym.make()` I get the following GCC error:
```
LinkError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```
I see that the same issue has been raised many times in the past (e.g. [this](https://github.com/openai/mujoco-py/issues/132) and [this](https://github.com/openai/mujoco-py/issues/284)). Does anybody know how to get it to run on Colab?

**To Reproduce**
Download MuJoCo and install `mujoco-py`.

**Expected behavior**
A fresh MuJoCo installation.

**Error Messages**
Following is the full error message:
```
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fdebug-prefix-map=/build/python3.7-pX47U3/python3.7-3.7.12=. -fstack-protector-strong -Wformat -Werror=format-security -g -fdebug-prefix-map=/build/python3.7-pX47U3/python3.7-3.7.12=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/lib/python3.7/dist-packages/mujoco_py -I/root/.mujoco/mujoco211/include -I/usr/local/lib/python3.7/dist-packages/numpy/core/include -I/usr/include/python3.7m -c /usr/local/lib/python3.7/dist-packages/mujoco_py/cymj.c -o /usr/local/lib/python3.7/dist-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/usr/local/lib/python3.7/dist-packages/mujoco_py/cymj.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fdebug-prefix-map=/build/python3.7-pX47U3/python3.7-3.7.12=. -fstack-protector-strong -Wformat -Werror=format-security -g -fdebug-prefix-map=/build/python3.7-pX47U3/python3.7-3.7.12=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/lib/python3.7/dist-packages/mujoco_py -I/root/.mujoco/mujoco211/include -I/usr/local/lib/python3.7/dist-packages/numpy/core/include -I/usr/include/python3.7m -c /usr/local/lib/python3.7/dist-packages/mujoco_py/gl/osmesashim.c -o /usr/local/lib/python3.7/dist-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/usr/local/lib/python3.7/dist-packages/mujoco_py/gl/osmesashim.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.7-pX47U3/python3.7-3.7.12=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /usr/local/lib/python3.7/dist-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/usr/local/lib/python3.7/dist-packages/mujoco_py/cymj.o /usr/local/lib/python3.7/dist-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/usr/local/lib/python3.7/dist-packages/mujoco_py/gl/osmesashim.o -L/root/.mujoco/mujoco211/bin -Wl,--enable-new-dtags,-R/root/.mujoco/mujoco211/bin -lmujoco210 -lglewosmesa -lOSMesa -lGL -o /usr/local/lib/python3.7/dist-packages/mujoco_py/generated/_pyxbld_2.1.2.14_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
---------------------------------------------------------------------------
DistutilsExecError                        Traceback (most recent call last)
[/usr/lib/python3.7/distutils/unixccompiler.py](https://localhost:8080/#) in link(self, target_desc, objects, output_filename, output_dir, libraries, library_dirs, runtime_library_dirs, export_symbols, debug, extra_preargs, extra_postargs, build_temp, target_lang)
    214 
--> 215                 self.spawn(linker + ld_args)
    216             except DistutilsExecError as msg:

33 frames
[/usr/lib/python3.7/distutils/ccompiler.py](https://localhost:8080/#) in spawn(self, cmd)
    909     def spawn(self, cmd):
--> 910         spawn(cmd, dry_run=self.dry_run)
    911 

[/usr/lib/python3.7/distutils/spawn.py](https://localhost:8080/#) in spawn(cmd, search_path, verbose, dry_run)
     35     if os.name == 'posix':
---> 36         _spawn_posix(cmd, search_path, dry_run=dry_run)
     37     elif os.name == 'nt':

[/usr/lib/python3.7/distutils/spawn.py](https://localhost:8080/#) in _spawn_posix(cmd, search_path, verbose, dry_run)
    158                           "command %r failed with exit status %d"
--> 159                           % (cmd, exit_status))
    160             elif os.WIFSTOPPED(status):

DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

LinkError                                 Traceback (most recent call last)
[<ipython-input-7-53120878d6a5>](https://localhost:8080/#) in <module>()
      1 import gym
      2 
----> 3 env = gym.make("Ant-v2")
      4 # import mujoco_py

[/usr/local/lib/python3.7/dist-packages/gym/envs/registration.py](https://localhost:8080/#) in make(id, **kwargs)
    143 
    144 def make(id, **kwargs):
--> 145     return registry.make(id, **kwargs)
    146 
    147 def spec(id):

[/usr/local/lib/python3.7/dist-packages/gym/envs/registration.py](https://localhost:8080/#) in make(self, path, **kwargs)
     88             logger.info('Making new env: %s', path)
     89         spec = self.spec(path)
---> 90         env = spec.make(**kwargs)
     91         # We used to have people override _reset/_step rather than
     92         # reset/step. Set _gym_disable_underscore_compat = True on

[/usr/local/lib/python3.7/dist-packages/gym/envs/registration.py](https://localhost:8080/#) in make(self, **kwargs)
[... 180 lines cut ...]
 No output.

**Additional context**
I installed the following additional packages as suggested in the Docker file:
```
!sudo apt-get install \
   patchelf \
   curl \
   git \
   libgl1-mesa-dev \
   libgl1-mesa-glx \
   libglew-dev \
   libglfw3 \
   libosmesa6-dev \
   python3-pip \
   python3-numpy \
   python3-scipy \
   net-tools \
   unzip \
   vim \
   wget \
   xpra \
   xserver-xorg-dev
```
PS: I tried to downgrade `mujoco-py`. All of the versions from `2.0.2.13` to `2.0.2.9` even fails to install using `pip`. Version `2.0.2.8` installs, but produces the same error mentioned above.

atonkamanda, 2022-02-22:
I've gone to the same error as you and finally found a fix, first the fix for a local setup won't work here for colab.

The real fix is to not do the old way of installing mujoco with the key and the binaries, causing dependencies and path issues.

Now that it is free, you can just do
```bash
!pip install free-mujoco-py
```

Without having to care about mjkey/environments variables.

You can check this example to make sure you are not missing any libraries or adding some useless one.
https://colab.research.google.com/drive/1KGMZdRq6AemfcNscKjgpRzXqfhUtCf-V?usp=sharing#scrollTo=3qeAEFtTG8dV

Should be fixed after that, tag me if it's not.

atonkamanda, 2022-02-22:
I should add that installing free-mujoco isn't something you add on the top of the other installation, otherwise it won't work, just get rid of the old files and do like in the notebook.

Hope it helps.

Saptam04, 2022-02-24:
@atonkamanda Thanks for the fix. In fact shortly after posting it, I found out this [Reddit post](https://www.reddit.com/r/reinforcementlearning/comments/r78mnx/has_anyone_succeeded_in_running_mujoco_in_colab/) and [this pull request ](https://github.com/openai/mujoco-py/pull/640) where they point to this exact notebook.

A bit of a rant: now that MuJoCo is freely available (kudos Deepmind), I think the authors and the maintainers REALLY should try to put in some effort to make it a little bit more user-friendly. As it stands, it is undoubtedly a hassle to install it even on the most popular platforms like Ubuntu and Debian. I recently had the opportunity of installing it on a Manjaro-based system, and I can confirm that it was truly a nightmare!

closed 2022-02-24 by Saptam04

Nov05, 2024-03-02:
> I've gone to the same error as you and finally found a fix, first the fix for a local setup won't work here for colab.
> 
> The real fix is to not do the old way of installing mujoco with the key and the binaries, causing dependencies and path issues.
> 
> Now that it is free, you can just do
> 
> ```shell
> !pip install free-mujoco-py
> ```
> 
> Without having to care about mjkey/environments variables.
> 
> You can check this example to make sure you are not missing any libraries or adding some useless one. https://colab.research.google.com/drive/1KGMZdRq6AemfcNscKjgpRzXqfhUtCf-V?usp=sharing#scrollTo=3qeAEFtTG8dV
> 
> Should be fixed after that, tag me if it's not.

yep, this works for me. thank you very much!

```
!apt-get install -y \
    libgl1-mesa-dev \
    libgl1-mesa-glx \
    libglew-dev \
    libosmesa6-dev \
    software-properties-common
!apt-get install -y patchelf
!pip install free-mujoco-py
```