openai/mujoco-py #692: Installing MuJoCo on CentOS - distutils.errors.CompileError: command 'gcc' failed with exit status 1

RuhanaAzam, 2022-02-24:
**Describe the bug**
Hi, I am trying to get MuJoCo running on a CentOS  cluster. I get a compilation error due to missing OpenGL packages. I'm assuming the issue has to do with these packages missing: `libosmesa6-dev, libgl1-mesa-glx, libglfw3`. Although, CentOS does not support these packages. Does anyone know a way to get Mujoco working on CentOS?

**To Reproduce**
I have followed the instructions for the setup: downloaded the binaries, placed them in "~/.mujoco/mujoco210", and installed mujoco-py via pip. Then, run `import mujoco_py`.

**Error Messages**
```
/home/$USER/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory
 #include <GL/osmesa.h>

...

During handling of the above exception, another exception occurred:
  File "/home/$USER/miniconda3/envs/phiTrain/lib/python3.7/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
```

**Desktop **
 - OS: CentOS 7.9.2009
 - Python Version 3.7.11
 - Mujoco Version 2.1
 - mujoco-py version 2.1.2.14

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: /usr/local/cuda/lib64::/home/$USER/.mujoco/mujoco210/bin
 - output of: `echo $HOME`  : /home/$USER

dyllanwli, 2022-03-30:
Got the same issues on `Ubuntu 20.04 LTS`, `Python 3.8.3`, `Mujoco Version 2.1` and `mujoco-py version 2.1.2.14`

whliu90, 2022-04-02:
Hello, there! I got the same issues on CentOS 7. I tried this：
```
conda install -c menpo osmesa
C_INCLUDE_PATH=:$HOME/.conda/envs/<env-name>/include python -c "import mujoco_py"
```
, then I got an error like "No such file or directory: 'patchelf'". According to #652, I installed 
`sudo yum install patchelf`
,and re-installed mujoco-py with `python setup.py install` . After that, everything goes right magically.