openai/mujoco-py #699: distutils.errors.DistutilsExecError: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

Evanwu1125, 2022-03-28:
**Describe the bug**
distutils.errors.DistutilsExecError: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

**To Reproduce**
After I have successfully installed mujoco_py==1.50.1.68, when I want to import this package, this issue occurs.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
running build_ext
building 'mujoco_py.cymj' extension
"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Ic:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py -IC:\Users\ASUS\.mujoco\mjpro150\include -ID:\Anaconda\envs\pytorch\lib\site-packages\numpy\core\include -ID:\Anaconda\envs\pytorch\include -ID:\Anaconda\envs\pytorch\Include "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\ATLMFC\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tcc:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py/gl/dummyshim.c /Foc:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.7\Release\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py/gl/dummyshim.obj -fopenmp -w
cl: 命令行 warning D9025 :正在重写“/W3”(用“/w”)
cl: 命令行 warning D9002 :忽略未知选项“-fopenmp”
dummyshim.c
"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Ic:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py -IC:\Users\ASUS\.mujoco\mjpro150\include -ID:\Anaconda\envs\pytorch\lib\site-packages\numpy\core\include -ID:\Anaconda\envs\pytorch\include -ID:\Anaconda\envs\pytorch\Include "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\ATLMFC\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tcc:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\cymj.c /Foc:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.7\Release\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\cymj.obj -fopenmp -w
cl: 命令行 warning D9025 :正在重写“/W3”(用“/w”)
cl: 命令行 warning D9002 :忽略未知选项“-fopenmp”
cymj.c
c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\cymj.c(728): fatal error C1083: 无法打开包括文件: “mjmodel.h”: No such file or directory
Traceback (most recent call last):
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 412, in compile
    self.spawn(args)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 517, in spawn
    return super().spawn(cmd, env=env)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\ccompiler.py", line 917, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\spawn.py", line 69, in spawn
    "command %r failed with exit code %s" % (cmd, exitcode))
distutils.errors.DistutilsExecError: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 283, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 54, in load_cython_ext
    cext_so_path = builder.build()
  File "c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 131, in build
    dist.run_commands()
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\dist.py", line 1196, in run_command
    super().run_command(command)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\dist.py", line 986, in run_command
    cmd_obj.run()
  File "D:\Anaconda\envs\pytorch\lib\site-packages\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 339, in run
    self.build_extensions()
  File "c:\users\asus\.mujoco\mujoco-py-1.50.1.0\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 75, in build_extensions
    build_ext.build_extensions(self)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\Cython\Distutils\old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "D:\Anaconda\envs\pytorch\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 414, in compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

**Desktop (please complete the following information):**
 - OS: [Window 10]
 - Python Version [3.7.11]
 - Mujoco Version [1.50]
 - mujoco-py version [e.g. 1.50.1.68]

zzu-zhangzhen, 2022-05-03:
have you deal with the issue? i meet same issue

Evanwu1125, 2022-05-03:
I have solved this issue by downloading Microsoft visual c++ Build Tools 2015 instead of Microsoft visual studio

zzu-zhangzhen, 2022-05-04:
> I have solved this issue by downloading Microsoft visual c++ Build Tools 2015 instead of Microsoft visual studio

i  have  download  Microsoft visual c++ Build Tools 2015, but it still appear Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools" ,Is there anything else I need to set up

Evanwu1125, 2022-05-04:
Setting up mujoco in a new environment may solve this issue.

AuroraHashcat, 2024-09-05:
same issue but I have installed build tools 2015 and run it with anaconda environment. Maybe since I also have vs2022 then there is a conflict?