openai/mujoco-py #700: mujoco-py returning command 'x86_64-linux-gnu-gcc' failed with exit status 1 when running script

fmalato, 2022-04-07:
**Describe the bug**
I followed the [instructions](https://github.com/openai/mujoco-py) to install `mujoco-py` version `2.1.2.14` along with `mujoco 2.1.1`. The procedure goes well and everything gets installed correctly, but when I try to run my script (`python3 main.py`) it returns the following error:
```Compiling /home/federima/.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/federima/.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/federima/.local/lib/python3.8/site-packages/mujoco_py -I/home/federima/.mujoco/mujoco210/include -I/home/federima/.local/lib/python3.8/site-packages/numpy/core/include -I/home/federima/.local/lib/python3.8/site-packages/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /home/federima/.local/lib/python3.8/site-packages/mujoco_py/cymj.c -o /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/federima/.local/lib/python3.8/site-packages/mujoco_py/cymj.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/federima/.local/lib/python3.8/site-packages/mujoco_py -I/home/federima/.mujoco/mujoco210/include -I/home/federima/.local/lib/python3.8/site-packages/numpy/core/include -I/home/federima/.local/lib/python3.8/site-packages/mujoco_py/vendor/egl -I/usr/include/python3.8 -c /home/federima/.local/lib/python3.8/site-packages/mujoco_py/gl/eglshim.c -o /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/federima/.local/lib/python3.8/site-packages/mujoco_py/gl/eglshim.o -fopenmp -w
creating /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8
creating /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8/mujoco_py
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/federima/.local/lib/python3.8/site-packages/mujoco_py/cymj.o /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/federima/.local/lib/python3.8/site-packages/mujoco_py/gl/eglshim.o -L/home/federima/.mujoco/mujoco210/bin -Wl,--enable-new-dtags,-R/home/federima/.mujoco/mujoco210/bin -lmujoco210 -lglewegl -o /home/federima/.local/lib/python3.8/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8/mujoco_py/cymj.cpython-38-x86_64-linux-gnu.so -fopenmp
/usr/bin/ld: cannot find -lmujoco210
/usr/bin/ld: cannot find -lglewegl
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/usr/lib/python3.8/distutils/unixccompiler.py", line 215, in link
    self.spawn(linker + ld_args)
  File "/usr/lib/python3.8/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.8/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.8/distutils/spawn.py", line 157, in _spawn_posix
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "main.py", line 25, in <module>
    env = gym.make('geometry_mover:geometry_mover-v0')
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 75, in make
    cls = load(self.entry_point)
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 18, in load
    mod = importlib.import_module(mod_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/federima/point_mover/mujoco_geometry_mover/geometry_mover/envs/__init__.py", line 1, in <module>
    from geometry_mover.envs.geometry_mover import GeometryMover
  File "/home/federima/point_mover/mujoco_geometry_mover/geometry_mover/envs/geometry_mover.py", line 4, in <module>
    from gym.envs.mujoco import mujoco_env
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/mujoco/__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "/home/federima/.local/lib/python3.8/site-packages/gym/envs/mujoco/mujoco_env.py", line 12, in <module>
    import mujoco_py
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 296, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/federima/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 249, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
[... 10 lines cut ...]
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 550, in build_extension
    self.compiler.link_shared_object(
  File "/usr/lib/python3.8/distutils/ccompiler.py", line 713, in link_shared_object
    self.link(CCompiler.SHARED_OBJECT, objects,
  File "/usr/lib/python3.8/distutils/unixccompiler.py", line 217, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

Sounds like some kind of linking issue related to `lmujoco210` and `lglewegl`, as you can see in the upper part of the error message, but I'm not sure how to fix it.

I tried to apply all the solutions that I found from similar questions, but none worked.

**Desktop (please complete the following information):**
 - OS: KDE neon User edition 5.19
 - Python Version: 3.8.0
 - Mujoco Version: 2.1.1
 - mujoco-py version: 2.1.2.14

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: /home/federima/.mujoco/mujoco210/bin:/usr/lib/nvidia
 - output of: `echo $HOME`: /home/federima
 - output of: `echo $USER`: federima

fmalato, 2022-04-19:
> have you fixed the issue? I also meet the same issue, and try many methods until now . I don't know how to deal with the problem.

Yes, I managed to "solve" by using a legacy version of MuJoCo (namely 2.0) with the free license and `mujoco-py` 2.0.2.8...

guodashun, 2022-04-20:
I solved this problem by downgrading the mujoco version from 2.1.x to 2.1.0.
There may be something incompatible in the higher version.