openai/mujoco-py #707: distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

wj210, 2022-04-26:
This error happened while i was trying to import mujoco_py in a docker file, i tried solutions for "x86_64-linux-gnu-gcc" error but it seems the cause of the error is different and the solution does not help. 
Here is the error log:
```
running build_ext
building 'mujoco_py.cymj' extension
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib/python3.6
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib/python3.6/site-packages
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib/python3.6/site-packages/mujoco_py
creating /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib/python3.6/site-packages/mujoco_py/gl
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I.local/lib/python3.6/site-packages/mujoco_py -I/home/user/.local/lib/python3.6/site-packages/mujoco_py -I/home/user/.mujoco/mujoco200/include -I/usr/local/lib/python3.6/dist-packages/numpy/core/include -I/usr/include/python3.6m -c /home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/user/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_f_9mujoco_py_4cymj_9PyMjModel__set':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:31875:80: error: 'mjModel {aka struct _mjModel}' has no member named 'key_mpos'; did you mean 'key_qpos'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_2d(__pyx_v_p->key_mpos, __pyx_v_p->nkey, (3 * __pyx_v_p->nmocap))); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 1494, __pyx_L1_error)
                                                                                ^~~~~~~~
                                                                                key_qpos
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:31890:80: error: 'mjModel {aka struct _mjModel}' has no member named 'key_mquat'; did you mean 'body_quat'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_2d(__pyx_v_p->key_mquat, __pyx_v_p->nkey, (4 * __pyx_v_p->nmocap))); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 1495, __pyx_L1_error)
                                                                                ^~~~~~~~~
                                                                                body_quat
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_12PyMjvPerturb_7active2___get__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:69927:55: error: 'mjvPerturb {aka struct _mjvPerturb}' has no member named 'active2'; did you mean 'active'?
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->active2); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3352, __pyx_L1_error)
                                                       ^~~~~~~
                                                       active
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_12PyMjvPerturb_7active2_2__set__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:69982:22: error: 'mjvPerturb {aka struct _mjvPerturb}' has no member named 'active2'; did you mean 'active'?
   __pyx_v_self->ptr->active2 = __pyx_v_x;
                      ^~~~~~~
                      active
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_f_9mujoco_py_4cymj_10PyMjvScene__set':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:76537:82: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framergb'; did you mean 'camera'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_float_1d((&(__pyx_v_p->framergb[0])), 3)); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3743, __pyx_L1_error)
                                                                                  ^~~~~~~~
                                                                                  camera
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_10PyMjvScene_10framewidth___get__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:77244:53: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framewidth'
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->framewidth); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3774, __pyx_L1_error)
                                                     ^~
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_10PyMjvScene_10framewidth_2__set__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:77299:20: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framewidth'
   __pyx_v_self->ptr->framewidth = __pyx_v_x;
                    ^~
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_9linewidth___get__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:78679:34: error: incompatible type for argument 1 of 'PyFloat_FromDouble'
   __pyx_t_1 = PyFloat_FromDouble(__pyx_v_self->ptr->linewidth); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3878, __pyx_L1_error)
                                  ^~~~~~~~~~~~
In file included from /usr/include/python3.6m/Python.h:93:0,
                 from /home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:63:
/usr/include/python3.6m/floatobject.h:45:24: note: expected 'double' but argument is of type 'float *'
 PyAPI_FUNC(PyObject *) PyFloat_FromDouble(double);
                        ^~~~~~~~~~~~~~~~~~
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_9linewidth_2__set__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:78734:32: error: assignment to expression with array type
   __pyx_v_self->ptr->linewidth = __pyx_v_x;
                                ^
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_12legendoffset___get__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:78865:55: error: 'mjvFigure {aka struct _mjvFigure}' has no member named 'legendoffset'; did you mean 'legendoff'?
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->legendoffset); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3886, __pyx_L1_error)
                                                       ^~~~~~~~~~~~
                                                       legendoff
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_12legendoffset_2__set__':
/home/user/.local/lib/python3.6/site-packages/mujoco_py/cymj.c:78920:22: error: 'mjvFigure {aka struct _mjvFigure}' has no member named 'legendoffset'; did you mean 'legendoff'?
   __pyx_v_self->ptr->legendoffset = __pyx_v_x;
                      ^~~~~~~~~~~~
[... 47 lines cut ...]
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
[... cut ...]
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
The command '/bin/sh -c python3 -c "import mujoco_py"' returned a non-zero code: 1
```
did anyone encountered similar error? its showing this weird logs such as "no such member"... that i do not understand.

swagatk, 2022-05-03:
In my case, I get this error when I import the mujoco_py module in python. I installed the package in a conda environment on Ubuntu 20.04 without any error. 

```
~/.mujoco/mujoco-py/examples$ python3
Python 3.8.13 (default, Mar 28 2022, 11:38:47) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
running build_ext
gcc -pthread -B /home/swagat/anaconda3/envs/mujoco_py/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py -I/home/swagat/.mujoco/mujoco210/include -I/home/swagat/.local/lib/python3.8/site-packages/numpy/core/include -I/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/vendor/egl -I/home/swagat/anaconda3/envs/mujoco_py/include/python3.8 -c /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/cymj.c -o /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/swagat/anaconda3/envs/mujoco_py/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py -I/home/swagat/.mujoco/mujoco210/include -I/home/swagat/.local/lib/python3.8/site-packages/numpy/core/include -I/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/vendor/egl -I/home/swagat/anaconda3/envs/mujoco_py/include/python3.8 -c /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/gl/eglshim.c -o /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/gl/eglshim.o -fopenmp -w
gcc -pthread -shared -B /home/swagat/anaconda3/envs/mujoco_py/compiler_compat -L/home/swagat/anaconda3/envs/mujoco_py/lib -Wl,-rpath=/home/swagat/anaconda3/envs/mujoco_py/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/cymj.o /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/gl/eglshim.o -L/home/swagat/.mujoco/mujoco210/bin -R/home/swagat/.mujoco/mujoco210/bin -lmujoco210 -lglewegl -o /home/swagat/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/mujoco_py-2.1.2.14-py3.8.egg/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/lib.linux-x86_64-3.8/mujoco_py/cymj.cpython-38-x86_64-linux-gnu.so -fopenmp
gcc: error: unrecognized command line option ‘-R’
Traceback (most recent call last):
  File "/home/swagat/.local/lib/python3.8/site-packages/setuptools/_distutils/unixccompiler.py", line 204, in link
    self.spawn(linker + ld_args)
  File "/home/swagat/.local/lib/python3.8/site-packages/setuptools/_distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/home/swagat/.local/lib/python3.8/site-packages/setuptools/_distutils/spawn.py", line 87, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/usr/bin/gcc' failed with exit code 1

During handling of the above exception, another exception occurred:
```

zzu-zhangzhen, 2022-05-13:
> What packages do you have installed？

rems75, 2022-05-13:
@wj210 I had the exact same issue. 
Solved it by upgrading to Mujoco 210 (found e.g here https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz) and reinstalling mujoco-py.

AthanasiosPetsanis, 2022-10-15:
> I can confirm that reinstalling Mujoco210 & compiling mujoco-py from source as suggested in [this](https://docs.google.com/document/d/1eBvfKoczKmImUgoGMbqypODBXmI1bD91/edit) link solved the above problem. Thanks @rems75 for the suggestion.

Following the steps in the link you suggest fixed it for me. (I have Ubuntu, thus the conda steps were unnecessary and I omitted them). Thank you!

YSheng17, 2023-03-05:
RUN pip3 install --user mujoco-py==2.0.2.8 before import mujoco_py

mg19950, 2023-04-03:
Upgrading to Mujoco 210 is not solving the issue for me, I am still getting the same error! :/ What should I do?

zebin-huang, 2023-09-10:
> RUN pip3 install --user mujoco-py==2.0.2.8 before import mujoco_py

this command works for me. Ubuntu 20.04 with Mujoco 210