openai/mujoco-py #708: mujoco_py/cymj.c:84607:22: error: ‘mjrContext’ {aka ‘struct mjrContext_’} has no member named ‘glewInitialized’; did you mean ‘glInitialized’?

sheepc, 2022-04-30:
**Describe the bug**
when import mujoco_py report this error, I had try to change gcc version 7 8 9,  but is no useful. 

**Error Messages**
>> import mujoco_py
Compiling /home/sheepc/mujoco-py/mujoco_py/cymj.pyx because it depends on /home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/Cython/Includes/cpython/type.pxd.
[1/1] Cythonizing /home/sheepc/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/sheepc/anaconda3/envs/mujoco/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/sheepc/mujoco-py/mujoco_py -I/usr/local/mujoco/include -I/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/numpy/core/include -I/home/sheepc/mujoco-py/mujoco_py/vendor/egl -I/home/sheepc/anaconda3/envs/mujoco/include/python3.8 -c /home/sheepc/mujoco-py/mujoco_py/cymj.c -o /home/sheepc/mujoco-py/mujoco_py/generated/_pyxbld_2.1.2.14_38_linuxgpuextensionbuilder/temp.linux-x86_64-3.8/home/sheepc/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/sheepc/mujoco-py/mujoco_py/cymj.c: In function ‘__pyx_pf_9mujoco_py_4cymj_12PyMjrContext_15glewInitialized___get__’:
/home/sheepc/mujoco-py/mujoco_py/cymj.c:84552:55: error: ‘mjrContext’ {aka ‘struct mjrContext_’} has no member named ‘glewInitialized’; did you mean ‘glInitialized’?
84552 |   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->glewInitialized); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 4175, __pyx_L1_error)
      |                                                       ^~~~~~~~~~~~~~~
      |                                                       glInitialized
/home/sheepc/mujoco-py/mujoco_py/cymj.c: In function ‘__pyx_pf_9mujoco_py_4cymj_12PyMjrContext_15glewInitialized_2__set__’:
/home/sheepc/mujoco-py/mujoco_py/cymj.c:84607:22: error: ‘mjrContext’ {aka ‘struct mjrContext_’} has no member named ‘glewInitialized’; did you mean ‘glInitialized’?
84607 |   __pyx_v_self->ptr->glewInitialized = __pyx_v_x;
      |                      ^~~~~~~~~~~~~~~
      |                      glInitialized
Traceback (most recent call last):
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/unixccompiler.py", line 173, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] +
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/ccompiler.py", line 917, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/spawn.py", line 68, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/usr/bin/gcc' failed with exit code 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sheepc/mujoco-py/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 296, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 249, in _build_impl
    dist.run_commands()
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/sheepc/mujoco-py/mujoco_py/builder.py", line 149, in build_extensions
    build_ext.build_extensions(self)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/sheepc/anaconda3/envs/mujoco/lib/python3.8/site-packages/setuptools/_distutils/unixccompiler.py", line 176, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/bin/gcc' failed with exit code 1
>>> 

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Python Version 3.8
 - Mujoco Version 2.1.5
 - mujoco-py version 2.1.2.14

**Environment**
export MUJOCO_PY_MUJOCO_PATH=/usr/local/mujoco
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/mujoco/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so

ksshen0000, 2022-05-01:
I guess that the contributor of mujoco is changing their variable names, only the mujoco210 is available, you can download from this page. https://github.com/deepmind/mujoco/releases/tag/2.1.0

sheepc, 2022-05-01:
Thanks you! It is useful

closed 2022-05-01 by sheepc

langfengQ, 2022-05-21:
> I guess that the contributor of mujoco is changing their variable names, only the mujoco210 is available, you can download from this page. https://github.com/deepmind/mujoco/releases/tag/2.1.0

Thanks!

sohamohsam48, 2022-07-16:
> I guess that the contributor of mujoco is changing their variable names, only the mujoco210 is available, you can download from this page. https://github.com/deepmind/mujoco/releases/tag/2.1.0

Thank you

KE7, 2023-06-10:
This won't work for ARM based macs using M1/M2 :(