openai/mujoco-py #709: errors.CompileError: command

zzu-zhangzhen, 2022-05-03:
when i import nujoco-py , the issue is occur,if you deal with the issue ,please help me ! thanks!

C:\Users\12995\Anaconda\envs\safety-gym\python.exe C:/Users/12995/PycharmProjects/test/main.py
Compiling C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.pyx because it changed.
[1/1] Cythonizing C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib\site-packages
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py
creating C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\gl
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py -IC:\Users\12995\.mujoco\mujoco210\include -IC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\numpy\core\include -IC:\Users\12995\Anaconda\envs\safety-gym\include -IC:\Users\12995\Anaconda\envs\safety-gym\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /TcC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py/gl/dummyshim.c /FoC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py/gl/dummyshim.obj -fopenmp -w
dummyshim.c
cl: ������ warning D9025 :������д��/W3��(�á�/w��)
cl: ������ warning D9002 :����δ֪ѡ�-fopenmp��
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py -IC:\Users\12995\.mujoco\mujoco210\include -IC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\numpy\core\include -IC:\Users\12995\Anaconda\envs\safety-gym\include -IC:\Users\12995\Anaconda\envs\safety-gym\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /TcC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.c /FoC:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\generated\_pyxbld_2.1.2.14_37_windowsextensionbuilder\temp.win-amd64-cpython-37\Release\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.obj -fopenmp -w
cl: ������ warning D9025 :������д��/W3��(�á�/w��)
cl: ������ warning D9002 :����δ֪ѡ�-fopenmp��
cymj.c
C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.c(84546): error C2039: "glewInitialized": ���� "mjrContext_" �ĳ�Ա
C:\Users\12995\.mujoco\mujoco210\include\mjrender.h(72): note: �μ���mjrContext_��������
C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.c(84546): error C2198: ��__Pyx_PyInt_From_int��: ���ڵ��õĲ���̫��
C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\cymj.c(84601): error C2039: "glewInitialized": ���� "mjrContext_" �ĳ�Ա
C:\Users\12995\.mujoco\mujoco210\include\mjrender.h(72): note: �μ���mjrContext_��������
Traceback (most recent call last):
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 412, in compile
    self.spawn(args)
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 517, in spawn
    return super().spawn(cmd, env=env)
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\ccompiler.py", line 917, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\spawn.py", line 69, in spawn
    "command %r failed with exit code %s" % (cmd, exitcode))
distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/12995/PycharmProjects/test/main.py", line 1, in <module>
    import mujoco_py
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\mujoco_py\builder.py", line 249, in _build_impl
    dist.run_commands()
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\dist.py", line 967, in run_commands
    self.run_command(cmd)
[... cut ...]
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "C:\Users\12995\Anaconda\envs\safety-gym\lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 414, in compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

Process finished with exit code 1

henbudidiao, 2022-05-12:
Have you solved it? I have the same problem. I reinstalled vs and changed environment variables, which didn't work.

zzu-zhangzhen, 2022-05-13:
> Have you solved it? I have the same problem. I reinstalled vs and changed environment variables, which didn't work.
I reinstalled the Ubuntu system

henbudidiao, 2022-05-13:
I have solved.I still use win10.See link:https://zhuanlan.zhihu.com/p/502112539

我已经解决了此问题，还是用的win10系统，解决方法请看这个链接：https://zhuanlan.zhihu.com/p/502112539

zzu-zhangzhen, 2022-05-14:
> I have solved.I still use win10.See link:https://zhuanlan.zhihu.com/p/502112539
> 
> 我已经解决了此问题，还是用的win10系统，解决方法请看这个链接：https://zhuanlan.zhihu.com/p/502112539
 

Thanks!!!!